UtilitechAS/amsreader-firmware #764: Error in GUI for amperage

tronde-ams, 2024-04-08:
Norway /230V 3ph IT / aidon 6525 / generic esp32

v. 2.2.8 showed I2 as "not available" (correct). v. 2.3.1 shows both 0.00A and 1.0A and messes up with text over text.

[image]

rogerthn2019, 2024-04-08:
Negative AC current?
[image]

gskjold, 2024-04-08:
> v. 2.2.8 showed I2 as "not available" (correct). v. 2.3.1 shows both 0.00A and 1.0A and messes up with text over text.

Found problem, will fix

> Negative AC current?

This is either incorrect parsing of data from your meter or the data sent is not accurate. Would be great if you could run a verbose debug and send me a copy of the payload

tronde-ams, 2024-04-08:
> Found problem, will fix
>
>
> This is either incorrect parsing of data from your meter or the data sent is not accurate. Would be great if you could run a verbose debug and send me a copy of the payload

Can you use this?
[debug.txt](https://github.com/UtilitechAS/amsreader-firmware/files/14910861/debug.txt)

Values for I2 does not change. Have been stuck all day.
My meter is Aidon 6525, correctly identifeid by the interface.

tronde-ams, 2024-04-08:
> Negative AC current? [...]

Aidon 6525 does not give value for I2. If you try to calculate it, you can end up with negative values, which is obviously wrong. That is why I2 should be shown as "not available" as it was in FW 2.2.8.

gskjold, 2024-04-09:
@tronde-ams please try this firmware:
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14916281/esp32.zip)

@rogerthn2019 has production, so he should se negative numbers. Also, he is on a kamstrup meter. This is where i need the debugging

tronde-ams, 2024-04-09:
> @tronde-ams please try this firmware: [esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14916281/esp32.zip)
> 

It is OK

rogerthn2019, 2024-04-09:
@gskjold Do you need more?
[telnet_2024-04-09.log](https://github.com/UtilitechAS/amsreader-firmware/files/14916787/telnet_2024-04-09.log)

rogerthn2019, 2024-04-09:
PS
I do have Import and Export at the same time but this is not visible on the Real time plot.
It could be the total sum that is shown on the Real time plot?
[image]

gskjold, 2024-04-09:
@rogerthn2019 Inspected the data from your meter, but the fields for current is given in "unsigned long" format, which does not contain a signed value, thus there is no way for the firmware to know which phase is importing or exporting.

rogerthn2019, 2024-04-09:
> @rogerthn2019 Inspected the data from your meter, but the fields for current is given in "unsigned long" format, which does not contain a signed value, thus there is no way for the firmware to know which phase is importing or exporting.

OK but negative AC current?

closed 2024-04-09 by gskjold

ArnieO, 2024-04-09:
@gskjold : Is export taken into account in some way the new Real time plot?

rogerthn2019, 2024-04-10:
> The format your meter uses is unable to report negative values, so it will always be positive, even when you are exporting.

OK, I can see that the per phase values (data.json) only have Voltage and Current.
But, in AC systems Power (W, var, VA?) can be positive and negative but Current (A) can't?

ArnieO, 2024-04-10:
> OK, I can see that the per phase values (data.json) only have Voltage and Current.
> But, in AC systems Power (W, var, VA?) can be positive and negative but Current (A) can't?

The meters report import and export as separate parameters, both with positive values. 
Current (A) is only reported as a positive number (without sign). 

Some 3-phase meters seem to report import and export power simultaneously (presumably if there is export on on phase and import on other(s)). But still with only positive current values. 

Other meters seem to always report the sum of export and import power - so that only one of those is non-zero at any time.

rogerthn2019, 2024-04-10:
> The meters report import and export as separate parameters, both with positive values. Current (A) is only reported as a positive number (without sign).

Hence my question: Negative AC current?
[image]

ArnieO, 2024-04-10:
> Hence my question: Negative AC current?

OK, I understand what you mean now! 😄 
If you look under the first posting in this thread, this is tagged as a bug two days ago. 😉 
Not 100% sure if it is fixed with v2.3.2/2.3.3, as it is not explicitly mentioned in the release notes. -> @gskjold ?

gskjold, 2024-04-10:
The original issue is fixed.

When you have production, meters report negative current when exporting on that phase.

rogerthn2019, 2024-04-10:
> The original issue is fixed.

Agree!

> When you have production, meters report negative current when exporting on that phase.

Do not agree and I think that @ArnieO do agree with me?

> Current (A) is only reported as a positive number (without sign).

ArnieO, 2024-04-10:
> The original issue is fixed.
> 
> When you have production, meters report negative current when exporting on that phase.

Does not seem consistent with what you wrote here? https://github.com/UtilitechAS/amsreader-firmware/issues/764#issuecomment-2045572218

gskjold, 2024-04-10:
True, his meter does not, but others do. Firmware does not know

rogerthn2019, 2024-04-10:
It might be that some meters have a flag for "negative current" but as far as I know
**Current (A) cannot be negative in AC systems?**

tronde-ams, 2024-04-10:
> It might be that some meters have a flag for "negative current" but as far as I know **Current (A) cannot be negative in AC systems?**

That is correct. No negative AC current. 

BUT - it is not wrong if a meter _defines_ power in to the house as positive and power out of the house as negative. 

The power meter is there to figure out how much you should pay for energy. If power in is positive, and power out is negative you end up by paying (or getting paid for the difference. P = P_in + (-P_out).

We must remember that energy accunting is the main objective for the meter. Values for voltages and currents are are a "nice to have" addition.

Another thing to consider, is that the accuracy of power meters is not that great. Most of them are certified to +/- 1% accuracy.