UtilitechAS/amsreader-firmware #765: Has something changed regarding static IP?

tronde-ams, 2024-04-08:
Don't know whether it is a bug or just me being clueless.

Generic Esp32.

All versions including 2.2.8 followed the settings for static IP, but 2.3.1 does not seem to care about it. 
This is not a big problem for me, because I know how to force my DHCP to assign one spesific IP.

I don't know much about network, but I find it kind of strange that it changed just when I upgraded to 2.3.1. Don't spend time on this if the problem is considered to be with me, I live well with tweaking DHCP in the router.

rogerthn2019, 2024-04-08:
Thanks!
Was about to upgrade and might need to adjust my DHCP.

ArnieO, 2024-04-08:
Nothing should be changed regarding static IP / DHCP. 

> All versions including 2.2.8 followed the settings for static IP, but 2.3.1 does not seem to care about it.

Can you elaborate on what your issue is? I use static IP set in the device and do not see any problem.

tronde-ams, 2024-04-08:
> Nothing should be changed regarding static IP / DHCP.
> 
> > All versions including 2.2.8 followed the settings for static IP, but 2.3.1 does not seem to care about it.
> 
> Can you elaborate on what your issue is? I use static IP set in the device and do not see any problem.

2.2.8 had setting for static IP used since very long time ago over many differnt FW. All versions including that came up with that IP after a upgrade to new FW, but 2.3.1 got assigned a dynamic IP. Same on two different units. As I said, I know wery little about network, so the problem can very well be with me or some setting in the router I don't understand or know about. I have solved the issue by tweaking the DHCP table, so no real problem, but since this FW behaved differently I felt I should ask.

rogerthn2019, 2024-04-08:
Static IP OK
[image]
[image]

ArnieO, 2024-04-08:
> I have solved the issue by tweaking the DHCP table, so no real problem, but since this FW behaved differently I felt I should ask.

We appreciate that! There are many changes in v2.3 compared to 2.2, so there could be unresolved issues. 

Just to be sure, I did an upgrade from v2.2.28 of a device with static IP now, upgrading to v2.3.1. 
It came up on its static IP after upgrade. So it seems to work as it should.

LeonSnow0, 2024-04-09:
I seem to have the same problem with the static ip not sticking after updating to 2.3.1

This is the current ip on the POW-K+
[image]

And this is my settings
[image]

When i reboot the device it gets the correct ip, but after a while it seems to get a new ip from the dhcp server for some unknown reason. This have happend 2-3 times since i updated yesterday

ArnieO, 2024-04-09:
Potentially same issue as https://github.com/UtilitechAS/amsreader-firmware/discussions/738#discussioncomment-9061913

gskjold, 2024-04-10:
Please try the attached firmware:
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14936395/esp8266.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14936397/esp32c3.zip)
[esp32s3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14936398/esp32s3.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14936399/esp32.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14936400/esp32s2.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14936401/esp32solo.zip)

https://github.com/UtilitechAS/amsreader-firmware/wiki/Flashing-via-web-GUI

LeonSnow0, 2024-04-11:
My provider is also altibox.

Router is a Genexis FiberTwist FT1 and the POW-K+ is connected through a ZyXEL WX3401 wireless ap

ArnieO, 2024-04-11:
@tronde-ams @LeonSnow0 : Thank you for additional info.
Can you check if the test version in this message remedies the issue? https://github.com/UtilitechAS/amsreader-firmware/issues/765#issuecomment-2048217607

LeonSnow0, 2024-04-11:
I flashed the firmware now.

After the update it was still using the DHCP address, but after i switched settings from static to DHCP and back it came up with the correct ip.
I'll leave it for a day or two and see if it sticks to the static ip or not and give you an update.

tronde-ams, 2024-04-11:
I tested my other unit just now, and it does not stick to Static IP.

Buth units appear to be equal hardware (160 MHz Esp32). One unit behaves as expected, one not.

I did also try to change back and forth between static and DHCP to no avail.

???

tronde-ams, 2024-04-11:
Forget about previous post.

Stupid me forgot to upgrade FW :-(. Really sorry!

The other unit does also seem to behave as expected with new FW.

gskjold, 2024-04-11:
How found the issue, new test firmware:
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14950910/esp32.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14950911/esp32solo.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14950912/esp32s2.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14950913/esp32c3.zip)
[esp32s3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14950914/esp32s3.zip)

LeonSnow0, 2024-04-11:
After flashing e080c7d network configuration settings are behaving normally and changes takes effect. Booted with the static ip and hopefully stays on it.

tronde-ams, 2024-04-11:
e080c7d appears to be OK on both units. Have tried differnent IPs too.

LeonSnow0, 2024-04-12:
Seems like the problem is fixed for me. My device have stayed on the static ip for the last 22 hours and there haven't been any other issues.

tronde-ams, 2024-04-13:
My units have been up for more than 36 hours on same static IP by now.

closed 2024-04-13 by gskjold