UtilitechAS/amsreader-firmware #766: Import value stuck

Cartoon-Dude, 2024-04-09:
Since the update to v2.3.1, the import value is stuck.

[image]
[image]

It's staying to the highest value, and also not logical to have both import and export at the same time.
With a reboot the import value is reseted, and back to displaying the highest.

ESP8266
Iskra AM550

ArnieO, 2024-04-09:
Which of the values is stuck - Power or Energy?

gskjold, 2024-04-09:
The power, this specific meter should only show import or export, not both. I just found the issue.

closed 2024-04-09 by gskjold