UtilitechAS/amsreader-firmware #767: Real time plot and user interface settings disappear

ma10as123, 2024-04-09:
**Describe the bug**
Real time plot and user interface settings disappear when updating to 2.3.2
[image]
[image]
[image]
[image]

gskjold, 2024-04-09:
Thanks!
2.3.3. should fix your issue

ma10as123, 2024-04-09:
Yes it works now :)

closed 2024-04-09 by ma10as123