UtilitechAS/amsreader-firmware #772: Blue LED after update

sm6tbe, 2024-04-12:
Hi,
I did an update (V2.3.1) via WIFI(webb browser), and after that the Blue LED is always on (impossible to log into it). Try to reset but still only blue LED always on.
Try to update via USB cable but all USB comports become COM3. so its impossible to push in firmware. It look like be a brick.
How to make it work? Possible to use FTDI?
Br
Patrik

**Hardware information:**
 - Country: Sweden
 - Meter: Aidon
 - Encryption enabled [yes/no]
 - AMS reader: POW-P1
 - Release v2.3.1

ArnieO, 2024-04-12:
Stuck blue light does not sound good, I have difficulties understanding how even a bricked firmware could do that - as the ble part of the LED is not controlled by the microcontroller, but from the incoming data signal. Did something else, more "physical"  happen in the same instance?

It is possible to reflash over FTDI (see Firmware User Manual that you can [download here](https://www.amsleser.no/index.php?controller=attachment&id_attachment=53), chapter 9). Note that the Rx/Tx marking on the PCB might be confusing. 

But FTDI should not be necessary, as the microcontroller has "native" USB. After you have connected USB cable to the computer, a COM port comes up when you do this:
- Press and hold AP button
- Press and release RES button
- Release AP button

But the steady blue light is bothering me. There might be ha hardware failure on your board, in which case we will of course send you a new one. 

But try the reflashing first (with [newest release firmware v2.3.3](https://github.com/UtilitechAS/amsreader-firmware/releases/download/v2.3.3/ams2mqtt-esp32s2-2.3.3.zip)). 
If unsuccessful please email me on post@amsleser.no

sm6tbe, 2024-04-12:
I have same feeling, the blue LED always on is not a good sign, this came
after upgrade firmware to V2.3.1, first web interface stopped, but it was
still feeding data to my home assistant, so i wanted to upgrade to V2.3.3,
but after disconnect it from the electric meter, it never worked again,
always blue LED.
I have done this
Press and hold AP button and so on....

But still same, and comport always become COM3 on all ports, and its
impossible to get any contact to it. So the i guess it has something to do
with the USB "interface" part.

sm6tbe, 2024-04-12:
> :
> Isn't the LED always blue when disconnected from the meter?

Hello,
Yes, its always on....nice blue colour, but not doing what it is supposed
to do :)

Den fre 12 apr. 2024 kl 12:31 skrev Gunnar Skjold ***@***.***

ArnieO, 2024-04-12:
> Isn't the LED always blue when disconnected from the meter?

Correct. Always blue when disconnected.
I assumed it was connected to the meter while those observations were done.

sm6tbe, 2024-04-12:
Yes when connected also, always blue, no other colour....and basically
stone dead, don't do anything except show blue light. and if try to connect
via USB cable as told before....its like a brick.

ArnieO, 2024-04-12:
> But still same, and comport always become COM3 on all ports, and its
impossible to get any contact to it. So the i guess it has something to do
with the USB "interface" part.

Let us try an other route; it might be easier to discover the right COM port in this way:

This is an online tool that works in Chrome and some other browsers, that allows you to access the local serial ports from a browser. Does not work with Firefox:

- Download and unzip this file: https://github.com/UtilitechAS/amsreader-firmware/releases/download/v2.3.3/ams2mqtt-esp32s2-2.3.3.zip
- Open this site in Chrome browser: https://esp.huhn.me/
- Click the "Connect button". An empty window should open up.
- Press/hold AP, press/release RES, release AP
- The port of the ESP should now show up like this:
[image]
- Select it and press Connect.
- A page will be presented where you can enter the memory location and binary file for the four files that are needed for flashing an ESP32 device. 
- Select each of the four files from the downloaded ZIP file, and indicate the correct memory locations for each of them, like this:
[image]
- Double check that the files and memory locations are set exactly as shown here. 
- Press the "PROGRAM" button.
- You should now see the upload to the device in the "output" window at the bottom of the screen.
- When finished, press the RES button.
- Depending on whether the device still remembers previous settings, you may or may not see "boot" activity on the LED. 
- If no activity on the LED, or if the LED goes to steady green (which is blue and yellow at the same time; assuming device is not connected to meter), search for access point AMS2MQTT - and connect to it.
- Open http://192.168.4.1/ to configure your device.

ArnieO, 2024-04-12:
> Yes when connected also, always blue

Not a good sign if it is steady blue when connected to the meter. If above described attempt to reflash fails, please contact me by email as indicated in previous message to arrange a replacement.

gskjold, 2024-04-12:
Couldn't it also be that the meter disabled the output? Not sure why that would happen, but still

sm6tbe, 2024-04-12:
I will do the test in a previous "post", i just want to reply on this one,
the meter working fine, have a equalizer connected to read the data, and
meter is pushing out nice and fine data.

Den fre 12 apr. 2024 kl 13:32 skrev Gunnar Skjold ***@***.***

sm6tbe, 2024-04-12:
Hello,
i will test this one, and reply.....i cross my fingers.

ArnieO, 2024-04-16:
I just wrote a blog post where the cable update method is explained in more detail: https://www.amsleser.no/module/ets_blog/blog?id_post=34

You said it did not work for you, and it is maybe because you did not have the needed driver. Driver installation is explained in the blog post.

gskjold, 2024-04-21:
Any updates on this?

sm6tbe, 2024-04-21:
Hi,
I think i wrote before what happen, but ok. I uploaded everything according
to instructions, and it worked for about 24hrs, then it stopped again, but
i did same thing again. And after second time, it has since then been
stabile. And no problems.
Im very happy that i got those instructions to use an online upload.
Best regards
Patrik

from mobile

gskjold, 2024-04-21:
Thank you, thats great to hear!

closed 2024-04-21 by gskjold