UtilitechAS/amsreader-firmware #773: Weekend price excusion ignored

blumentarzan, 2024-04-13:
**Describe the bug**
My energy price is defined as followed:
High tariff: mo-fr 07:00 to 19:00
Low tariff: all other time

Created that configuration:
[image]

Today saturday mornig I have still high tariff:
[image]

**To Reproduce**
See configuration above. Check price on a weekend day.

**Expected behavior**
On weekend with that configuration always low tariff

**Screenshots**
See above

**Hardware information:**
 - Country: Switzerland
 - Meter: Landis+Gyr
 - Encryption enabled No
 - AMS reader: ESP32-C3
 - M-bus adapter (if applicable): No

**Relevant firmware information:**
 - Version: 2.3.3
 - MQTT: yes
 - MQTT payload type: Home Assistant
 - HAN GPIO: GPIO7
 - HAN baud and parity: ?
 - Temperature sensors: ?
 - ENTSO-E API enabled: ?

**Additional context**
Thanks for taking care on that. Great project!!

ThorAlex87, 2024-04-13:
I just noticed having a problem that is probably related: Since midnight tonight the tariffs seem to be ignored, but the tax is added as intended: 
[image]
[image]

I had a look at past prices in Home Assistant, and it's clear the tariffs are added correctly until midnight tonight, then they stop being added. I also updated to 2.3.4 today and did a reboot, with no change. I also tried changing from "+" to "fixed" (no idea what the difference is) with no change.

rpirsc13, 2024-04-13:
> I just noticed having a problem that is probably related: Since midnight tonight the tariffs seem to be ignored, but the tax is added as intended (prices are for NO3): ![Untitled](https://private-user-images.githubusercontent.com/111507745/322199863-cb23b646-b351-4e80-9133-39b700bf3c71.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMwMTgzODEsIm5iZiI6MTcxMzAxODA4MSwicGF0aCI6Ii8xMTE1MDc3NDUvMzIyMTk5ODYzLWNiMjNiNjQ2LWIzNTEtNGU4MC05MTMzLTM5YjcwMGJmM2M3MS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxM1QxNDIxMjFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02ZjJkNzU0NTc4ZTdkMWEzNmZkZjkzYTczZTg0ODM5ZTkyYTEyNTMzN2ExMmI2OTY2NTJmYzdjMmI2N2NlYzE5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.VNoBlVDCmzYU7VIJ38m7lfJ8LVat-XO5rC6xw_Hi9QI) ![Untitled2](https://private-user-images.githubusercontent.com/111507745/322199870-f808b0fc-b86b-47d3-b198-28176a331367.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMwMTgzODEsIm5iZiI6MTcxMzAxODA4MSwicGF0aCI6Ii8xMTE1MDc3NDUvMzIyMTk5ODcwLWY4MDhiMGZjLWI4NmItNDdkMy1iMTk4LTI4MTc2YTMzMTM2Ny5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxM1QxNDIxMjFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NTc4NTIwMmZmOTUwOTRlMmYxY2RjNzFlYmQ3ZjU4OGI3OGE0YWU5NjQ0Y2VlMWUxNGYyZDhjODg5NTAxZmRkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.LOlfX_kTr9QNvE-nIUok-rv_Xq1GD9A-snpV2wYM2M0)
> 
> I had a look at past prices in Home Assistant, and it's clear the tariffs are added correctly until midnight tonight, then they stop being added. I also updated to 2.3.4 today and did a reboot, with no change. I also tried changing from "+" to "fixed" (no idea what the difference is) with no change.

Have you tried removing the date constraints ? I have also noticed that price changes are completely ignored for entries with date constraints ( have not tried day constraints )

ThorAlex87, 2024-04-13:
You are right, removing the dates fixed it. Weird, as it worked perfectly  for 6 days and we are well within the date range I had set.

gskjold, 2024-04-14:
Two different issues. The problem with the date restrictions is that date and month have been mixed when sent to backend. Here is a test build that should resolve both issues:
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14970698/esp32s2.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14970699/esp32solo.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14970700/esp8266.zip)
[esp32s3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14970701/esp32s3.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14970702/esp32c3.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14970703/esp32.zip)

ThorAlex87, 2024-04-14:
Thanks, test build solved my issue!

closed 2024-04-21 by gskjold