UtilitechAS/amsreader-firmware #774: Export gauge to show hour price like import

wknutsen, 2024-04-13:
**Is your feature request related to a problem? Please describe.**
No, but its hard to know what the export price is with new price configuration in place in version 2.3.3

**Describe the solution you'd like**
Is it possible to have export price shown on the export gauge like the import gauge ?
[image]

Nice work with the new version it is real nice :)

ThorAlex87, 2024-04-13:
> I understand the interest of such a solution - but export prices vary depending on the grid operator.
> 
> Some give you spot price, some give you a fixed price per kWh, some let you store it in a "bank" allowing you to use it without charge later - and so on.

The one company that offerd fixed export price in Norway has stopped. As for solkonto Fjordkraft is currently the only one with kWh based one, and it is temporarily closed for new customers with rumours sayng it will be back with a 2000kWh limit, making it a lot less appealing. So at least here in in Norway there will soon be very few alternatives to spot price for export. I currently have solkonto, and I'd still like the price shown as suggested, as it is very interesting for comparison. 

Maybe add a "show export prices" toggle to enable this, and also an export price graph similar to the import one?

gskjold, 2024-04-13:
With the new price configuration, this makes sense, can fix this

closed 2024-04-21 by gskjold