UtilitechAS/amsreader-firmware #775: Possible bug while importing real time calculations from FW 2.2.8

tronde-ams, 2024-04-13:
Generic Esp32

I did not pay attention to this before I got my electricity bill today, but something got wrong between 2.2.8 and 2.3x. I have some memory about it being present before 2.3.4, but can't say for sure.

My other unit was not active before 2.3.1, so I can't compare.

I did not save config, so I do only have a screenshot.

This is 2.2.8 and is correct.

[image]

This is 2.3.4 and the value for last month kWh is wrong. It is the same value as last month cost. Can it be a mix-up there?

[image]

tronde-ams, 2024-04-14:
I am certain it is a mix-up in the GUI.

This is from the saved config of 2.3.4 which shows wrong values just now. 

energyaccounting 6 4 6.02 385.35 **1246.35** 0.00 0.00 0.00 10 3.34 2 4.16 3 3.37 4 3.58 5 3.92 **1872.46** 0.00

1872 is present there, but not in the GUI.

gskjold, 2024-04-14:
You are absolutely right about that, thanks!

ArnieO, 2024-04-14:
We received message by email from an other customer who has an issue that seems related. 
- Older Pow-U with ESP8266
- Upgrade from v2.2.28 to 2.3.4

This is shown in GUI, 1868 kWh last month:
[image]

While energy company website says 2819 kWh. 

Also: 
During first 3 hours after upgrade, hourly readings were averaged, while this error message was observed: 
[image]
After that, hourly messages no longer averaged, error message disappeared.

closed 2024-04-21 by gskjold