UtilitechAS/amsreader-firmware #778: Home Assistant: "integration mqtt has state class total_increasing" for monthly, daily and hourly meters

Xrlls, 2024-04-15:
**Describe the bug**
Home Assistant logs:
```
2024-04-12 15:05:10.311 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.elmaaler01_dayuse from integration mqtt has state class total_increasing, but its state is not strictly increasing. Triggered by state 8.8 (8.84) with last_updated set to 2024-04-12T13:00:13.384689+00:00. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
```
This happens for the monthly, daily and hourly meters for produced and used when they reset.

This is similar to https://github.com/UtilitechAS/amsreader-firmware/issues/620 except for the information on what to change below :)

**To Reproduce**
Connect to Home Assistant using MQTT, wait for monthly, hourly or daily meters to reset.

**Expected behavior**
No errors should be logged.

**Screenshots**
N/A

**Hardware information:**
N/A

**Relevant firmware information:**
https://developers.home-assistant.io/docs/core/entity/sensor/#available-state-classes

For a meter that can change both in positive and negative direction state class `total_increasing` is not applicable because it implies that the meter only increases. Instead it should be either state class `measurement` or `total`. As it is not a total meter in the classical understanding I would prefer `measurement`.

**Additional context**
Add any other context about the problem here.

stigvi, 2024-04-18:
[image]

There should be no warnings as reset to zero is allowed and expected. But decreasing a small amount is not allowed.

gskjold, 2024-04-19:
Very interesting indeed

gskjold, 2024-04-19:
Looking closer at the warning:

> Triggered by state 8.8 (8.84) 

This is most likely caused by a reboot of the device, causing the current hour to reset which in turn caused the daily and monthly values to decrease

closed 2024-04-19 by gskjold

Xrlls, 2024-04-22:
Thanks for the analysis! I will make some tests with reboots of the device to see whether this reproduces the issue.