UtilitechAS/amsreader-firmware #779: P1 not working after latest update

ArnieO, 2024-04-15:
Hej!Efter senaste uppdateringen så fungerar inte min Pow -P1, den blinkar grönt men webbsidan fungerar inte. Provat att dra ur sladden

ArnieO, 2024-04-15:
@Farjestad:
Which version did you update to?

Farjestad, 2024-04-15:
> Moved from other thread, message from @Farjestad:

Den senaste som kom här om dagen. Kommer inte ihåg siffrorna Skickat från min Galaxy

ArnieO, 2024-04-15:
Some users had an issue with v2.3.3, when used with static IP: It reverted to DHCP, and was given a different IP address by the router, see https://github.com/UtilitechAS/amsreader-firmware/issues/765

The issue should be fixed with v2.3.4 that was released Saturday at 11:46. If updated to latest before that, this could be the reason for your issue. 

If so, please download v2.3.4 from this link: https://github.com/UtilitechAS/amsreader-firmware/releases/tag/v2.3.4

Then use the method described here to upgrade: https://github.com/UtilitechAS/amsreader-firmware/issues/772#issuecomment-2051547787

Farjestad, 2024-04-15:
Sorry this tool did not work, and now it is blue blink

 

Then use the method described here to upgrade: #772 (comment) <https://github.com/UtilitechAS/amsreader-firmware/issues/772#issuecomment-2051547787>

ArnieO, 2024-04-16:
Sorry, but you need to be a bit more extensive in order to help us understand your situation.

Farjestad, 2024-04-16:
After update I can't reach the P1, did have a static ip and can't ping. And it is not in ip list in DHCP. And blink blue now.HansSkickat från min Galaxy

ArnieO, 2024-04-16:
So this sounds like the problem I mentioned, that should be fixed by v.2.3.4:
 https://github.com/UtilitechAS/amsreader-firmware/releases/tag/v2.3.4

tronde-ams, 2024-04-16:
> HiThe problem I that I can't reach the P1 it is not connected to my wifi network any more.Hans

Use your phone and see if you have a wi-fi network named AMS2MQTT.
If so, connect your phone to that network.

Then enter 192.168.4.1 into the address field of an web browser on your phone, and you should see the initial configuration page for the P1.

Farjestad, 2024-04-17:
Hi!No AMS2MQTT wifi.If I unplugg first green 1sec then some quickly yellow.HansSkickat från min Galaxy

ArnieO, 2024-04-17:
@Farjestad 
I'm trying to understand and help you, but your messages are very short.
Please try to update it over USB cable according to the explanation in this new blogpost: 
https://www.amsleser.no/module/ets_blog/blog?id_post=34

Farjestad, 2024-04-17:
Ok USB Is there a USB on the P1?Skickat från min Galaxy

https://www.amsleser.no/module/ets_blog/blog?id_post=34

ArnieO, 2024-04-17:
> Ok USB Is there a USB on the P1?

Yes. 
P1 user manual: https://www.amsleser.no/index.php?controller=attachment&id_attachment=42

Farjestad, 2024-04-17:
Ok finally finding a usb that's work, so programing done, but can't see any wifi from P1, and now it is blue light and some red blinkSkickat från min Galaxy

ArnieO, 2024-04-17:
OK - so far so good. It remembers the old credentials (incorrect Wifi setting), so we have to do a Factory Reset to reset it. 
Do this:
- Press and hold the AP button until the LED changes to steady white. This will change it to Access point mode. 
- After som seconds you will find that the Wifi network (SSID) named AMS2MQTT appears (look for it on your phone or PC). 
- Connect to AMS2MQTT (no password)
- Open a browser and go to http://192.168.4.1/
- The GUI will appear. 
- Open the configuration page and scroll to the bottom.
- On the left side, there is a button "Factory reset". Use it. 

The device will now restart in AP mode, and has forgotten the old settings. 
- Connect to AMS2MQTT again, and open http://192.168.4.1/ in a browser.
- You should now see v.2.3.4 in the upper left corner (newest firmware). 
- Select SSID (Wifi access point) and provide password. You can now also define static IP again.
- Follow remaining instructions on the screen

Farjestad, 2024-04-17:
Hi!

 

After "Factory reset" can’t find Wi-Fi AMS2MQTT! 

And can see green and some yellow blink

//Hans

ArnieO, 2024-04-17:
Did you check that it had upgraded to v2.3.4?

Farjestad, 2024-04-17:
Used this see fotoSkickat från min Galaxy

ArnieO, 2024-04-17:
To include photo you need to NOT respond by email, but go to this thread on Github.

Farjestad, 2024-04-17:
[image]

https://github.com/UtilitechAS/amsreader-firmware/assets/151819645/80a01735-a008-4574-9f13-c28afe7f61c7

Looking like this now

ArnieO, 2024-04-17:
Thank you! 👍 
Yes, that is the correct files you have used. 
The video is a little short, but from what I see there, it could look like a reboot loop - which is not good. 
Could you be so kind as to post a longer video, at least 10 seconds?

Farjestad, 2024-04-17:
https://github.com/UtilitechAS/amsreader-firmware/assets/151819645/a1290c55-d3d3-4011-9f3a-f4aaf85e06cd

ArnieO, 2024-04-17:
Thank you! What I see here is: 

1. Reboot blink sequence - but it comes two times for some reason, which is strange. 
2. At 00:15 there is an error blink code: 3 blinks. This means it has not been able to connect to your Wi-Fi. (See [Firmware User Manual](https://www.amsleser.no/index.php?controller=attachment&id_attachment=53) chapter 10). 
3. At 00:16 a short blue blink, this is when data arrives from the meter. 
4. At 00:18 the error blink is repeated (you will se it repeat every 3 second). 

Så the main problem here seems to be that the device is not connecting to your Wi-Fi. 

Do the following: 

- Press the AP key until you get steady yellow light. 
- Find and connect to AMS2MQTT
- Open a browser and go to http://192.168.4.1/
- Go to the configuration page and check that you have selected the right access point for your Wi-Fi
- Re-enter your Wi-Fi password, and doble/triple check that you enter it correctly. 
- Post a screenshot of the config page, to let me verify your settings.

Farjestad, 2024-04-18:
[image]

Farjestad, 2024-04-18:
[image]

Farjestad, 2024-04-18:
jorne is my wifi

Farjestad, 2024-04-18:
[image]

ArnieO, 2024-04-19:
Strange indeed. 
The last time I heard of a situation like this, the user discovered (by pure chance!) that a "space" sign (" ") was the last character of his Wifi SSID name.

As I have no better ideas: Could you try to (at least temporarily) rename your Wi-Fi SSID and retry?

Farjestad, 2024-04-19:
Sorry but now I can´t get it AP mode :(

ArnieO, 2024-04-19:
> Sorry but now I can´t get it AP mode :(

If you press and hold the AP button for 5-10 seconds - nothing happens?

Farjestad, 2024-04-19:
[image]
And here you can see no space after jorne

gskjold, 2024-04-20:
I think he meant that a space was accidentally added to the SSID in the amsreader config, not on the router. 

I would recommend disconnecting Pow-P1 from the meter and run it on USB power while setting it up and NOT connect it to the meter until everything works. 

Do a factory reset before you start:
- Click reset
- Immediately after, click and hold AP button
- LED should cycle from green to yellow to red
- When it comes to red, release AP button
- Some flashing will confirm the factory reset
- Your device should now be fully reset and AMS2MQTT AP should be visible
- Configure your Wi-Fi config

When your device has reconnected and you are back in the GUI, set buffer size to 512 and also remove the "802.11b" tick if not already removed.

You can now reconnect to the meter. Let me know how this works out

Farjestad, 2024-04-21:
Ok! lite bit stupid of me, there was easier to use the USB and sitt inside the huse do do this. So now it works!!!!
And thanks for yours patience :-)

//Hans

gskjold, 2024-04-21:
Excellent,thank you for reporting back to us

closed 2024-04-21 by gskjold