UtilitechAS/amsreader-firmware #780: Landis+Gyr E450: Dip in TPi every minute

avandorp, 2024-04-15:
Landis+Gyr E450 as provisioned by Regionalwerke Baden seems to have another "every minute" issue. The value of total imported energy drops slighty every 60 seconds. Looking at the last 13000 values as stored by home assistant shows that the drop is between 0.077 kWh and 0.081 kWh (Usually 0.08 or 0.081). To make sure its not an artifact from home assistant I did look at the values in MQTT explorer:
[image]
My hunch was that it has again something to do with the different payloads the meter sends. But according to the documentation the payload every 5 minutes doesn't have total energy import information:
```OBIS Code; Attribute index	Register
Transmission interval 05s
0-8:25.9.0; 2	Object list push settings end user information 1
0-8:25.9.0; 1	OBIS Code  Push Settings end user information 1
0-0:96.1.0; 2	Device identification 1 (Producer serial number)
1-0:1.7.0; 2	Power import +P
1-0:2.7.0; 2	Power export -P
1-1:1.8.0; 2	Energy import +A (QI + QIV)
1-1:2.8.0; 2	Energy export  -A (QII + QIII)
1-1:5.8.0; 2	Reactive energy +Ri (QI)
1-1:6.8.0; 2	Reactive energy +Rc (QII)
1-1:7.8.0; 2	Reactive energy -Ri (QIII)
1-1:8.8.0; 2	Reactive energy -Rc (QIV)
1-0:31.7.0; 2	Current L1
1-0:51.7.0; 2	Current L2
1-0:71.7.0; 2	Current L3
Transmission interval 60s
0-9:25.9.0; 2	Object list push settings end user information 2
0-9:25.9.0; 1	OBIS Code  Push Settings end user information 2
1-1:1.8.1; 2	Energy import +A (QI + QIV) Tariff 1
1-1:1.8.2; 2	Energy import +A (QI + QIV) Tariff 2
1-1:2.8.1; 2	Energy export -A (QII + QIII) Tariff 1
1-1:2.8.2; 2	Energy export -A (QII + QIII) Tariff 2
```
Does amsreader use the added values from 1-1:1.8.1; 2 and 1-1:1.8.2; 2? Or some sort of averaged value if 1-1:1.8.0; 2 is missing?

I've noticed this when investigating unrelated home assistant issues and coming along those entries in the log files
`entity sensor.ams_0026_tpi has state class total_increasing, but its state is not strictly increasing.`

Definitely a low prio issue and probably something better fixed on the meter end.

Firmware 2.3.3/2.3.4 (older as well, but didn't specifically check)

ArnieO, 2024-04-15:
@gskjold : Does the firmware use 1.8.1 + 1.8.2 if 1.8.0 is missing?

If so, this could maybe be a rounding issue, as sum of 1.8.1 and 1.8.2 may not add up to exactly 1.8.0 due to number format limitation?

gskjold, 2024-04-15:
> Does the firmware use 1.8.1 + 1.8.2 if 1.8.0 is missing?

Yes

ArnieO, 2024-04-15:
> > Does the firmware use 1.8.1 + 1.8.2 if 1.8.0 is missing?
> 
> Yes

OK, so that is probably the issue here. My guess is that it is due to number format limitation, so that 1.8.1 + 1.8.2 != 1.8.0
I don't really see a sensible way around it.