UtilitechAS/amsreader-firmware #781: problem after update

ghozt1989, 2024-04-19:
**Describe your problem**
I got to update the other day and suddenly I can't get any data in anymore. I can see that there is a buffer size selection option now, but what should this be on, it says 64 right now.
Maybe this is where my mistake lies?

**Hardware information:**
 - Country: Denmark
 - Meter: [Kamstrup Omnipower]
 - Encryption enabled [Yes]
 - AMS reader: [ESP32]
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: [2.3.4]
 - MQTT: [yes]
 - MQTT payload type: [e.g. JSON]
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

stigvi, 2024-04-20:
I have tried different buffer sizes. 64 seems small. I currently have it on 2048, but 512 or 256 will probably be fine.

But it is hard to say what the optimal value is........

ArnieO, 2024-04-20:
If you have issues with not getting data after upgrade, there could be several reasons. 
In general, it is a good idea to do a backup of the configuration before updating. If you have issues after an upgrade, upload that config back to the device.

To get help here, please state which hardware you are using, and post a screenshot of your config page. 

The optimal value for the buffer size is the smallest that do not give error messages in red on the top bar. If you get no data at all and no error message, check bitrate and parity.

closed 2024-05-23 by gskjold