UtilitechAS/amsreader-firmware #782: Landis+Gyr E450

olupton, 2024-04-21:
Thanks for the great work! The following is just something I noticed; it is not especially important to me.

(I was also a little surprised, having seen https://github.com/UtilitechAS/amsreader-firmware/issues/740, that the `Q` field is still zero.)

**Describe the bug**
I just updated my Pow-U to v2.3.5, and it has started reporting per-phase currents and voltages (I1, I2, I3, U1, U2, U3) where previously these were zero.

Naively it seems that the reported voltages are off by a factor of 10, as all three are reported as ~24.
An example MQTT JSON payload coming from the meter is
```
{"id": ...,"name": ...,"up":2578,"t":1713702165,"vcc":3.262,"rssi":-81,"temp":-127.00,"data":{"lv":"","meterId": ...,"type":"","P":609,"Q":0,"PO":0,"QO":0,"I1":0.82,"I2":1.51,"I3":0.57,"U1":23.90,"U2":23.80,"U3":23.80,"tPI":1832.609,"tPO":0.000,"tQI":0.000,"tQO":0.000,"rtc":0},"realtime":{"h":0.26,"d":26.9,"t":10,"x":8.96,"he":0.00,"de":0.0}}
```
showing voltages around 24 (V, presumably), currents around 1 (A, presumably) giving a total power around 600 (W, presumably).

**To Reproduce**
Look at MQTT messages.

**Expected behavior**
Values in the nominal 230-240V range.

**Hardware information:**
 - Country: Switzerland
 - Meter: Landis+Gyr E450 
 - Encryption enabled no
 - AMS reader: Pow-U

**Relevant firmware information:**
 - Version: 2.3.5
 - MQTT: yes
 - MQTT payload type: JSON
 Remaining fields: I have not changed any of these settings.
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

avandorp, 2024-04-21:
Do you have information from your provider on the data that is sent? My E450 comes with the following info: https://github.com/UtilitechAS/amsreader-firmware/blob/main/doc/Switzerland/RWB_SmartMeter_Bedienungsanleitung.pdf (see table on page 5). The same meter seems to be provisioned in different ways.

olupton, 2024-04-21:
I'm afraid not. The only thing I've seen from them is https://www.romande-energie.ch/images/files/particuliers/compteurs_intelligents/manuel_compteur_landis.pdf, but I don't think it's very helpful here.

avandorp, 2024-04-21:
You can simply try to set a multiplier for voltages: Setting/Meter/Multipliers/Volt->10.

ArnieO, 2024-04-22:
@olupton : I think your meter sends ASCII, meaning you could run a [Telnet debug](https://www.amsleser.no/module/ets_blog/blog?id_post=24) and translate the hex output to ASCII to read what the meter reports.

arkimos, 2024-04-22:
Is anyone able to see the Amperage in AMS reader, who is using Landis+Gyr E450
My SM is an L+G E450 and they is sending Amperage every 5 sec. like the Import Value (W).
This is validated by decoding the Debugging Data with Guru X.  

It there an option to show, Amperage in UI and send also over MQTT?

(still using V2.3.4. Until now I’m not able to do the V2.3.5 Update)

ArnieO, 2024-04-22:
> (still using V2.3.4. Until now I’m not able to do the V2.3.5 Update)

https://www.amsleser.no/module/ets_blog/blog?id_post=34

arkimos, 2024-04-23:
> > (still using V2.3.4. Until now I’m not able to do the V2.3.5 Update)
> 
> https://www.amsleser.no/module/ets_blog/blog?id_post=34

Nice option this simplified method. 
Flashing goes straight forward and very easy
```
Took 15785ms to write 1307760 bytes
Erase size 0, blocks 0, block size 0x4000, offset 0x0000, encrypted no
Done!
To run the new firmware please reset your device.
```

But after reset, it looks like the device is bricked. The PC goes crazy. Every second comes a blim-blim from the PC, endless. 
No Wifi, and Terminal show endless
`invalid header: 0xffffffff`
I think something went wrong, but not sure what the problem is.

ArnieO, 2024-04-23:
> I think something went wrong, but not sure what the problem is.

That sounds strange... 
Are you sure you used the right files for your microcontroller? 
What hardware are you running?

olupton, 2024-04-23:
> You can simply try to set a multiplier for voltages: Setting/Meter/Multipliers/Volt->10.

I tried this, but it doesn't affect the U1/U2/U3 values published to MQTT as far as I can see.

> I think your meter sends ASCII, meaning you could run a [Telnet debug](https://www.amsleser.no/module/ets_blog/blog?id_post=24) and translate the hex output to ASCII to read what the meter reports.

I tried this too. The telnet debug worked fine, but it was not clear how to interpret the results. For example I see `HDLC frame`, `LLC frame`, `GBT frame`, `DLMS frame` and `Using application data` followed by sequences of bytes (formatted as 2 hex characters). Those bytes are mostly not printable ASCII. The only printable sequence I noticed is the 8-character sequence that is published over MQTT as the `meterId` field in the JSON.

---

As there was talk of amperage graphs, this is what I see with 2.3.5:
[image]

arkimos, 2024-04-25:
Flashed again by using esptool. Looks like flash process is running fine. 
```
esptool.py v4.7.0
Serial port COM3
Connecting...
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: 48:31:b7:86:28:84
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x0014ffff...
Compressed 11360 bytes to 8615...
Wrote 11360 bytes (8615 compressed) at 0x00001000 in 0.2 seconds (effective 530.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.0 seconds (effective 498.1 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 734.6 kbit/s)...
Hash of data verified.
Compressed 1307760 bytes to 835583...
Wrote 1307760 bytes (835583 compressed) at 0x00010000 in 9.9 seconds (effective 1061.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
But after resetting, same behavior. It seems ESP is looping during startup. No WiFi no LED light, nothing.

gskjold, 2024-06-06:
Please try the firmware below:
[esp32s3.zip](https://github.com/user-attachments/files/15633553/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/15633561/esp32solo.zip)
[esp32.zip](https://github.com/user-attachments/files/15633564/esp32.zip)
[esp8266.zip](https://github.com/user-attachments/files/15633567/esp8266.zip)
[esp32c3.zip](https://github.com/user-attachments/files/15633570/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/15633572/esp32s2.zip)

Flashing instructions: https://github.com/UtilitechAS/amsreader-firmware/wiki/Flashing-via-web-GUI

closed 2024-06-15 by gskjold

Mysonemo, 2025-03-19:
> Flashed again by using esptool. Looks like flash process is running fine.
>
> ```
> esptool.py v4.7.0
> Serial port COM3
> Connecting...
> Chip is ESP32-C3 (QFN32) (revision v0.4)
> Features: WiFi, BLE, Embedded [...]

Same issue to me how did you do finally ?

ArnieO, 2025-03-20:
> Same issue to me how did you do finally ?

The referred posting is old. Which firmware version are you having issues with?