UtilitechAS/amsreader-firmware #783: OTA from 2.3.4 to 2.3.5 fails

avandorp, 2024-04-21:
OTA via link in top banner or via "info/firmware/Select firmware file for upload" both fail on my ESP32-C3. Previous upgrade experience would suggest that it might work after some more tries. Currently I've only tried it probably two dozen times. And I'm pretty sure that upgrading via USB after disassembly of the installation would work (I will try that later).

Using the link in the top banner reloads the UI a few seconds after pressing OK, showing "Previous upgrade attempt from v2.3.4 to v2.3.5 failed -100" in the info/firmware box. Debug log shows:

```
[frames]
(V) LNG
(E) Serial buffer overflow
(E) Serial FIFO overflow
(E) Update failed
(W) Used 2401ms to handle web
(V) HDLC frame:
[more frames]
```

Using the upload method takes much longer (and I see my computer having egress network traffic, indicating that the firmware file is sent) and reboots the device. Debug log shows:
```
[frames]
(V) LNG
(D) handleFileUpload file: /firmware.bin
(D) handleFileUpload Open file and write: 0
(D) handleFileUpload Written: 0
(E) Serial buffer overflow
(E) Serial FIFO overflow
(E) Serial FIFO overflow
(E) Serial FIFO overflow
(E) Serial FIFO overflow
(E) Serial FIFO overflow
(E) Serial FIFO overflow
(E) Serial FIFO overflow
(E) Serial FIFO overflow
(E) Serial FIFO overflow
(E) Serial FIFO overflow
(E) Serial FIFO overflow
(E) Serial FIFO overflow
(E) Serial FIFO overflow
(D) handleFileUpload Ended
(W) Used 24474ms to handle web
(E) Serial buffer overflow
[...frames...]
(W) loop() used 24635ms
(I) Successfully connected to MQTT
(I)   Subscribing to [homeassistant/status]
(I) Received online status from HA, resetting sensor status
(V) HDLC frame:
(V) 7E A0 7D CE  FF 03 13 D0  45 E0 40 00  02 00 00 6C 
(V) 02 04 12 00  03 09 06 01  01 01 08 00  FF 0F 02 12 
(V) 00 00 02 04  12 00 03 09  06 01 01 02  08 00 FF 0F 
(V) 02 12 00 00  02 04 12 00  03 09 06 01  01 05 08 00 
(V) FF 0F 02 12  00 00 02 04  12 00 03 09  06 01 01 06 
(V) 08 00 FF 0F  02 12 00 00  02 04 12 00  03 09 06 01 
(V) 01 07 08 00  FF 0F 02 12  00 00 02 04  12 00 03 09 
(V) 06 01 01 08  08 00 FF 0F  02 12 00 00  FF 09 08 
(W) Boundry flag missing
(V)   payload:
(V) 7E A0 7D CE  FF 03 13 D0  45 E0 40 00  02 00 00 6C 
(V) 02 04 12 00  03 09 06 01  01 01 08 00  FF 0F 02 12 
(V) 00 00 02 04  12 00 03 09  06 01 01 02  08 00 FF 0F 
(V) 02 12 00 00  02 04 12 00  03 09 06 01  01 05 08 00 
(V) FF 0F 02 12  00 00 02 04  12 00 03 09  06 01 01 06 
(V) 08 00 FF 0F  02 12 00 00  02 04 12 00  03 09 06 01 
(V) 01 07 08 00  FF 0F 02 12  00 00 02 04  12 00 03 09 
(V) 06 01 01 08  08 00 FF 0F  02 12 00 00  FF 09 08 36 
(V) 33 33 32 36  34 31 33 06  00 00 02 44  06 00 00 00 
(V) 00 06 00 2D  14 96 06 00  00 0F 01 06  00 00 CA A5 
(V) 06 00 00 00  6D 06 00 00  17 25 06 00  0E 27 C7 12 
(V) 00 3E 12 00  81 12 00 6B  69 42 7E 
(D) Sending translation file for language: en
(D) Language file /translations-en.json
[...frames...]
(I) Rebooting
```
Somehow I would expect that handling frames, subscribing/sending to MQTT would be stopped during upgrade and only resumed after reboot. But whether that would solve the problem I don't know.

ArnieO, 2024-04-22:
See also the blogpost "Updating device over USB cable, simplified method" 
https://www.amsleser.no/module/ets_blog/blog?id_post=34

msasdk, 2024-04-22:
I noticed the exact same with 2.3.5 on esp32-c3. This is the first time I've experienced this. MQTT is disabled in my case.

avandorp, 2024-04-22:
> See also the blogpost "Updating device over USB cable, simplified method" https://www.amsleser.no/module/ets_blog/blog?id_post=34

Thanks. I will use USB, but I always have to physically dismantle my installation, so I usually go for OTA first. In how far is the procedure shown in your blog post "simpler" than just using flash.sh? 😁 On a more serious note: Point 9 of your blog post seems to be missing memory locations.

avandorp, 2024-04-22:
> I noticed the exact same with 2.3.5 on esp32-c3. This is the first time I've experienced this. Manual upload did not work either. MQTT is disabled in my case.

I've now also tried with MQTT disabled. No luck.

ArnieO, 2024-04-22:
> In how far is the procedure shown in your blog post "simpler" than just using flash.sh? 😁

Not familiar with that one! At least, this is a solution with no local installs required (except the driver).

> On a more serious note: Point 9 of your blog post seems to be missing memory locations.

Ouch... I did copy/paste and forgot that this editor does not handle graphics directly. 
Fixed now, thank you for the notification!

avandorp, 2024-04-22:
> > In how far is the procedure shown in your blog post "simpler" than just using flash.sh? 😁
> 
> Not familiar with that one! At least, this is a solution with no local installs required (except the driver).

I meant the flash.sh in the released zip-Files (using esptool).

gskjold, 2024-04-22:
Not many Linux users, so the flash.sh is mostly not relevant :)

msasdk, 2024-04-22:
Stuck on 2.3.4 here. Tried to upgrade to 2.3.5 using the provided flash scripts (both using the upgrade and flash argument) and in both cases the device seems to fail to boot (have not logged the UART output). Reverting back to 2.3.4 and the device boots normally. My devices uses the ESP32-C3-MINI-1-N4 module.

avandorp, 2024-04-22:
I wonder what the -100 error is, it comes relatively quickly when updating via the top banner link.

gskjold, 2024-04-23:
I suggest C3 users just stick to 2.3.4 for now, will fix for 2.3.6

gskjold, 2024-04-23:
> I wonder what the -100 error is, it comes relatively quickly when updating via the top banner link.

https://github.com/espressif/arduino-esp32/blob/cf448906b3836fbe9368934713b697469254c62f/libraries/HTTPUpdate/src/HTTPUpdate.h#L35

avandorp, 2024-04-23:
"HTTP_UE_TOO_LESS_SPACE". That explains the fast failing. I will try whether updating by usb works. Might take a few days.

avandorp, 2024-04-24:
No luck updating to 2.3.5 with USB either. The device stays silent after updating, no WiFi or LED. It's still visible via the USB cable and pressing the reset button makes the USB device vanish and reappear, but it neither connects to the wifi nor opens an AP when pressing the AP button. No LEDs blink on reboot or at anytime. Flashing again with 2.3.4 has it working again at once (all partitions or only firmware.bin doesn't matter).

One thing that I've noticed is that when going from 2.3.4 to 2.3.5 is that the USB port vanishes shortly after bootup (possibly after enabling WiFi). To flash a new firmware I have to time the flashing command to approximately 1 second after inserting the USB cable. I did see that with a former version as well (unfortunately I don't remember which one). Downgrading doesn't have that problem, possibly because the WiFi never comes up.

adsa95, 2024-05-19:
I have the same problem with the Pow-K+ trying to upgrade from v2.2.23 to 2.3.5.
[image]

ArnieO, 2024-05-20:
> I have the same problem with the Pow-K+ trying to upgrade from v2.2.23 to 2.3.5.

Can you try using one of the other upgrade methods, as described in [this article](https://www.amsleser.no/module/ets_blog/blog?id_post=34)?

adsa95, 2024-05-20:
> > I have the same problem with the Pow-K+ trying to upgrade from v2.2.23 to 2.3.5.
> 
> Can you try using one of the other upgrade methods, as described in [this article](https://www.amsleser.no/module/ets_blog/blog?id_post=34)?

I tried the method of manually downloading the .bin file from the releases page and upgrade using the web interface, it worked, thanks!

ArnieO, 2024-05-20:
> I tried the method of manually downloading the .bin file from the releases page and upgrade using the web interface, it worked, thanks!

Good to hear! 👍 
We see that some users have this issue, and some even need need the third fallback method explained in the article. 
We are trying to figure out the reason, but have not managed to recreate it yet. 
Our current hypotheses is that there is something in the user Wi-Fi that creates issues - but we do not know what it could be yet.

RoLiFru, 2024-05-31:
I am stuck on 2.3.4 as well. Neither the one-click method nor using the web-interface and uploading the .bin-file worked. Unfortunately, when I try to start method 3 shown here (https://www.amsleser.no/module/ets_blog/blog?id_post=34) I do not find/see the instructions in step 9. As I do not want to brick my device: Can somebody please post the instructions which file has to be put to which memory location? Thanks a lot!

ArnieO, 2024-05-31:
> Can somebody please post the instructions which file has to be put to which memory location? Thanks a lot!

Very sorry that the illustrations were missing in the article, I have inserted them now. (You may need to do F5 or Ctrl+F5 to reload the page.)

RoLiFru, 2024-06-01:
Thanks for the illustrations! I followed along and uploaded the 4 files and got 4 times the confirmation, that the upload was finished. However, after pressing the RES button, the device seems to be stuck in a boot loop with no working program: There is no LED activity, no AP (even after pressing the AP button for some seconds) and when I try to connect the device again to the ESPwebtool, the port continuosly pops up, hides, pops up again and so forth in the connection window. I have repeated the whole procedure (after pressing/holding AP, pressing RES and releasing AP) for several times with the esp32c3-2.3.5 files and the 2.3.4 files - all with the same result. Any help to bring my device back to work would be very much appreciated.

ArnieO, 2024-06-01:
Ah... it is a C3. 
We have had feedback earlier that the C3 binary for v2.3.5 is not working.
Please try the test binary in this thread: https://github.com/UtilitechAS/amsreader-firmware/issues/786#issuecomment-2139355460

RoLiFru, 2024-06-02:
Thank you, Arnie. Tested this binary as well - same result, unfortunately. Maybe I am doing something wrong with the programming? I am sure, I indicated the four files to the memory locactions shown. I noticed that programming would only work for the first file (and not for all four in one step), hence I uploaded the first, then the second, the third and the fourth and then pressed RES. When I upload only one and then press RES I am also in the (as I presume) boot loop.

Noschvie, 2024-06-02:
Same here, have an ESP32-C3 device from @dbeinder and I'm not able to get the device running after an upgrade from 2.2.28 to 2.3.5 using https://esp.huhn.me

ArnieO, 2024-06-02:
There is a known issue with the C3 binaries that @gskjold is working on (but not with highest priority, as we do not use C3 in our products). But of course - we want to publish working binaries for all versions, as far as possible.

RoLiFru, 2024-06-02:
ESP32-C3 v2.3.4 is back to life. The C3-bootloader.bin must be addressed to 0x0000 instead of 0x1000 as shown in the illustration. Would be great to have a working 2.3.5 for C3 published, though.

Noschvie, 2024-06-02:
@RoLiFru thanks for your hint, works using https://esp.huhn.me/ and the binaries from thread: https://github.com/UtilitechAS/amsreader-firmware/issues/786#issuecomment-2139355460

But with 2.3.5 I get an HAN error and will have to start a Telnet session...

avandorp, 2024-06-02:
> ESP32-C3 v2.3.4 is back to life. The C3-bootloader.bin must be addressed to 0x0000 instead of 0x1000 as shown in the illustration. Would be great to have a working 2.3.5 for C3 published, though.

For the correct addresses see the flash.sh file inside the zip file for the targeted platform.

entotentrefirefemskjetten, 2024-06-02:
On a side note, I password protected the config page and cannot remember the Username/password. Is there a default username? Or did I make that as well? lol

ArnieO, 2024-06-03:
> On a side note, I password protected the config page and cannot remember the Username/password. Is there a default username? Or did I make that as well? lol

The solution to that situation is to do an [Emergency factory reset](https://www.amsleser.no/module/ets_blog/blog?id_post=15).

closed 2024-06-06 by gskjold

msasdk, 2024-06-06:
Why closed?

avandorp, 2024-06-15:
> Why closed?

OTA update from 2.3.4 to 2.3.6 worked for my C3 device.