UtilitechAS/amsreader-firmware #785: Better support for switzerland / more sensor

Jack-Harper, 2024-04-25:
The default configuration is only partially working for Switzerland.
There is many missing "sensor" (tension, current, A price, B price,...).

I have the "datasheet" for the communication code for romande energie (see below).
Can this help to have more sensor readouts?

[image]

I am quite new to git and programming sorry if this is inappropriate.

ArnieO, 2024-04-26:
Hi and welcome!
We already know this meter/payload: https://github.com/UtilitechAS/amsreader-firmware/discussions/698

Jack-Harper, 2024-04-26:
Hello, ok thanks for the information. Should I do something with issues? Close it / delete it?

Can I ask the status in the other issue or is it unwelcome?

ArnieO, 2024-04-26:
> Hello, ok thanks for the information. Should I do something with issues? Close it / delete it?
> 
> Can I ask the status in the other issue or is it unwelcome?

Nothing is unwelcome! 
I propose we keep this thread open - but I took the liberty of editing the "Subject" of this thread to better reflect your issue.

And I already asked for status in the other thread earlier this morning: https://github.com/UtilitechAS/amsreader-firmware/discussions/698#discussioncomment-9233329
We will surely get a status update on this from @gskjold when he is available. 

In the mean time: Which hardware are you running? 
(We prefer the start post to use the template so that we get all relevant information to understand the situation.)