UtilitechAS/amsreader-firmware #786: Can't flash XIAO ESP32C3

sonite, 2024-05-02:
**Describe your problem**
I can't get my XIAO ESP32C3 to work with the latest release. I have tried 3 methods:

ESP32C3 Flash download tool V3.9.5
With: esp32c3-2.3.5.bin @ 0x0

In terminal:
22:44:41.861 -> ESP-ROM:esp32c3-api1-20210207
22:44:41.861 -> Build:Feb  7 2021
22:44:41.861 -> rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
22:44:41.861 -> Saved PC:0x4005885e
22:44:41.861 -> SPIWP:0xee
22:44:41.861 -> mode:DIO, clock div:2
22:44:41.861 -> load:0x3c0f0020,len:0x48cf8
22:44:42.106 -> load:0x3fc8ba00,len:0x2c10
22:44:42.106 -> load:0x40380000,len:0x46e0
22:44:42.106 -> load:0x42000020,len:0xe938c

With the separate files:
esp32c3-2.3.5\bootloader.bin @ 0x1000
esp32c3-2.3.5\partitions.bin @ 0x8000
esp32c3-2.3.5\boot_app0.bin @ 0xe000
esp32c3-2.3.5\frimware.bin @ 0x10000

In terminal:
22:47:40.619 -> invalid header: 0xffESP-ROM:esp32c3-api1-20210207
22:47:40.619 -> Build:Feb  7 2021
22:47:40.619 -> rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
22:47:40.619 -> Saved PC:0x4004c0dc
22:47:40.619 -> invalid header: 0xffffffff

Visual Studio + Platformio:

Use the [env:esp32c3]
Build & upload -> 

In terminal:
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf752
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0xf4
load:0x403cc710,len:0x8a0
load:0x403ce710,len:0x2274
entry 0x403cc710

I have used it as a nat router so the hardware works.

sonite, 2024-05-03:
New Setup in Platformio
```python
[env:esp32c3]
platform = espressif32
framework = arduino
board = seeed_xiao_esp32c3
board_build.mcu = esp32c3
board_build.f_cpu = 160000000L
board_build.f_flash = 80000000L
board_build.flash_mode = qio
build_flags = ${common.build_flags}
lib_ldf_mode = off
lib_compat_mode = off
lib_deps = ${esp32.lib_deps}
lib_ignore = ${common.lib_ignore}
extra_scripts = ${common.extra_scripts}
``` 
It's unreliable but AMS2MQTT network is created and found. Can some times load the first config page 192.168.4.1.
But as soon as I have saved Generic ESP and just picked a pin, SAVE. It stops working proper.

Terminal give me and then stops, It's like it just jump over the setting page of network:
```python
09:10:54.086 -> (I) AMS bridge started

09:10:54.086 -> [0m(I) Voltage: 0.00V

09:10:54.086 -> [0m(D) ESP32 LittleFS

09:10:54.086 -> [0m(D)  size: 1441792, used: 8192

09:10:54.118 -> [0m(I) -----------------------------------------------

09:10:54.118 -> [0m(I) --Network configuration--

09:10:54.118 -> [0m(I) SSID:                 ''

09:10:54.118 -> [0m(I) Psk:                  ''

09:10:54.118 -> [0m(I) Hostname:             'ams-ec5d'

09:10:54.118 -> [0m(I) IPv6:                 'No'

09:10:54.118 -> [0m(I) mDNS:                 'Yes'

09:10:54.118 -> [0m(I) 802.11b:              'Yes'

09:10:54.118 -> [0m(I) 

09:10:54.118 -> [0m(I) --MQTT configuration--

09:10:54.118 -> [0m(I) Enabled:              No

09:10:54.118 -> [0m(I) 

09:10:54.118 -> [0m(I) --Web configuration--

09:10:54.118 -> [0m(I) Security:             0

09:10:54.118 -> [0m(I) 

09:10:54.118 -> [0m(I) --Meter configuration--

09:10:54.153 -> [0m(I) HAN RX:               2

09:10:54.153 -> [0m(I) HAN RX pullup         Yes

09:10:54.153 -> [0m(I) Baud:                 0

09:10:54.153 -> [0m(I) Parity:               0

09:10:54.153 -> [0m(I) Invert serial:        No

09:10:54.153 -> [0m(I) Buffer size:          64

09:10:54.153 -> [0m(I) Distribution system:  2

09:10:54.153 -> [0m(I) Main fuse:            40

09:10:54.153 -> [0m(I) Production Capacity:  0

```

gskjold, 2024-05-03:
Try 2.3.4

sonite, 2024-05-03:
> Try 2.3.4

TY so much for replying. 2.3.4 is slightly better, but something breaks after the "AMS reader SNAPSHOT" or vendor settings.

When using the original setup in platfomio:
```python
[env:esp32c3]
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.07.00/platform-espressif32.zip
framework = arduino
board = esp32-c3-devkitm-1
board_build.mcu = esp32c3
build_flags = ${common.build_flags}
lib_ldf_mode = off
lib_compat_mode = off
lib_deps = ${esp32.lib_deps}
lib_ignore = ${common.lib_ignore}
extra_scripts = ${common.extra_scripts}
``` 
This won't allow the Terminal to run. But The AP starts.
I can enter the IP in my browser and: 
[image]

I tried this multiple times and chose the two different board types with UART0 and GPIO2, Save. The Network page is shown for about 10s (192.168.4.1/setup) and it dies.

When using slightly different settings in platformio I get some results from the Terminal. It just seems like the installation process stops when going to network setup. After that the device have saved the "zero" setting and just stops:

```python
[env:esp32c3]
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.10.01/platform-espressif32.zip
framework = arduino
board = seeed_xiao_esp32c3
board_build.mcu = esp32c3
board_build.f_cpu = 160000000L
board_build.f_flash = 80000000L
board_build.flash_mode = qio
build_flags = ${common.build_flags}
lib_ldf_mode = off
lib_compat_mode = off
lib_deps = ${esp32.lib_deps}
lib_ignore = ${common.lib_ignore}
extra_scripts = ${common.extra_scripts}
``` 

```python
15:32:34.235 -> (I) AMS bridge started
15:32:34.235 -> [0m(I) Voltage: 0.00V
15:32:34.235 -> [0m(D) ESP32 LittleFS
15:32:34.235 -> [0m(D)  size: 1441792, used: 8192
15:32:34.347 -> [0m(I) No configuration, booting AP
15:32:34.347 -> [0m(I) Entering setup mode
15:32:34.347 -> [0m(I) (setupHanPort) Setting up HAN on pin -1/-1 with baud 0 and parity 0
15:32:34.514 -> [0m(D) Hardware serial
15:32:34.514 -> [0m

``` 

```python
5:36:55.745 -> (I) AMS bridge started
15:36:55.745 -> [0m(I) Voltage: 0.00V
15:36:55.745 -> [0m(D) ESP32 LittleFS
15:36:55.745 -> [0m(D)  size: 1441792, used: 8192
15:36:55.745 -> [0m(I) -----------------------------------------------
15:36:55.745 -> [0m(I) --Network configuration--
15:36:55.745 -> [0m(I) SSID:                 '  
15:36:55.745 -> [0m(I) Psk:                  '
15:36:55.745 -> [0m(I) IP:                   '   
15:36:55.745 -> [0m(I) Gateway:              '                                 
15:36:55.745 -> [0m(I) Subnet:               '                            
15:36:55.745 -> [0m(I) DNS1:                 '                              ?ж ? ?Є ?J) B  ?0  ?'
15:36:55.745 -> [0m(I) DNS2:                 '       ?ж ? ?Є ?J) B  ?0  ?'
[... 52 lines cut ...]
15:36:55.849 -> [0m(I) 
15:36:55.849 -> [0m(I) --Domoticz configuration--
15:36:55.849 -> [0m(I) Enabled:              No
15:36:55.849 -> [0m(I) 
15:36:55.849 -> [0m(I) --NTP configuration--
15:36:55.849 -> [0m(I) Enabled:              Yes
15:36:55.849 -> [0m(I) Timezone:                             
15:36:55.849 -> [0m(I) Server:                                        
15:36:55.849 -> [0m(I) DHCP:                 Yes
15:36:55.849 -> [0m(I) 
15:36:55.849 -> [0m(I) 
15:36:55.849 -> [0m(I) --UI configuration--
15:36:55.891 -> [0m(I) Language:             en
15:36:55.891 -> [0m(I) --UI configuration--
15:36:55.891 -> [0m(I) Enabled:              Yes
15:36:55.891 -> [0m(I) Hostname:                         
15:36:55.891 -> [0m(I) Client ID:            ffffffff-ffff-ffff-ffff-ffffffffffff
15:36:55.891 -> [0m(I) Interval:             255
15:36:55.891 -> [0m(I) -----------------------------------------------
15:36:55.891 -> [0m(I) Connecting to WiFi network:       
15:36:55.891 -> [0m(I) Marking /translations-en.json for download
15:36:55.964 -> [0m(I) (setupHanPort) Setting up HAN on pin 20/-1 with baud 0 and parity 0
15:36:56.043 -> [0m(D) Hardware serial
15:36:56.043 -> [0m
```

sonite, 2024-05-03:
2.2.28 works so far that I at last see the device in my home network and occupationally can configure it in its UI. Tomorrow I'll try 2.2.27 that works wit my ESP8266. E.g. this version is unstable for my XIAO ESP32C3.

ArnieO, 2024-05-04:
It seems we have an issue with later builds for C3.

sonite, 2024-05-05:
Tested 2.2.27
But it freezes after 15-39s. It seems to fetch the time and than just stop. I've removed the line ESP.deepSleep on row 352 in AmsToMqttBridge.cpp. I'm a bit unexperienced. Any suggestion what else I can do?
[image]

sonite, 2024-05-05:
Observation by visuals. Every time HAN turns red. it stops. Another note. The XIAO ESP32C3 pin number 20-21 is not selectable:
[image]
[image]

ArnieO, 2024-05-05:
Thank you for the update - but as our readers do not use C3, we are not able to put high priority on solving. the "C3 issue". 

Try older firmware versions (I am puzzled that 2.2.27 apparently do not work) and do an "Erase flash" before flashing - to be sure the device is clean.

sonite, 2024-05-05:
> Thank you for the update - but as our readers do not use C3, we are not able to put high priority on solving. the "C3 issue".
> 
> Try older firmware versions (I am puzzled that 2.2.27 apparently do not work) and do an "Erase flash" before flashing - to be sure the device is clean.

Thank you for replying and highlighting the importance of 'Erase Flash.' I've spent a couple of hours trying out the firmwares. While I can't definitively say if they work or not, here's what I've discovered:

[image]

The reason I initially thought versions 2.3.4, 2.2.27, and 2.2.28 didn't work was probably because I attempted to change the board configuration to 'seeed_xiao_esp32c3,' a habit I carried over from version 2.3.5. However, they do work the original setting board = esp32-c3-devkitm-1 .

I utilized Flash Download Tool 3.9.5 to establish the USB connection upon reset and erased with QIO settings set as SPI mode. Following that, I used 'Erase Flash' in Visual Platformio just to ensure a clean slate before uploading all these firmwares.

ArnieO, 2024-05-05:
Your observations are then the same as some others have seen: C3 builds for v2.2.x work, but not v2.3.x.

sonite, 2024-05-07:
Three other observations (2.3.4 up n running):

- The colour disappeared from the top border (purple)
- Serial parity error. ESP32C3 wont find the correct settings it self, whilst my ESP8266 v 2.2.27 did. I had to manually add baud rate 2400 and parity 8N1.
- The worst -> HAN stopped working after ~24, 36 hours. Only HAN. Reboot did not sort it. Now I've re-flashed and its operational again. Will see if the problem persist (I tried to restart my energy meter with no luck and double checked the physical connections)

But the other stuff works, MQTT sending to my Home assistant. ENTSO-E API ok.

gskjold, 2024-05-30:
Please try attached version:
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/15499121/esp32c3.zip)

avandorp, 2024-06-01:
> Your observations are then the same as some others have seen: C3 builds for v2.2.x work, but not v2.3.x.

2.3.4 works on my C3, OTA fails for 2.3.5 because of insufficient space, didn't try USB flashing yet because I would have to dismantle the installation.

gskjold, 2024-06-03:
Have you tried the firmware from my previous comment?

sonite, 2024-06-03:
> Have you tried the firmware from my previous comment?

Hi, Thanks fo all the good work. Since I have 2.3.5 up n running after so much issues. I have to see if I can free another C3 for this project. Don't want to break something that works. Maybe at the end of the week. BR S

closed 2024-06-06 by gskjold

dbeinder, 2024-06-07:
> EDIT: First file was wrong, try this one: [esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/15499711/esp32c3.zip)

@gskjold Your d0621e9 build works on C3 when flashed via USB/Serial, but web upgrade is still a problem, due to size limit:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381d6a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0xf4
load:0x403cc710,len:0x8a4
load:0x403ce710,len:0x2228
entry 0x403cc710
(I) AMS bridge started
(I) Voltage: 0.00V
WiFi.Mode(0) CUR=0
WiFi.Mode(1070184508) no change
(D) ESP32 LittleFS
(D)  size: 1441792
(I) Found firmware
(I)  flashing
(D)  firmware size: 1308848
(D)  available: 1306624
(I) -----------------------------------------------
```

gskjold, 2024-06-07:
I've made a version where remote debugging have been stripped out:
[esp32c3.zip](https://github.com/user-attachments/files/15742884/esp32c3.zip)

It seems to be the only way to make this work for now...

dbeinder, 2024-06-15:
For devs who want to keep hacking on their DIY devices with RemoteDebug before #805 is finished, here's a branch with modified partitions and working firmware upload: https://github.com/dbeinder/amsreader-firmware/tree/c3-big-app