UtilitechAS/amsreader-firmware #787: ISKRA Am550, P1 port encrypted set-up for  supported?

andrwie, 2024-05-03:
**Describe your problem**

I would like to use amsreader for for my Smartmeter
Want to check if my SmartMeter configuration is supported by amsreader.

The SmartMeter has the following P1 port description (information I got form my energy provider):
Protocol: DLMS
Security: Cosem Suite 0; High Level Security (HLS); AES 128

[image]

My first trials with the latest firmware failed (decryption key used, no authentication key available)

**Hardware information:**
 - Country: Austria
 - Meter: ISKRA Am550
 - Encryption enabled: yes
 - AMS reader: ESP8266
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: [e.g. 2.1.0]
 - MQTT: [yes/no]
 - MQTT payload type: [e.g. JSON]
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

Noschvie, 2024-05-03:
What's the name of your province / energy provider ?

andrwie, 2024-05-03:
the province is Carinthia (Kärnten)
and the energy provider is  KELAG

ArnieO, 2024-05-03:
You wrote "ISKRA Am550", but that is maybe the same as "Iskraemeco AM550"?
If so, we have adapted the firmware to run on Iskraemeco AM550 in Switzerland, see this thread: https://github.com/UtilitechAS/amsreader-firmware/issues/785

To understand why you have not succeeded, please run a Telnet debug as [described here](https://www.amsleser.no/module/ets_blog/blog?id_post=24) and post the result. 

Also, to have some more detailed information on you meter: Can you help me navigate the website https://www.kelag.at to find more technical information on the meter? (I can read quite a bit of German, so reading documentation is fine. Navigating the website to find documentation is a bit more tricky.)

andrwie, 2024-05-04:
Thanks for your suggestions.

I am quite sure, that the smartmeter you are referring to thread #785 is the same type. However the configuration of the P1 port can differ. e.g.: mine delivers encrypted date whereas the other one does not encrypt. I do not know it there is additional differences.

I checked the homepage of my provider - there is no specific data about the smartmeter P1 interface. But I can forward a document they sent me when I asked for the activation of the P1 interface:
[Technische Anleitung Kundenschnittstelle Smart Meter V3.2.pdf](https://github.com/UtilitechAS/amsreader-firmware/files/15210708/Technische.Anleitung.Kundenschnittstelle.Smart.Meter.V3.2.pdf)
Juat found one link to an operating manual for the ISKRAAM550: https://kaerntennetz.at//dokumente/Smart%20Meter/Zaehlerbeschreibung_Iskraemeco.pdf

I also die the telnet "exercise" you suggested - unfortunately just the same message - no matter what config of AMSreader I try:
"(W) No HAN data received last 30s, single blink"
Does this poit to a hardware problem? Cna I try something else?

ArnieO, 2024-05-05:
> I can forward a document they sent me when I asked for the activation of the P1 interface:
> [Technische Anleitung Kundenschnittstelle Smart Meter V3.2.pdf](https://github.com/UtilitechAS/amsreader-firmware/files/15210708/Technische.Anleitung.Kundenschnittstelle.Smart.Meter.V3.2.pdf)

Thank you, this is useful.

1. It confirms that the interface is indeed following the Dutch [P1 companion standard](https://www.netbeheernederland.nl/_upload/Files/Slimme_meter_15_a727fce1f1.pdf).
2. It describes which parameters are sent. (the table "Welche Daten werden ausgegeben")

So our hardware and firmware should be able to decrypt and decode this meter.

However, your Telnet debug shows that the reader does not get any data - so we need to find out why.

Let me run through a list of clarification questions:

1. Which hardware are you running? You wrote "ESP8266", so it cannot be our Pow-P1 (it is ESP32). Specifically, if this is a Wemos, you need to be careful regarding which input you use (search in other threads here). 
2. The data signal on this meter is an Open Collector output:
[image]
This means that you WILL SEE NO DATA if you just connect the data output to a GPIO: You need a pullup resistor to 3,3V, and it should be order-of-magnitude 3 kohm. 

If this is what you have on your hardware:
- Add "/vendor" to the URL and configure the device as "Generic ESP8266" - and select the GPIO you are using.
- Make sure the bitrate is set to 115 200 baud and parity set to 8N1.

If you still have issues (no data Telnet debug), please post screendump of your config page.

andrwie, 2024-05-06:
Thankx for the heads up. I will go on with debugging as you suggested.

The Board I am using is a Wemos D1 clone with an E12-F chip mounted.
The pin I have used to read P1 data is GPIO13
You mentioned I need to be careful about choosing the right GPIO for Wemos boards - is GPIO13 ok?

Next I will do a hardware check (DC test) on my interface circuitry - pulling RJ12 connector pin5 to gnd or let it float - to see if the signal arrives at GPIO13

ArnieO, 2024-05-06:
Thank you for information regarding your setup. 
GPIO13/D7 should be fine, it is only the Rx line that cannot be used. 

If you did not have a 3k pullup to 3,3V on the data line you cannot expect to see any data from the meter.

You will also need to invert the incoming signal and increase the buffer size from the default.
Try this setting:
[image]

andrwie, 2024-05-07:
I finally found the problem in my hardware set-up.
To send data, the smart meter needs to get the 5V supplied by the user on pin1, which I did not do (was just pulling the data pin and the request pin to my local 5V supply - this is not sufficient for my smart meter)

Now, I am getting data in the telnet terminal window.
However amsreader does not decrypt or convert it to power numbers - no data dispalyed in the main window
Decryption is activated and my key is inserted...
[image]

below is a Telnet dump - could you please have a look at it and advice on what to try next?
---------------------------------------
* Commands:
    ? or help -> display these help of commands
    q -> quit (close this connection)
    m -> display memory available
    v -> set debug level to verbose
    d -> set debug level to debug
    i -> set debug level to info
    w -> set debug level to warning
    e -> set debug level to errors
    s -> set debug silence on/off
    l -> show debug level
    t -> show time (millis)
    profiler:
      p      -> show time between actual and last message (in millis)
      p min  -> show only if time is this minimal
      P time -> set debug level to profiler
    c -> show colors
    filter:
          filter <string> -> show only debugs with this
          nofilter        -> disable the filter
    cpu80  -> ESP8266 CPU a 80MHz
    cpu160 -> ESP8266 CPU a 160MHz

* Please type the command and press enter to execute.(? or h for this help)
***
(V) HDLC frame:
(V) 7E A0 77 CF  02 23 13 BB  45 E6 E7 00  DB 08 49 53
(V) 4B 69 74 AC  00 8F 5F 20  00 03 D7 84  16 5F 78 9E
(V) 00 9F 85 8B  18 0E 95 3A  40 42 33 85  0F C5 97 31
(V) 99 A0 48 3A  05 91 1C 11  9D ED 0F D1  7D 9A 3B CF
(V) 06 2C C3 B1  2F 70 85 7A  D3 20 CE 90  69 BE 88 50
(V) 52 7B 15 90  95 0D B2 90  0F 1B AA 67  81 D4 C5 20
(V) F0 76 FA 57  2C C3 24 1E  A6 FD 8F E7  20 9B 85 B2
(V) 8B 90 E3 87  80 1D 4A 08  7E
(V) LLC frame:
(V) E6 E7 00 DB  08 49 53 4B  69 74 AC 00  8F 5F 20 00
(V) 03 D7 84 16  5F 78 9E 00  9F 85 8B 18  0E 95 3A 40
(V) 42 33 85 0F  C5 97 31 99  A0 48 3A 05  91 1C 11 9D
(V) ED 0F D1 7D  9A 3B CF 06  2C C3 B1 2F  70 85 7A D3
(V) 20 CE 90 69  BE 88 50 52  7B 15 90 95  0D B2 90 0F
(V) 1B AA 67 81  D4 C5 20 F0  76 FA 57 2C  C3 24 1E A6
(V) FD 8F E7 20  9B 85 B2 8B  90 E3 87 80  1D
(V) GCM frame:
(V) DB 08 49 53  4B 69 74 AC  00 8F 5F 20  00 03 D7 84
(V) 0F 00 03 D7  85 0C 07 E8  05 07 02 15  0B 11 00 FF
(V) 88 80 02 0C  09 06 00 06  19 09 00 FF  09 0D 31 31
(V) 32 31 33 37  38 33 38 31  31 39 39 09  04 15 0B 11
(V) 00 09 05 07  E8 05 07 02  06 00 DB 45  00 06 00 06
(V) 6B E5 06 00  16 36 F8 06  00 23 00 B0  06 00 00 00
(V) 09 06 00 00  00 00 09 00  09 00
(V) DLMS frame:
(V) 0F 00 03 D7  85 0C 07 E8  05 07 02 15  0B 11 00 FF
(V) 88 80 02 0C  09 06 00 06  19 09 00 FF  09 0D 31 31
(V) 32 31 33 37  38 33 38 31  31 39 39 09  04 15 0B 11
(V) 00 09 05 07  E8 05 07 02  06 00 DB 45  00 06 00 06
(V) 6B E5 06 00  16 36 F8 06  00 23 00 B0  06 00 00 00
(V) 09 06 00 00  00 00 09 00  09 00
(D) Received valid DLMS at 46
(V) Using application data:
(V) 02 0C 09 06  00 06 19 09  00 FF 09 0D  31 31 32 31
[... 222 lines cut ...]
(V) 67 64 32 CC  85 70 E8 C6  C2 13 E9 67  D2
(V) GCM frame:
(V) DB 08 49 53  4B 69 74 AC  00 8F 5F 20  00 03 D7 8A
(V) 0F 00 03 D7  8B 0C 07 E8  05 07 02 15  0B 17 00 FF
(V) 88 80 02 0C  09 06 00 06  19 09 00 FF  09 0D 31 31
(V) 32 31 33 37  38 33 38 31  31 39 39 09  04 15 0B 17
(V) 00 09 05 07  E8 05 07 02  06 00 DB 45  00 06 00 06
(V) 6B E5 06 00  16 36 F8 06  00 23 00 B0  06 00 00 00
(V) 00 06 00 00  00 33 09 00  09 00
(V) DLMS frame:
(V) 0F 00 03 D7  8B 0C 07 E8  05 07 02 15  0B 17 00 FF
(V) 88 80 02 0C  09 06 00 06  19 09 00 FF  09 0D 31 31
(V) 32 31 33 37  38 33 38 31  31 39 39 09  04 15 0B 17
(V) 00 09 05 07  E8 05 07 02  06 00 DB 45  00 06 00 06
(V) 6B E5 06 00  16 36 F8 06  00 23 00 B0  06 00 00 00
(V) 00 06 00 00  00 33 09 00  09 00
(D) Received valid DLMS at 46
(V) Using application data:
(V) 02 0C 09 06  00 06 19 09  00 FF 09 0D  31 31 32 31
(V) 33 37 38 33  38 31 31 39  39 09 04 15  0B 17 00 09
(V) 05 07 E8 05  07 02 06 00  DB 45 00 06  00 06 6B E5
(V) 06 00 16 36  F8 06 00 23  00 B0 06 00  00 00 00 06
(V) 00 00 00 33  09 00 09 00
(V) DLMS
(W) No HAN data received last 30s, single blink

ArnieO, 2024-05-07:
Great - and this seems to be valid data! 

You can translate it to something readable by doing this:

- Copy out the six lines of hexadecimal numbers below "DLMS frame:"
- Go to this page: https://www.gurux.fi/GuruxDLMSTranslator
- Paste the hex values in the upper frame
- Delete all the "(V)" in the start of the lines
- Press the "To XML" button

This gives:
[image]

This seems to fit with the payload format in the document from your provider:
[image]

- The third line is the time ("Uhrzeit"), and OBIS code 0.9.1 shall be in the format hh:mm:ss
The four bytes are 15 / 0B / 16 / 00, which i assume is 21:11:22 , and I don't know what the last byte is.
The fourth line is the date, OBIS code 0.9.2 shall be in the format YY.MM.DD or DD.MM.YY. 
The bytes are 07E8 / 05 / 07 / 02, which gives: 2024.05.07, and I don't know what the last byte is.

And so on. 

I think the problem here is that we have not encountered this payload format before, so it has to be implemented in the code. So you'll have to wait until @gskjold finds time to do that. He's on vacation right now so please have some patience.

andrwie, 2024-05-08:
that's really good news - thankx.

I would appreciate if you would include my smart meter's payload config in your code.

gskjold, 2024-05-23:
Please try the attached firmware:
[esp32s3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/15421526/esp32s3.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/15421527/esp32.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/15421529/esp32s2.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/15421530/esp8266.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/15421531/esp32c3.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/15421533/esp32solo.zip)

andrwie, 2024-05-24:
thankx for the update - I tried it. But it still doe not show the decoded Values in the main window
[image]

ArnieO, 2024-05-25:
> any other test I shall try?

No thanks - just wait until @gskjold finds time to make a new attempt. 😉 

Decoding the data is not the issue here, so Telnet output is no longer needed. That part was already confirmed earlier.

gskjold, 2024-05-30:
Here is another test version:
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/15499113/esp32s2.zip)

andrwie, 2024-05-30:
Thank you so much for your time and effort. I tried the new version. Unfortunately it is not showing data in the main screen.
[image]

Anything I can try?

ArnieO, 2024-05-31:
> Anything I can try?

Are you still seeing data as before when you do a Telnet debug?

gskjold, 2024-06-01:
I have a hard time testing this locally. Right now I can't see what would prevent it from parsing your data. If you feel comfortable sharing your decryption key, you can send that to gunnar@amsleser.no , maybe I have more luck putting your full payload in my emulator

andrwie, 2024-06-01:
@ ArnieO: I do see data in telnet terminal - seems to be tha same format us before - nothing on the main window though.

@ gskjold: Just sent my decryption key to gunnar@amsleser.no - hope this helps

gskjold, 2024-06-04:
Thank you @andrwie , just what I needed!
Here is a new test build:
[esp32s3.zip](https://github.com/user-attachments/files/15568969/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/15568970/esp32solo.zip)
[esp32.zip](https://github.com/user-attachments/files/15568971/esp32.zip)
[esp32c3.zip](https://github.com/user-attachments/files/15568972/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/15568973/esp32s2.zip)
[esp8266.zip](https://github.com/user-attachments/files/15568974/esp8266.zip)

andrwie, 2024-06-05:
I tried the new version - IT IS WORKING - thankx
[image]

gskjold, 2024-06-06:
Good to hear!