UtilitechAS/amsreader-firmware #789: HAN Port problems - HAN: Unknown data received, check meter config

Kimsejer, 2024-05-05:
After updating to v2.3.5 im getting errors on my HAN, so no data is loading.

HAN: Unknown data received, check meter config
HAN: Serial buffer full

Is has work for years before, so is pretty strang that it dosent now.

A clear and concise description of what the problem is.

**Hardware information:**
 - Country: Denmark
 - Meter: Landis+Gyr E360
 - Encryption enabled: No
 - AMS reader: Pow-P1
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: v.2.3.5
 - MQTT: yes
 - MQTT payload type: Home Assistant
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]
 
[image]

Kimsejer, 2024-05-05:
Fixed by editing the buffer size from 128 to 768.

closed 2024-05-05 by Kimsejer

ArnieO, 2024-05-05:
Yeah... the default buffer size is too small for most P1 meters. We have increased the default value in the latest firmware - but your device inherited the setting from before the upgrade.