UtilitechAS/amsreader-firmware #790: Energy use last 24 hours randomly missing hour data - added to next hour

Buruglen, 2024-05-06:
**Describe the bug**
The table showing energy use last 24 hours is randomly missing an hour of data. The data is then added to the next hour.
It seems that the system has not recognized that it is a new hour.

The data in Norwegian Elhub is correct, so the meter has the raw data.

**To Reproduce**
This happens randomly. Only seen with firmware 2.3.5.

**Expected behavior**
That all hour data is correct.

**Screenshots**
[image]
[image]
[image]

**Hardware information:**
 - Country: Norway
 - Meter: Aidon 6540
 - Encryption enabled no
 - AMS reader: Pow-U+

**Relevant firmware information:**
 - Version: 2.3.5
 - MQTT: Yes
 - MQTT payload type: Home Assistant
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: 2400 8E1
 - Temperature sensors 
 - ENTSO-E API enabled: Yes

**Additional context**
This often makes the Tariff peaks wrong as the data for two hours are added.

ArnieO, 2024-05-06:
Thank you for reporting, but this is a known and already reported issue that we're working on getting to the bottom of, see https://github.com/UtilitechAS/amsreader-firmware/issues/743

closed 2024-05-06 by ArnieO

Buruglen, 2024-05-06:
OK. Did not look so far back, as I had never seen it with earlier releases, only 2.3.5.

ArnieO, 2024-05-06:
> OK. Did not look so far back, as I had never seen it with earlier releases, only 2.3.5.

And it seems to occur very randomly - with (for most of those that see it) long time between each appearance. Which makes it difficult to debug. 
We think it started with v2.3.x.

Please contribute with the information @gskjold requests in the other thread!

Buruglen, 2024-05-06:
Just after installation of 2.3.5 I had it every day for a while. My Pow-U+ is powered all the time via USB. Please let me know if you want me to debug this in any way.

ArnieO, 2024-05-06:
Thank you for that, but please ask in the other thread on this subject, where @gskjold collects information to help figure this out: https://github.com/UtilitechAS/amsreader-firmware/issues/743