UtilitechAS/amsreader-firmware #792: Day and month plot numbers are gone.

ycagwyn, 2024-05-08:
All of a sudden the day and month plot numbers were replaced by some very huge numbers. I have tried to patch the figures into the .cfg file and after some struggle managed to upload them to the device.  Graphs looks OK until there is a change of hour when it gets back to show one very large number for previous hour.  Downloaded the configuration after upload and noticed that  the latter half (export) of day and month plot series were missing.
Instant power readings are still transmitted OK to the MQTT-server every 10 second while energy measurements looks as bad as on the GUI.

[image]

 - Country: Sweden
 - Meter: Kamstrup
 - Encryption enabled: no
 - AMS reader: Pow-K+
 - Chip: esp32s2 (160MHz)

 - Version: 2.3.5
 - MQTT: yes
 - MQTT payload type: raw(full)
 - HAN GPIO: ?
 - HAN baud and parity: 2400 8E1
 - ENTSO-E API enabled: not yet

ArnieO, 2024-05-08:
Huh, interesting. 
Do you see any sign in your data that this could be a number format / rollover error? 
What order of magnitude are the "normal" numbers?

ycagwyn, 2024-05-09:
Hi Arnie!

Thank you for the quick reaction.
Since I’m not that closely familiar with the product I cannot tell whether it’s format /rollover problem or not but I would guess so.
My normal daily import/export readings are 0-60 kWh and hourly 0-15 kWh.
This is how it looks today:
￼

Daily figures seemed OK after my path yesterday and here is what the downloaded .cfg says now:

amsconfig
version v2.3.5
boardType 5
meterBaud 2400
meterParity 8N1
meterInvert 0
meterDistributionSystem 2
meterMainFuse 25
meterProductionCapacity 10
dayplot 6 1715238018 320277.240 3 20000 18000 16000 15000 10000 12000 10000 0 0 0 1000 1000 1000 2000 1000 2000 1000 6000 6000 18000 79000 24000 6000 14000 30917.920 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 30915000 0 0 0 0 0 0
monthplot 7 1715205617 320151.360 3 12000 18000 17000 19000 33000 23000 28000 347000 18000 55000 35000 14000 26000 60000 22000 23000 33000 31000 57000 64000 38000 33000 28000 49000 41000 46000 28000 26000 15000 14000 0 30915.420 36000 29000 38000 36000 1000 27000 38000 30915000 27000 6000 1000 16000 11000 3000 18000 5000 15000 12000 0 3000 2000 15000 33000 30000 0 1000 3000 19000 22000 30000 0
energyaccounting 6 5 68.56 287.41 557.62 -26.03 -16.09 189.48 8 79.00 9 20.00 7 46.96 4 5.51 5 9.97 1138.79 325.80

Ha de´
Kenneth

ArnieO, 2024-05-09:
I see nothing special about those values, just normal household consumptions.

So where is the problem...

You marked off the total accumulated energy meter reading yesterday: 
[image]
Is this the wrong value? 
This should correspond to the value you see on the LCD screen of your meter.
What do you see on the meter LCD? A photo would be great.

ycagwyn, 2024-05-09:
￼

ycagwyn, 2024-05-09:
[image]

ycagwyn, 2024-05-09:
[image]
Todays view

ArnieO, 2024-05-09:
Thank you, that is interesting. 
It seems to show 3 decimals..?
Mine does not show decimals on the LCD.

Is the accumulated export value also wrong in your GUI?
[image]

ArnieO, 2024-05-09:
Can you run a Telnet debug so that we can see what data the meter sends? 
https://www.amsleser.no/module/ets_blog/blog?id_post=24

ycagwyn, 2024-05-09:
According to the netowner Boo Energi and the PV readings from APSystems ECU-C I have exported around 3800 kWh since installation end of June 2023 so I guess that is what should be presented as accumulated export.
I'll try to run a Telnet debug as soon as I figured out how its done on Mac

ArnieO, 2024-05-09:
> I'll try to run a Telnet debug as soon as I figured out how its done on Mac

Good question! I'd greatly appreciate if you could send me a short description when you find out, so that I can add that information to the bog post. 

The kWh values in your GUI are off by a factor ten. Please open the config page and set multiplier to 0.1 to fix that. We think that could be a root cause for the issues you are seeing. 
[image]

Is this something that suddenly happened, or did it show correct values previously? 
Maybe your meter has received a firmware update. We unfortunately have no way of knowing the number of decimals the meter sends, and it is not consistent. So in some cases a correction must be done by using the multipliers.

ycagwyn, 2024-05-09:
Change of multipliers made no visible difference.
[image]
**.... until I used dot(.) instead of comma(,) as decimal separator in the GUI !!!!!**

Now accumulated import and export figures seem to be more relevant and I will try to patch the plot numbers into the cfg once again if that is OK with you?

Everything worked perfect until 2024-05-07 17:00 as can be seen at the Homey insights graph having figures 0-20 kWh suddenly changing to 30 MWh. The sudden decrease at end of the 8/5 is probably a result of my patching efforts.
[image]

ycagwyn, 2024-05-09:
**Change of multipliers had no effect until I used dot(.) as decimal separator instead of comma(,) which is default at the GUI!** Saw that decimals are used in Power saving dBm as well.
Now my accumulated import and export numbers appears to be OK. Do you (as I) think it could be an idea to patch the plot numbers back into the cfg file?

ArnieO, 2024-05-09:
It is unfortunately difficult to find a way to accept both decimal comma and point in the GUI, we are a bit stuck with the libraries we use. 

> Do you (as I) think it could be an idea to patch the plot numbers back into the cfg file?

If you mean editing the cfg file and load it back: Yes, sure. You can always keep a copy and revert if it does not do the trick. Editing the cfg file is a good way to restore in such a situation.

ycagwyn, 2024-05-09:
So the conclusion is that my problem probably was caused by a firmware upgrade of the power meter which is fixed by changing multipliers kWh from 1 to 0.1. Also found that decimal sign used in GUI does not follow browsers locale.
Thank you so much for your help.
(I'll try Telnet on Mac and supply instructions if successful.)

ArnieO, 2024-05-09:
> So the conclusion is that my problem probably was caused by a firmware upgrade of the power meter which is fixed by changing multipliers kWh from 1 to 0.1. 

Power meter firmware upgrade is my best hypothesis, if this suddenly appeared.

> Also found that decimal sign used in GUI does not follow browsers locale.

Unfortunately correct.

> Thank you so much for your help.

You're welcome!

> (I'll try Telnet on Mac and supply instructions if successful.)

That would be great! 👍 

(Please close thread if you consider issue solved.)

ycagwyn, 2024-05-09:
Sorry but got this after reset of "Enable debugging" and reboot. Retried typing 0.1 as kWh multiplier. What did I do wrong, do I have to wait for accumulated figures?
[image]

ycagwyn, 2024-05-09:
Yes wait was the answer, sorry about that.

ArnieO, 2024-05-09:
Fantastic - thank you! Will add it to the blogpost ASAP.