UtilitechAS/amsreader-firmware #797: Not using assigned fixed IP

fth1963, 2024-05-26:
**Describe the bug**
I have assigned amsreader to use a specific fixed (/24) IP address, but it uses another.

**To Reproduce**
I'm not sure how to reproduce.  I have assigned IP 192.168.1.98/24 to it but after searching I have found out it uses 192.168.1.122. Looking at the settings I can see that it still shows my original setting for a fixed IP 192.168.1.98/24. 

**Expected behavior**
I expect it to use the IP address I have assigned in the settings.

**Screenshots**
[image]

**Hardware information:**
 - Country: Denmark
 - Meter: AMS Reader v 2.3.1
 - Encryption enabled: yes
 - AMS reader: Pow K+
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.3.1
 - MQTT: yes
 - MQTT payload type: Don't know looking directly a web page
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]
 - Chip: esp32s2 (160MHz)

**Additional context**
Add any other context about the problem here.

fth1963, 2024-05-26:
After reboot it used the original fixed IP.

ArnieO, 2024-05-27:
Strange. 
Could there be an IP conflict on your gateway, with more than one device requesting to use the same static IP?

gskjold, 2024-06-06:
Upgrade your device, this was fixed in 2.3.4

closed 2024-06-06 by gskjold