UtilitechAS/amsreader-firmware #799: Energy use last 24 hours - values disappearing/disappeared

spacecosmos, 2024-05-27:
**Describe the bug**
The data and values for **Energy use last 24 hours** are disappearing (and some already disappeared). Not sure what's going on.

**Expected behavior**
Having the data _not_ disappear.

**Screenshots**
[image]

**Hardware information:**
 - Country: Norway
 - Meter: Kamstrup
- Chip: esp32s2 (160MHz)
- Device: [Pow-K+]

**Additional context**
It's all been running fine until I saw this. Haven't updated or done anything either, except from a couple of reboots to see if it sorted itself out.

spacecosmos, 2024-05-28:
Still a problem after multiple reboots:

[image]

ArnieO, 2024-05-28:
What about the rest of your page?
Does it report power while this goes on - or has the Kamstrup Omnipower stopped sending data?

ArnieO, 2024-05-28:
Could this be what is going on? https://www.amsleser.no/module/ets_blog/blog?id_post=33

spacecosmos, 2024-05-31:
Sorry for the late reply. I tried to take out the device and left it disconnected for an hour hoping that would solve it - it didn't. The values display for a while just after rebooting and disappearing again. I just left everything as is an suddenly yesterday evening it started displaying correct values again. I haven't done anything and nothings really changed. So I'm a little confused.

[image]

ArnieO, 2024-05-31:
Yes, this is confusing. 
As mentioned in the article I linked, there is a firmware update available from Kamstrup for the Kamstrup Omnipower. We emailed all grid companies in Norway in March, recommending them to do the update - but did not get much response. And we know (based on reported issues from customers that we helped them solve) that some grid companies have not done the upgrade. 

So we can speculate and cross fingers - and hope that your grid company has done the upgrade now and that this is the cause and cure for what you have seen. 🤞

closed 2024-05-31 by spacecosmos