UtilitechAS/amsreader-firmware #800: Realtime export calculations is not right.

Pocke01, 2024-05-28:
**Describe the bug**
Started to notice after upgrade to 2.3.5 (upgraded from 2.2.28 to 2.3.5) that the calculations didn't seem right. 

This is the numbers from my inverter and electric company 
Electric company is 2 days behind in there numbers.
Current month imported: 835 kWh
Current month exported: 1279 kWh
Inverter Production today: 18 kWh. 

This is from AMS reader:
Realtime Export calculations in AMS ui
Day 95.40 kWh 14.5 SEK
Month 128795 kWh 467 SEK.   (looks like it should be 1287.95 kWh)

AMS Sensors in home assistant say: 
Current day produced 149,31 kWh
Current day used 41,26 kWh
Current month produced 128 849,16 kWh
Current month used 1 219 141,25 kWh

**To Reproduce**

**Expected behavior**

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: Sweden
 - Meter: Kamstrup
 - Encryption enabled: no
 - AMS reader: Pow-K+
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.3.5
 - MQTT: yes
 - MQTT Home Assistant
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: 2400 8N1
 - ENTSO-E API enabled: Yes

**Additional context**
last month total export is the same as my electric company so that one looks like its working.

ArnieO, 2024-05-28:
Thank you for the report, there seems to be a multiplier issue.

We have heard that some Swedish grid companies recently did a firmware upgrade on Kamstrup Omnipower meters that changed the resolution of the incoming data. And as there is nothing in the data that enables us to know the number of decimals, you need to correct this manually. From your data, it looks like they did that firmware upgrade on your meter this month. 

Go to the config page, and in the "Meter" tile activate Multipliers:
[image]

Then you need to change at least the "kWh" to 0.01 (to divide the incoming value by 100). This will scale both for Import and for Export. 

Your displayed value for "Current month used" will be wrong until we change month.

Pocke01, 2024-05-28:
yeah it looks like that might be the problem.

Thanks for the help!

ArnieO, 2024-05-30:
Hi @Pocke01 : 
You reopened the thread this morning. Do you still have issues?

Pocke01, 2024-05-30:
Hi, 
I still have problems but I must have re opened by misstake cause I don’t have all the data I think I need to find what’s is the problem. 

It looks like I have to change multiplier to 0.1 to get right on hour/day numbers. But the import / export month numbers seems to be off with different multipliers.

Need to get todays data from my grid company. But they are 2 days behind in the data I can see from them. 

[image]

ArnieO, 2024-05-30:
You can use the display on your power meter to verify the multipliers. 

1. Check the kWh counter on the display of your meter, it should be the same value as you see under the upper left gauge (13556). If not, adjust kWh-multiplier.
2. Press the green button on our Kamstrup meter 3 times so that it displays power (kW). Check that the gauge (upper left) shows the same power as what you see on the display of the meter. If not, adjust Watt-multiplier.

gskjold, 2024-05-30:
Please keep in mind that after adjusting a multiplier, historic data will still be wrong. Only new data will follow the new multiplier, and it might take a couple of hours for the day graph to correct itself, and a day or two for the month graph. Real time calculations will continue to be wrong until we enter the next month.

Pocke01, 2024-05-30:
Checked meter and changed multiplier to 0.1.  I´ll close this since its not a bug.  
Will check during next month to se if it looks normal again.  

Lots of thanks for the help!