UtilitechAS/amsreader-firmware #806: Addressable RGB LED of ESP32-S3-DevKitC-1 usable ?

alexwala, 2024-06-12:
**Describe your problem**
Is the addressable RGB LED of [ESP32-S3-DevKitC-1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html) usable ?

**Hardware information:**
 - AMS reader: ESP32-S3-DevKitC-1

**Relevant firmware information:**
 - Version: 2.3.5

ArnieO, 2024-06-12:
Our firmware does not use, and cannot use, an addressable LED.

alexwala, 2024-06-12:
thanks for the fast respond !

ArnieO, 2024-06-12:
You're welcome!
The two reasons for using an ordinary RGB LED are:

1. The blue part of the RGB LED is actuated directly from the incoming data signal. In that way it will indicate incoming data independently of whether the microcontroller is running or not. So as soon as the board is powered, there is a visual indicator of incoming data.
2. Increased code size (library for addressable LED)

alexwala, 2024-06-12:
Thanks for the explanation !

alexwala, 2024-07-05:
Adapted FW to work with [T-Dongle S3](https://www.lilygo.cc/products/t-dongle-s3) on board addressable RGB LED (using [FastLED](https://github.com/FastLED/FastLED) ;)