UtilitechAS/amsreader-firmware #807: Unexpected Drops in Energy Production and Income Readings Throughout the Day

candycab, 2024-06-13:
Description:
I am experiencing unexpected drops in the energy production and income readings throughout the day using the AMS reader. 
There are sudden drops in the values before they increase again. This issue occurs intermittently and affects the accuracy of the data displayed in Home Assistant.

Steps to Reproduce:
Set up the AMS reader with the latest firmware version.
Configure MQTT settings as per the documentation.
Observe the energy production and income readings in Home Assistant throughout the day.

Expected Behavior:
The energy production and income readings should increase steadily throughout the day without sudden drops.

[image]
[image]

Hardware information:
 - Country: Norway
 - Meter: Kamstrup
 - AMS reader: ESP8266, Pow-K

**Relevant firmware information:**
 - Version: 2.3.5
 - MQTT: yes

wknutsen, 2024-06-14:
I think this is fixed in next version. and is related to bug #788

ArnieO, 2024-06-14:
v2.3.6 is right around the corner now! 
Expected later today if nothing unexpected shows up.

gskjold, 2024-06-15:
2.3.6 is out, but it looks to me like your device reboots a lot

closed 2024-06-29 by gskjold

candycab, 2024-06-30:
I'm still experiencing problems despite upgrading to version 2.3.6. 
@gskjold You mentioned that the device restarts frequently—how did you determine this?
[image]
[image]
[image]

ArnieO, 2024-07-01:
> I'm still experiencing problems despite upgrading to version 2.3.6.
@gskjold You mentioned that the device restarts frequently—how did you determine this?

Power meters with "HAN NVE" configuration sends accumulated energy (kWh) only once per hour (at whole hour). 
Such drops are usually an indication that the the device reboots in between those messages, as the firmware makes a running estimate/accumulation of kWh based on power readings. If device reboots, that estimate is restarted. When passing whole hour, kWh is reset to the value reported by the meter - and the curve jumps back up.

However, it does not look like your device is rebooting often; "Up 4 days":
[image]

We are starting to see some unexpected issues with the old ESP8266 devices, see for instance https://github.com/UtilitechAS/amsreader-firmware/issues/815
If you too have made a "long" upgrade I would try a Factory reset as I proposed in that thread. 
I would also try to disable MQTT as a test - and observe if there are drops in the Consumption figures in the "Real time calculation" tile.

We currently do not have an "upgrade to ESP32" system - but that might be a good idea. When did you buy your Pow-K?

ArnieO, 2024-07-01:
You may want to try the test firmware posted here: https://github.com/UtilitechAS/amsreader-firmware/issues/815#issuecomment-2199477693