UtilitechAS/amsreader-firmware #810: Price current hour sent to MQTT randomly during one hour

Cartoon-Dude, 2024-06-17:
I noticed the current hour price is sent only during an hour randomly every few days to homeassistant MQTT.
[image]
Everything related to price is marked in homeassistant as "unavailable" most of the time.
[image]
Unfortunately only this value can be use to calculate the cost with the homeassistant energy stuff.

v2.3.6

ArnieO, 2024-06-17:
Are you reading data directly from Entso-E (using API key), or are you reading from our intermediary server (no API key)?

Experience shows that reading from Entso-E is sometimes not so reliable, which is why we set up the intermediary server (with an API that is far easier for the device).

ArnieO, 2024-06-17:
Can you show your configuration for reading prices?
[image]

Cartoon-Dude, 2024-06-17:
[image]

gskjold, 2024-06-17:
Do you have a fixed energy price? There could be some issue if fetch is not enabled and you rely on a fully fixed price

ArnieO, 2024-06-17:
> Do you have a fixed energy price? There could be some issue if fetch is not enabled and you rely on a fully fixed price

His screenclip above confirms that he has fixed prices, which is why he has not activated "Enable price fetch" (and should not need to). Maybe a snag related to that then?

ArnieO, 2024-06-21:
Seems to be a bug. We'll look into it.
Thank you for reporting!

NiteCrwlr, 2024-06-23:
Hello. I can confirm this, same problem here.

gskjold, 2024-06-27:
@NiteCrwlr do you also have "Enable price fetch from server" disabled?

Also, check that your device is not rebooting randomly, just to rule out this as a cause

gskjold, 2024-06-29:
The attached build should fix this issue:

[esp32s3.zip](https://github.com/user-attachments/files/16039336/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/16039337/esp32solo.zip)
[esp32.zip](https://github.com/user-attachments/files/16039338/esp32.zip)
[esp32c3.zip](https://github.com/user-attachments/files/16039339/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/16039340/esp32s2.zip)
[esp8266.zip](https://github.com/user-attachments/files/16039341/esp8266.zip)

NiteCrwlr, 2024-06-29:
> @NiteCrwlr do you also have "Enable price fetch from server" disabled?
> 
> Also, check that your device is not rebooting randomly, just to rule out this as a cause

Yes, "Enable price fetch from server" is disabled.
Device does not reboot, it has a stable uptime.
I will try the fixed firmware, thank you for your effort!

NiteCrwlr, 2024-06-30:
Looks good! Price sensors all available since the update yesterday evening! 👍🏼

Cartoon-Dude, 2024-07-04:
It is now perfect, and the switch to day/night tariff is working.

closed 2024-07-04 by Cartoon-Dude