UtilitechAS/amsreader-firmware #811: AMSreader reports 10x actual power consumption

adsa95, 2024-06-17:
**Describe the bug**
AMSreader seems to assume power meter readings are reported with two decimals for meter kWh readings even when they are reported with three decimals. This leads to all values based on this being reported as 10x the actual value.

About  a month ago my reader started reporting 10x power consumption on all values based on power meter kWh values (hourly, daily etc, but the current live consumption still seems correct). Before reporting this here I contacted my power company as I thought maybe they updated the meter software which could be relevant to know in this error report. Apparently they haven't updated the software but made a configuration change to report the meter value with three decimals instead of two, due to a regulation change in Sweden.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Values to be reported correctly.

**Screenshots**
Here is my daily power consumption during the transition (exported from Home Assistant):
[image]

**Hardware information:**
 - Country: Sweden
 - Meter: Kempower
 - Encryption enabled: No
 - AMS reader: Pow-K+

**Relevant firmware information:**
 - Version: 2.3.5
 - MQTT: Yes
 - MQTT payload type: Home Assistant

**Additional context**
I guess this could be an issue on the power meter side instead of the reader side if the values are exchanged as floats and the decimal is in the wrong place. If so let me know how/what to debug so that I can confirm or debunk this :)

ArnieO, 2024-06-17:
Thank you for explaining the reason for this; we assumed it was linked to a firmware update on the meters - but I now understand it is a regulatory issue. 

I made a blogpost, making it easier for others to find this information: https://www.amsleser.no/module/ets_blog/blog?id_post=35

closed 2024-06-17 by ArnieO

ruffy91, 2024-07-04:
I have the same problem in Switzerland. However after setting the multiplicator for kWh the value for total accumulated power is correct but it is still 10 times too high for individual phases. Is there a separate setting for the phases I didn't find or is this a bug?

ArnieO, 2024-07-05:
> I have the same problem in Switzerland. However after setting the multiplicator for kWh the value for total accumulated power is correct but it is still 10 times too high for individual phases. Is there a separate setting for the phases I didn't find or is this a bug?

Interesting to hear that this issue shows up in other countries too - so thank you for notifying us!

This sounds like a bug. 
In most countries the payload does not deliver power per phase, so my guess is that the multiplicator has not been implemented in the code for power per phase.

@gskjold will look into this.

ruffy91, 2024-09-04:
The code where the 10x error (https://www.amsleser.no/module/ets_blog/blog?id_post=35) comes from is this: 

`        if(meterType == AmsTypeKamstrup) {
            if(listType >= 3) {
                activeImportCounter *= 10;
                activeExportCounter *= 10;
                reactiveImportCounter *= 10;
                reactiveExportCounter *= 10;
                l1activeImportCounter *= 10;
                l2activeImportCounter *= 10;
                l3activeImportCounter *= 10;
                l1activeExportCounter *= 10;
                l2activeExportCounter *= 10;
                l3activeExportCounter *= 10;
            }`

Maybe there is some method to detect swedish and swiss meters and automatically not applying this unnecessary multiplication instead of manually scaling it down again afterwards?

ArnieO, 2024-09-05:
> Maybe there is some method to detect swedish and swiss meters and automatically not applying this unnecessary multiplication instead of manually scaling it down again afterwards?

We do not see any way to detect this in a foolproof way. And it is not consistently different from country to country, as such changes are typically rolled out over time. 

There are OBIS code fields available in the standard that could have been used in the payload to indicate the number of decimals. Unfortunately, such codes are not used.