UtilitechAS/amsreader-firmware #812: Editing hourly data doesn't change tariff peaks

SVH-Powel, 2024-06-21:
**Describe the bug**
The title says it all. When editing hourly data because it has missed one hour (one hour is zero and the next one is twice as much), the tariff peaks is not changed and is from now on not related to actual data at all.

Version is 2.3.6

ArnieO, 2024-06-21:
I tried to reproduce this, by editing one hour to 8000:
[image]

Can you retry, and document step by step what you did?

SVH-Powel, 2024-06-21:
I don't have access to the amsreader at the moment. I am able to edit the data as you show, but after setting the value to 8kW, what is your monthly tariff peak?

ArnieO, 2024-06-21:
Ah - right. Sorry, I misunderstood. 
This is an enhancement point.

It is possible to fix the incorrect peak by editing the cfg file. I will make a "tips & tricks" article on how to do it later today.

ArnieO, 2024-06-21:
Until we get an "Edit data" also on "Tariff peaks": This article is now edited to indicate how to edit Tarrif peaks by using the config file: https://www.amsleser.no/module/ets_blog/blog?id_post=28

closed 2024-06-21 by ArnieO