UtilitechAS/amsreader-firmware #813: Unkown data received

Th3H0ff, 2024-06-26:
Before I start asking my utility I would really need some input from this forum.
I'm using a Wemos D1 mini with a [TTL to Mbus converter](https://www.ebay.com/itm/273122508071?_trkparms=amclksrc%3DITM%26aid%3D1110006%26algo%3DHOMESPLICE.SIM%26ao%3D1%26asc%3D20201210111314%26meid%3D4ae500c42d6c43dba4fbc3c8ed06db95%26pid%3D101195%26rk%3D1%26rkt%3D12%26sd%3D274011702607%26itm%3D273122508071%26pmt%3D1%26noa%3D0%26pg%3D2047675%26algv%3DSimplAMLv9PairwiseWebMskuAspectsV202110NoVariantSeed%26brand%3DUnbranded&_trksid=p2047675.c101195.m1851&amdata=cksum%3A2731225080714ae500c42d6c43dba4fbc3c8ed06db95%7Cenc%3AAQAGAAABEIIp3bTCXFAKw6S9crk9CpCrrtOgadqFiC10NVPbN0936eprYVB4SXfDa52e1mbPLFwdBXChmaGoQQy%252BBr%252F%252B6GwMcxzo7LNbOSruw8JYl4DlyP7oF0HAfDyaq70zO7kJ5TaNT4YP7AkcEgketK%252BSXiNU63J9O%252FNGJW%252FFzn%252B66oRh%252Ff51%252FOd%252BA7ejgOOSfiKyvRVua%252FiFnXrxVBuK6ejVzVdYQ8RkXWB%252BnAu4penvUGurSKh3gR%252B7GDYFpsU4WtTtZ59xj8h1jM8sf1rmoti2QMtmA6IzzxoTGR%252FY%252BwYIH5Ouq6dr2FeuKZP7nES3I5AT%252FKa7fdaUOT27KtkO6Rie2slNPzgZrUAyecfyWTwfQBCc%7Campid%3APL_CLK%7Cclp%3A2047675) which is connected to my Kamstrups power meter using the RJ45 type HAN module.

The amsreader telnet debug keep showing
(E) Ended up in default case while unwrapping...(tag FF)
(W) Unknown data received
(V)   payload:
(V) FF F7 FF FF  FF F7 BF FF  FF FF FF FF  7F BF EF FF
(V) FF FF FF 7F  FF FF BF FF  FB FF BF FF  FB FF FF DF
(V) FF FF FF FF  FF FB FF FF  FF DF FF FF  FF FF DF FF
(V) FF FF FF FF  BF FF FF FF  F7 FF FF F7  FF FF FF FF
(V) F7 FF FF F7  EF FF 7F F7  FF FF F7 FF  F7 FF FF FF
(V) FF F7 BF FF  FF FF EF FF  BF FF EF FF  FF FF BF FB
(V) 7F

at regular intervals. My utility provider claims they have activated the Han port.

**Hardware information:**
 - Country: Åland
 - Meter: Kamstrups 
 - Encryption enabled no
 - AMS reader: ESP8266
 - M-bus adapter (if applicable): TTL to Mbus

**Relevant firmware information:**
 - Version: 2.3.6
 - MQTT: yes
 - MQTT payload type: Home Assistant
 - HAN GPIO: D! (GPIO4)
 - HAN baud and parity: 2400 8N1
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

ArnieO, 2024-06-26:
My best guess is that your M-bus conversion fails. 

Some advise:
1. There are numerous threads here where users (more or less successfully) have set up systems using same or similar hardware. Search in closed threads under Issues. The two common issues seem to be:
- Inadequate decoupling on the Wemos D1 mini (depends on which version you have)
- Interfacing issues with the M-bus converter.
3. The Kamstrup HAN-NVE module (the one you use) is especially difficult to read (compared to meters with built-in HAN-NVE interface) due to its very limited allowed current draw (max 144 mW) from the M-bus signal. It is far easier to read data from the 6-pin plug inside the Kamstrup meter, where data arrives as a 0/3,3V signal. By taking the signal there you avoid the translation to M-bus levels (24/12V; the HAN-NVE module) and back to logic level (your M-bus adapter).
Pinout for the 6-pin plug: https://github.com/ArnieO/SmartMeterDocumentation#han-interface-description

Th3H0ff, 2024-06-26:
First of all. Thank you for the prompt and accurate answer. It was, as they say on LinkedIn, above and beyond.
Furthermore. I CANNOT believe this! You are 100% correct. I unlugged the HAN module from the meter and plugged the D1 mini into the data out pin along with ground and...it worked. HAN wen green in the gui right away. Man I spent 50€ on the HAN module, surely 50€ on different mbus-ttl chips and God knows how many hours digging through different threads and I was in the wrong place all along. :)
Crazy.

ArnieO, 2024-06-26:
> First of all. Thank you for the prompt and accurate answer. It was, as they say on LinkedIn, above and beyond. Furthermore. I CANNOT believe this! You are 100% correct. I unplugged the HAN module from the meter and plugged the D1 mini into the data out pin along with ground and...it worked. HAN wen green in the gui right away. Man I spent 50€ on the HAN module, surely 50€ on different mbus-ttl chips and God knows how many hours digging through different threads and I was in the wrong place all along. :) Crazy.

I've been working on this for several years, so this is very familiar stuff - and spent countless hours to construct what is now the Pow-K and the Pow-U.

By the way, you can draw up to 75 mA from the Vcc pin (which is at nominal 4,15V) of the 6-pin connector in the meter. So if you want a solution without external power it is possible to power from the meter - BUT you then need some serious capacitance to support the ESP while it's on Wi-Fi, or else it will reboot regularly. If you overload the Vcc pin too much, it will shut down the port for a few minutes.

Best of luck, and enjoy the firmware!

closed 2024-06-26 by ArnieO