UtilitechAS/amsreader-firmware #814: Pow-K+/Omnimeter/N1: Voltage values are most likely multiplied by 100

jlaur, 2024-06-27:
**Describe the bug**
The voltages for all phases were suddenly multiplied by 100 (it seems). As seen from an event log getting MQTT messages:
```
2024-06-27 14:49:12.776 [INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'Omnipower_Phase3_Voltage' changed from 231 V to 230 V
2024-06-27 14:49:42.722 [INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'Omnipower_Phase1_Voltage' changed from 233 V to 234 V
2024-06-27 14:49:52.732 [INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'Omnipower_Phase2_Voltage' changed from 230 V to 229 V
2024-06-27 14:50:02.741 [INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'Omnipower_Phase1_Voltage' changed from 234 V to 23452 V
2024-06-27 14:50:02.759 [INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'Omnipower_Phase2_Voltage' changed from 229 V to 23006 V
2024-06-27 14:50:02.789 [INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'Omnipower_Phase3_Voltage' changed from 230 V to 23088 V
```

and has been like that ever since.

I updated the firmware earlier this week, but didn't mess with anything today when this happened. I have other power meters still measuring correct voltage, and things connected to my sockets is still working, so I know for a fact that these voltages are faulty. 🙂

Maybe my grid company pushed a firmware upgrade to my meter?

I can easily work-around this by dividing by 100 when receiving MQTT messages, but I would rather have it fixed closer to the source - and it also affects the amsleser UI.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to main page.
2. Look at voltages.
3. Observe insane values.

**Expected behavior**
I would expect voltages to be closer to 230 V than 23000 V.

**Screenshots**
[image]

**Hardware information:**
 - Country: Denmark
 - Meter: OMNIPOWER
 - Encryption enabled: yes
 - AMS reader: Pow-K+
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.3.6
 - MQTT: yes
 - MQTT payload type: raw minimal
 - HAN GPIO: ?
 - HAN baud and parity: 2400 8N1
 - Temperature sensors: ?
 - ENTSO-E API enabled: yes

**Additional context**
Grid company: N1 (Northern Jutland)

ArnieO, 2024-06-28:
Ouch... I agree with you, this is probably an other power meter firmware adjustment. 
There was one in Sweden too around 1st May: https://www.amsleser.no/module/ets_blog/blog?id_post=35

It could be great if you could run a Tenet debug and post the output as a text file for analysis/confirmation. https://www.amsleser.no/module/ets_blog/blog?id_post=24

The fix is easy (the principle is explained in the first link): 
- Activate Multipliers
- Set Voltage multiplier to 0.01 (effektively dividing by 100)

What is unfortunate is that there are available OBIS codes that could have been part of the payload - that tell where the decimal point should be. But we have not seen any payload yet using those codes. So there is no way for the firmware to detect such a change (except perhaps an automatic "sanity check", but that could be a slippery slope). 

The fix is easy to do

jlaur, 2024-06-28:
@ArnieO - thanks for your quick response and tip for enabling multiplier. I hope the debug information is safe to share.

<details>
<summary>
Telnet session
</summary>

```
(V) HDLC frame:
(V) 7E A0 E2 2B  21 13 23 9A  E6 E7 00 0F  00 00 00 00
(V) 0C 07 E8 06  1C 05 0E 1C  00 FF 80 00  00 02 19 0A
(V) 0E 4B 61 6D  73 74 72 75  70 5F 56 30  30 30 31 09
(V) 06 01 01 00  00 05 FF 0A  10 35 37 30  36 35 36 37
(V) 33 30 32 38  31 38 37 33  30 09 06 01  01 60 01 01
(V) FF 0A 12 36  38 34 31 34  33 39 42 4E  32 34 34 31
(V) 30 31 30 31  31 09 06 01  01 01 07 00  FF 06 00 00
(V) 02 1F 09 06  01 01 02 07  00 FF 06 00  00 00 00 09
(V) 06 01 01 03  07 00 FF 06  00 00 00 00  09 06 01 01
(V) 04 07 00 FF  06 00 00 01  D6 09 06 01  01 1F 07 00
(V) FF 06 00 00  00 8C 09 06  01 01 33 07  00 FF 06 00
(V) 00 00 24 09  06 01 01 47  07 00 FF 06  00 00 00 B9
(V) 09 06 01 01  20 07 00 FF  12 5C 11 09  06 01 01 34
(V) 07 00 FF 12  5B 1D 09 06  01 01 48 07  00 FF 12 5C
(V) 46 66 66 7E
(V) LLC frame:
(V) E6 E7 00 0F  00 00 00 00  0C 07 E8 06  1C 05 0E 1C
(V) 00 FF 80 00  00 02 19 0A  0E 4B 61 6D  73 74 72 75
(V) 70 5F 56 30  30 30 31 09  06 01 01 00  00 05 FF 0A
(V) 10 35 37 30  36 35 36 37  33 30 32 38  31 38 37 33
(V) 30 09 06 01  01 60 01 01  FF 0A 12 36  38 34 31 34
(V) 33 39 42 4E  32 34 34 31  30 31 30 31  31 09 06 01
(V) 01 01 07 00  FF 06 00 00  02 1F 09 06  01 01 02 07
(V) 00 FF 06 00  00 00 00 09  06 01 01 03  07 00 FF 06
(V) 00 00 00 00  09 06 01 01  04 07 00 FF  06 00 00 01
(V) D6 09 06 01  01 1F 07 00  FF 06 00 00  00 8C 09 06
(V) 01 01 33 07  00 FF 06 00  00 00 24 09  06 01 01 47
(V) 07 00 FF 06  00 00 00 B9  09 06 01 01  20 07 00 FF
(V) 12 5C 11 09  06 01 01 34  07 00 FF 12  5B 1D 09 06
(V) 01 01 48 07  00 FF 12 5C  46
(V) DLMS frame:
(V) 0F 00 00 00  00 0C 07 E8  06 1C 05 0E  1C 00 FF 80
(V) 00 00 02 19  0A 0E 4B 61  6D 73 74 72  75 70 5F 56
(V) 30 30 30 31  09 06 01 01  00 00 05 FF  0A 10 35 37
(V) 30 36 35 36  37 33 30 32  38 31 38 37  33 30 09 06
(V) 01 01 60 01  01 FF 0A 12  36 38 34 31  34 33 39 42
(V) 4E 32 34 34  31 30 31 30  31 31 09 06  01 01 01 07
(V) 00 FF 06 00  00 02 1F 09  06 01 01 02  07 00 FF 06
(V) 00 00 00 00  09 06 01 01  03 07 00 FF  06 00 00 00
(V) 00 09 06 01  01 04 07 00  FF 06 00 00  01 D6 09 06
(V) 01 01 1F 07  00 FF 06 00  00 00 8C 09  06 01 01 33
(V) 07 00 FF 06  00 00 00 24  09 06 01 01  47 07 00 FF
(V) 06 00 00 00  B9 09 06 01  01 20 07 00  FF 12 5C 11
(V) 09 06 01 01  34 07 00 FF  12 5B 1D 09  06 01 01 48
(V) 07 00 FF 12  5C 46
(D) Received valid DLMS at 29
(V) Using application data:
(V) 02 19 0A 0E  4B 61 6D 73  74 72 75 70  5F 56 30 30
(V) 30 31 09 06  01 01 00 00  05 FF 0A 10  35 37 30 36
(V) 35 36 37 33  30 32 38 31  38 37 33 30  09 06 01 01
(V) 60 01 01 FF  0A 12 36 38  34 31 34 33  39 42 4E 32
(V) 34 34 31 30  31 30 31 31  09 06 01 01  01 07 00 FF
(V) 06 00 00 02  1F 09 06 01  01 02 07 00  FF 06 00 00
(V) 00 00 09 06  01 01 03 07  00 FF 06 00  00 00 00 09
(V) 06 01 01 04  07 00 FF 06  00 00 01 D6  09 06 01 01
(V) 1F 07 00 FF  06 00 00 00  8C 09 06 01  01 33 07 00
(V) FF 06 00 00  00 24 09 06  01 01 47 07  00 FF 06 00
(V) 00 00 B9 09  06 01 01 20  07 00 FF 12  5C 11 09 06
(V) 01 01 34 07  00 FF 12 5B  1D 09 06 01  01 48 07 00
(V) FF 12 5C 46
(V) DLMS
[... 101 lines cut ...]
(V) 00 00 00 00  09 06 01 01  03 07 00 FF  06 00 00 00
(V) 00 09 06 01  01 04 07 00  FF 06 00 00  01 CE 09 06
(V) 01 01 1F 07  00 FF 06 00  00 00 8B 09  06 01 01 33
(V) 07 00 FF 06  00 00 00 24  09 06 01 01  47 07 00 FF
(V) 06 00 00 00  BB 09 06 01  01 20 07 00  FF 12 5C 29
(V) 09 06 01 01  34 07 00 FF  12 5B 2F 09  06 01 01 48
(V) 07 00 FF 12  5B D3
(D) Received valid DLMS at 29
(V) Using application data:
(V) 02 19 0A 0E  4B 61 6D 73  74 72 75 70  5F 56 30 30
(V) 30 31 09 06  01 01 00 00  05 FF 0A 10  35 37 30 36
(V) 35 36 37 33  30 32 38 31  38 37 33 30  09 06 01 01
(V) 60 01 01 FF  0A 12 36 38  34 31 34 33  39 42 4E 32
(V) 34 34 31 30  31 30 31 31  09 06 01 01  01 07 00 FF
(V) 06 00 00 02  27 09 06 01  01 02 07 00  FF 06 00 00
(V) 00 00 09 06  01 01 03 07  00 FF 06 00  00 00 00 09
(V) 06 01 01 04  07 00 FF 06  00 00 01 CE  09 06 01 01
(V) 1F 07 00 FF  06 00 00 00  8B 09 06 01  01 33 07 00
(V) FF 06 00 00  00 24 09 06  01 01 47 07  00 FF 06 00
(V) 00 00 BB 09  06 01 01 20  07 00 FF 12  5C 29 09 06
(V) 01 01 34 07  00 FF 12 5B  2F 09 06 01  01 48 07 00
(V) FF 12 5B D3
(V) DLMS
```
</details>

ArnieO, 2024-06-28:
Not sure if it is sensitive, but the payloads contains the serial number of your meter. 
Edit your message now if you don't want it to be in the "open".

 This is the content of one of your payloads, decoded to XML format.

(I have omitted the lines that identify the type number and serial number of your meter.)

```
<DataNotification>
  <LongInvokeIdAndPriority Value="00000000" />
  <!--2024-06-28 14:28:00-->
  <DateTime Value="07E8061C050E1C00FF800000" />
  <NotificationBody>
    <DataValue>
      <Structure Qty="19" >
        <String Value="Kamstrup_V0001" />

       (...)

        <!--1.1.1.7.0.255-->
        <OctetString Value="0101010700FF" />
        <UInt32 Value="0000021F" />
        <!--1.1.2.7.0.255-->
        <OctetString Value="0101020700FF" />
        <UInt32 Value="00000000" />
        <!--1.1.3.7.0.255-->
        <OctetString Value="0101030700FF" />
        <UInt32 Value="00000000" />
        <!--1.1.4.7.0.255-->
        <OctetString Value="0101040700FF" />
        <UInt32 Value="000001D6" />
        <!--1.1.31.7.0.255-->
        <OctetString Value="01011F0700FF" />
        <UInt32 Value="0000008C" />
        <!--1.1.51.7.0.255-->
        <OctetString Value="0101330700FF" />
        <UInt32 Value="00000024" />
        <!--1.1.71.7.0.255-->
        <OctetString Value="0101470700FF" />
        <UInt32 Value="000000B9" />
        <!--1.1.32.7.0.255-->
        <OctetString Value="0101200700FF" />
        <UInt16 Value="5C11" />
        <!--1.1.52.7.0.255-->
        <OctetString Value="0101340700FF" />
        <UInt16 Value="5B1D" />
        <!--1.1.72.7.0.255-->
        <OctetString Value="0101480700FF" />
        <UInt16 Value="5C46" />
      </Structure>
    </DataValue>
  </NotificationBody>
</DataNotification>
```
What you see here is the OBIS code for each parameter, then the value in hexadecimal format. 
The phase voltages are these OBIS codes:
- 32.7.0 : Value 5C11, which is 23569
- 52.7.0 : Value 5B1D, which is 23325
- 72.7.0 : Value 5C46, which is 23622

In comparison, my Kamstrup Omnipower here in Norway sends these voltages:
```
<!--1.1.32.7.0.255-->
<OctetString Value="0101200700FF" />
<UInt16 Value="00F3" />
<!--1.1.52.7.0.255-->
<OctetString Value="0101340700FF" />
<UInt16 Value="00F3" />
<!--1.1.72.7.0.255-->
<OctetString Value="0101480700FF" />
<UInt16 Value="00F2" />
```

As you can see, the voltage values are here without decimals: 243, 243 and 242V.

**The conclusion is clear:**
Your meter is now sending phase voltage with two decimals. 
The fix you need to implement is described in my previous message.

It is indeed a strange change. Who needs 5-digit precision on phase voltage? It does not make sense to me.

closed 2024-06-28 by ArnieO