UtilitechAS/amsreader-firmware #815: POW-U constant rebooting on "newer" firmware

thorbearswe, 2024-06-29:
**Describe your problem**
I updated my POW-U yesterday. I believe from 2.2.3 to 2.3.6 and it seemed to go well and worked as it should for a couple of hours. 
but today i noticed in home assistant that i had no values delivered from the meter since yesterday evening. with external power everything works as it should, but when i disconnect the usb power the pow-u reboots and i get about 6-10 scs of uptime between reboots. if i connect external power the reboots stop and i get readings again. 
my guess is that something changed between the firmware versions that made the pow-u need more power.
i have tried to downgrade the firmware to see if it solves the problem but i have NOT had any success to back all the way to 2.2.3. 
this behaviour is observed with 2.3.4 and 2.3.5 as well.

**Hardware information:**
 - Country: Sweden
 - Meter: Kaifa MA304H4D
 - Encryption enabled: yes
 - AMS reader: POW-U esp8266 HWver 1.4
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.3.6
 - MQTT: yes
 - MQTT payload type: Home assistant
 - HAN GPIO: ?
 - HAN baud and parity: 2400 8E1
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: yes (fetch from remote server)

ArnieO, 2024-06-30:
This is a strange and unexpected issue, so let us try to debug.
You're running quite old hardware. One of the disadvantages with ESP8266 is that it has some flaws in the power saving.

Can you as a start provide a screenshot of your Config page?

thorbearswe, 2024-06-30:
yes, ofc.

[image]

ArnieO, 2024-07-01:
Thank you! I don't see any issues with this config.
We have sometimes seen issues with such "long" updates at the one you did.

I would in this situation try a Factory reset.
If you want to preserve historical values: Do a backup from the Info/status page first - and restore from the file after the Factory reset is done. 

Try this:
- Do a backup with all parameters stored (except "Include secrets").
- Do a Factory reset and restart the unit as per [Firmware User Manual](https://www.amsleser.no/index.php?controller=attachment&id_attachment=48) chapter 3.
- When device is running: Observe if it still reboots before restoring from backup.
- Restore parameters from backup
- If it now starts to reboot, please post the content of the backup file (configfile.cfg). (Delete potential sensitive information first)

gskjold, 2024-07-01:
I was working on a unrelated thing on friday which may affect stability for 8266. Could you please try the attached firmware:
[esp8266.zip](https://github.com/user-attachments/files/16039341/esp8266.zip)

Update guide: https://github.com/UtilitechAS/amsreader-firmware/wiki/Flashing-via-web-GUI

thorbearswe, 2024-07-02:
> * estore parameters from backup

ok, i have now tried a factory reset. entered my wifi credentials and nothing else (no mqtt or anything) cut the external power and i get a reboot right away. after that i have seen at most 13secs of uptime before the next reboot. (with external power = still ok).
I have also tried the test firmware. see reply to gskjold

this is the content of the cfg you requested:

amsconfig
version v2.3.6
boardType 7
netmode 1
hostname ams-3804
mdns 1
use11b 0
mqttHost *
mqttPort *
mqttClientId Kaifa MA304H4D
mqttPublishTopic Kaifa_MA304H4D
mqttPayloadFormat 4
mqttSsl 0
homeAssistantDiscoveryPrefix 
homeAssistantDiscoveryHostname ams-3804.local
homeAssistantDiscoveryNameTag 
meterBaud 2400
meterParity 8E1
meterInvert 0
meterDistributionSystem 2
meterMainFuse 20
meterProductionCapacity 0
gpioHanPin 12
gpioHanPinPullup 1
gpioApPin 0
gpioLedPin 2
gpioLedInverted 1
gpioLedPinRed 13
gpioLedPinGreen 14
gpioLedRgbInverted 1
gpioVccOffset 0.00
gpioVccMultiplier 1.000
gpioVccBootLimit 0.0
ntpEnable 1
ntpDhcp 1
ntpTimezone Europe/Stockholm
ntpServer 1.se.pool.ntp.org
priceEnabled 1
priceArea 10Y1001A1001A46L
priceCurrency SEK
thresholds 5 10 15 20 25 50 75 100 150 65535 1
dayplot 6 1719946857 139106.552 0 1039 1244 1093 1005 1192 1200 1202 1492 1383 1686 1141 1139 1700 1282 936 1165 1536 3134 4068 1225 1482 1139 1054 1244
monthplot 7 1719871211 139075.617 0 33841 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 32110 0
energyaccounting 6 7 12.51 12.51 10.09 0.00 0.00 0.00 1 2.69 2 3.13 0 0.00 0 0.00 0 0.00 32.11 0.00

thorbearswe, 2024-07-02:
> I was working on a unrelated thing on friday which may affect stability for 8266. Could you please try the attached firmware: [esp8266.zip](https://github.com/user-attachments/files/16039341/esp8266.zip)
> 
> Update guide: https://github.com/UtilitechAS/amsreader-firmware/wiki/Flashing-via-web-GUI

I've tested the test firmware but I still experience reboots without external power. also I got alot more checksum errors and even sporadick reboots with external power connected.

ArnieO, 2024-07-03:
> this is the content of the cfg you requested:

Thank you! I see nothing out of the ordinary. 

What puzzles me is: 
- You tried to roll back to v2.2.3 which you think is the version you were running before the upgrade; no luck. 
- We know for a fact that there are a quite large number of ESP8266-based devices running also with v2.3.x - without issues.

There have been situations (@gskjold can maybe elaborate) where devices need a full erase of the flash memory before getting back to normal behavior. This is not something particular with our firmware, but a generally known issue that may occur sometimes with ESP8266 when flashing firmware. It is however extremely seldom we see the need for that related to OTA (over-the-air) update, which is what you have done now.

But to do a flash erase on your device you would need an [FTDI module](https://www.kjell.com/se/produkter/el-verktyg/elektronik/utvecklingskit/arduino/moduler/luxorparts-usb-till-seriell-adapter-for-arduino-p88064). 
However, I cannot guarantee that it solves the issue you're seeing.

So to cut the chase, I prefer to send you a new Pow-U- please email us on post@amsleser.no so we can identify you and arrange a new shipment.

GaupeLnx, 2024-07-03:
Since a few days i also see Constant reboots on a POW-U+ Chip: esp32s2 (160MHz)  running version 2.3.6 which i updated a week ago or so. Not sure if i updated it from 3.5 or one or 2 versions before that. 
The device is constant connected to power.  It says Reason: Software reset (3/0)  Have not tried factory reset yet. Is that what i should try first?

ArnieO, 2024-07-04:
> Reason: Software reset (3/0)

Interesting and puzzling information. Most (if not all) software resets have been removed from our code - so this must be happening from within library code.

To help us narrowing down the issue; can you please:
1) Share a printout of your config page
2) Temporarily turn off MQTT to see if that ends the reset issue

GaupeLnx, 2024-07-04:
Can i "Turn off MQTT" by just deleting the MQTT address in the config? as i do not want to disable the MQTT as that would stop all my automations from working for days (reboots do not come often, last one was 9 hours ago)
[image]
[image]
[image]

ArnieO, 2024-07-04:
You turn off MQTT by deleting what is in the "Server" field.

thorbearswe, 2024-07-04:
> , I cannot guarantee that it solves the issue you're seeing.

i have spent the evening flashing my pow-u in different ways. 
I've tried a flash erase first, then downgraded just to do the updates one version a time. sadly i have had no progress in getting a stable uptime from my device...
so I gave up and sent you an email per request.

ArnieO, 2024-07-05:
> I gave up and sent you an email per request.

And it has now been taken care of - so I close the thread.

closed 2024-07-05 by ArnieO

GaupeLnx, 2024-07-05:
I ran  have ran my POW-U  with MQTT turned off and it was stable for 24 hours at least. Then i Factory reset it and restored the old config. Within 4 hours it rebooted again with reason: Software reset (3/0)

ArnieO, 2024-07-08:
> I ran have ran my POW-U with MQTT turned off and it was stable for 24 hours at least. Then i Factory reset it and restored the old config. Within 4 hours it rebooted again with reason: Software reset (3/0)

OK - so this indicates that there is something related to MQTT that provokes "Software reset (3/0)" in some setups. 

From your previous screenshot I see that you're doing IP assignment from DHCP (not static IP), and this is your MQTT settings:
[image]

Thank you for assisting us, this is obviously something we need to look into and see if we can remedy.

LarsKjeldsen, 2024-07-23:
My POW-K with ESP8266 did the same last month June 26 22:00 but started to work again at midnight July 1, I already ordered a new before it started to work again.
I kept the old 8266 in the meter, and it has then worked fine until today July 23 where it again started to reboot constantly.
I checked out the lates code and uploaded it to the device. same issue. 
I have attached serial output from the device (no HAN port attached) and all default except it is connected to my WiFi.
[ams dump.txt](https://github.com/user-attachments/files/16350832/ams.dump.txt)
It does NOT reboot when it acts as AP and not connected to my WiFi, but I guess a lot of other stuff is not in action.

The new POW-K with ESP32 is working fine but has only been in the meter from today.