UtilitechAS/amsreader-firmware #817: Could not connect to DEBUG Telnet port on my ESP32-C3

kusmi, 2024-07-01:
**Describe the bug**
I enabled debug&telnet, but could not access the device using telnet. I do have a ESP32-C3 device and just saw a commit comment about it: https://github.com/UtilitechAS/amsreader-firmware/commit/fd373a5846f93cdfff7c94a2560ee3bac754ee7b so I'm not sure if that is the reason for my telnet connection problem?

**To Reproduce**
Steps to reproduce the behavior:

- Enabled debug&telnet
- no telnet connection possible

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: Switzerland, Wettingen
 - Meter: L&G E450
 - AMS reader: gPlugM (ESP32-C3)

**Relevant firmware information:**
 - Version: 2.3.6
 - MQTT: yes
 - MQTT payload type: Homeassistant

**Additional context**
I try to debug my HAN port, as I don't receive any data. Not sure if my local electrical company has fully enabled the port yet.

ArnieO, 2024-07-02:
We know there are issues with v2.3.6 on some C3 devices. 
We'll need to change the memory mapping to solve it, so we  "know" the path towards a solution. 

That work does not currently have high priority, as it does not affect ESP32-S2 - which is the one we use in our products.
As we add features, S2 will eventually also be affected - so a solution will come at some point in time.

Until then, the advise for C3 users is to roll back to older firmware version.

closed 2024-07-02 by ArnieO

kusmi, 2024-07-02:
OK, I see, so I might have chosen the wrong hardware then 🥲
let's see what features do get dropped from C3 and I might upgrade to one of your device then...

ArnieO, 2024-07-02:
The maintenance and further development of our firmware is financed by sale of our own hardware.

The Swiss company gPlug have been selling devices with our firmware without any agreement with us. We invited them to sign a license agreement, but they have not wanted that. Our understanding is then that they prefer to exploit our work without contributing.

You had no way of knowing this, so no blame on you from our side!