UtilitechAS/amsreader-firmware #819: AMS reader time changes with timezone setting in OS

manfi001, 2024-07-04:
**Describe the bug**
AMS reader time changes with timezone setting in OS

**To Reproduce**
change timezone to less than 0, e.g. UTC-1 and AMS reader time changes to 1 hour less than real time.

**Expected behavior**
AMS reader time should be independent of timezone setting on readout device

**Hardware information:**
 - Country: Austria
 - Meter: Sagemcom
 - Encryption enabled [yes]
 - AMS reader: [ESP8266]
 - M-bus adapter: TSS721

**Relevant firmware information:**
 - Version: [2.3.6]
 - MQTT: [yes]
 - MQTT payload type: [JSON]
 - HAN GPIO: [UART2]
 - HAN baud and parity: [2400 8E1]
 - Temperature sensors [none]
 - ENTSO-E API enabled: [no]

**Additional context**
Discovered this behavior by chance during a recent stay in Latin America

manfi001, 2024-07-04:
Correction to the above posting: the date changes to the previous day

ArnieO, 2024-07-05:
Were you using the device in Latin America when you saw this, or did you read your home device remotely?

manfi001, 2024-07-05:
I read the home device remotely. But it's the same beaviour if I change the time zone of the PC or tablet at home and access the readers page.Sent from my Galaxy

  timezone setting in OS (Issue #819)

ArnieO, 2024-07-08:
Just to be sure I understand your report: 

- This is the date indicated in upper right corner of the GUI, not timestamp on parameters sent to MQTT - right?
- Is the indicated date correct according to the time zone set, or is there a 1 hour error?

manfi001, 2024-07-08:
- Yes, it's the date in the upper right corner, MQTT timestamp is not affected
- The indicated date is wrongly set one day back

ArnieO, 2024-07-09:
Thank you for the clarifications, this must be a bug. 
Using the device in other time zones is not common, you are probably the first one to "test" this.

closed 2024-08-06 by gskjold