UtilitechAS/amsreader-firmware #822: Undefined Month in setup and different views

A-Wangen, 2024-07-15:
**Describe the bug**
No month is available in the different views and setup. see Screenshots
And sometimes the date does not automatically update to the current time and date.
 
**To Reproduce**
Noting special is done to 

**Expected behavior**
A correct month in the differnt views

**Screenshots**
[image]
[image]
[image]

**Hardware information:**
 - Country: Norway
 - Meter: Aidon
 - Encryption enabled NO
 - AMS reader: Pow-U
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: v2.3.6
 - MQTT: yes
 - MQTT payload type: JSON
 - HAN GPIO: 
 - HAN baud and parity: 2400 8E1
 - Temperature sensors
 - ENTSO-E API enabled: 
 

**Additional context**
Add any other context about the problem here.

ArnieO, 2024-07-15:
It looks like you could have an NTP issue in your configuration? 
Please post screenshots of:
- Config screen
- Status/Info screen

Please include the status bar (top section of the screen) in at least one of the screenshots.

egiljae, 2024-07-28:
@A-Wangen Check your User interface -> Language setting. I had "Unknown" set, and was experiencing the same as you.

ArnieO, 2024-07-29:
> @A-Wangen Check your User interface -> Language setting. I had "Unknown" set, and was experiencing the same as you.

Interesting. Do you know how Language became "Unknown" for you?

egiljae, 2024-07-29:
> > @A-Wangen Check your User interface -> Language setting. I had "Unknown" set, and was experiencing the same as you.
> 
> Interesting. Do you know how Language became "Unknown" for you?

It happened after upgrading from a previous version. Some settings had to be manually updated since they where either not set or partially missing, eg. NTP was after update set to ".org" and had to be manually updated to "pool.ntp.org".

ArnieO, 2024-07-29:
> It happened after upgrading from a previous version. Some settings had to be manually updated since they where either not set or partially missing, eg. NTP was after update set to ".org" and had to be manually updated to "pool.ntp.org".

Thank you for this information. 
Such issues could typically arise if user does a "long" upgrade (not from previous release).
We will look into this issue and seek to avoid such problems in future upgrades.

closed 2024-08-06 by gskjold