UtilitechAS/amsreader-firmware #823: 2.3.5 and up breaks HAN data.

githubpassera, 2024-07-20:
**Describe the bug**
No data from HAN port 2.3.5+

Rolling back to 2.3.4 solves issue.
2.3.5 & 2.3.6
[image]

**Hardware information:**
 - Country: Sweden
 - Meter: Kaifa MA304H4
 - Encryption enabled no
 - AMS reader: ESP32

**Relevant firmware information:**
 - Version: 2.3.5 + 2.3.6
 - MQTT: YES
 - MQTT payload type: [e.g. JSON]
 - HAN GPIO: UART0
 - HAN baud and parity:2400 8E1

ArnieO, 2024-07-20:
Can you show us screendump of the config screen when it is not working?

githubpassera, 2024-07-20:
234 Working.
[image]
236 Not working
[image]

egiljae, 2024-07-21:
Also not able to get 2.3.6 working, no HAN data. 2.3.4 works just fine, same config.
[image]

egiljae, 2024-07-22:
> @egiljae: You are using a device with ethernet, maybe the issue is related to that in some way. I see that you have HAN Rx and HAN Tx on the same GPIO: ![image](https://private-user-images.githubusercontent.com/10295178/350884506-ded1e938-cd5f-4e15-bf53-d6da5ae5c61d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE2MzY1NjcsIm5iZiI6MTcyMTYzNjI2NywicGF0aCI6Ii8xMDI5NTE3OC8zNTA4ODQ1MDYtZGVkMWU5MzgtY2Q1Zi00ZTE1LWJmNTMtZDZkYTVhZTVjNjFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIyVDA4MTc0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlkYjlkMzRjZmI2MzFjNmFjMzA3OGVhZWVkYWI3ZWQ2YjcxYWVhYjA0ZGE3ZGQ0YjI3MGQ2NjFjNzVkNmVlZmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.KNLnDmYBsatELXmgR4YIeW47-_pDjSEVnINaSAmPHT8) That does not look sensible. In most situations you do not need "HAN Tx". Please try to delete it to run v2.3.6 (leave the field empty).

You are correct. Setting HAN TX to something other than GPIO4 works. I could not set it blank, since there is no option for it, neither could I find any reference to it in the config file. I just set it to a GPIO pin that is not in use. The HAN TX field was automatically set after upgrading from an older version.

[image]

ArnieO, 2024-07-22:
> You are correct. Setting HAN TX to something other than GPIO4 works.
(...)
The HAN TX field was automatically set after upgrading from an older version.

I suspected something like that. There are very few users on Ethernet, so feedback is very welcome! 
I will notify @gskjold that there should be an option to set the "HAN Tx" field blank, so that it will be fixed in future revisions.

githubpassera, 2024-07-23:
Yes, that was the issue, I set the TX to GPIO38 (not used) and 2.3.6 is now working, thank you.

closed 2024-07-23 by ArnieO