UtilitechAS/amsreader-firmware #827: low voltage on device without usb power

catohagen, 2024-08-05:
I dont know if its an issue with the device, or the port in the meter, but the voltage drops and the device isn't reporting any
data to home assistant.
It started happening after 2.3.5 firmware....I also tried 2.3.6 but it was the same.
If USB power connected it works 100% all the time and shows 3.28volts

removing USB power, and voltage drops under 3V with red background and device barely responds

Firmware
Installed version: v2.3.5

Chip: esp32s2 (160MHz)
Device: [Pow-U+](http://192.168.1.3/vendor)
MAC: 58:XX:XX:XX:XX:XX
AP MAC: 58:XX:XX:XX:XX:XX
Last boot: 05.08.2024 10:07
Reason: Vbat power on reset (1/0)

ArnieO, 2024-08-05:
It shouldn't behave like that. 
If I assume correctly from your name, this is a new device shipped in March -24.
You could (as a test) revert to v2.2.28, which was on it when it shipped - but downgrading from 2.3.x to 2.2.x means you'd have to reconfigure it from scratch.

My guess is that something in the settings got messed up during a firmware update. 
Can you:

1. Post screenshot of the Config screen
2. Make a [configuration backup](https://wiki.amsleser.no/en/firmware#backup-restore) (without "secrets") and post the content
-

catohagen, 2024-08-05:
heres screenshot + cfg

[image]

amsconfig
version v2.3.5
boardType 7
netmode 1
hostname ams-1267
ip 192.168.1.3
gateway 192.168.1.1
subnet 255.255.255.0
dns1 8.8.8.8
mdns 1
use11b 1
mqttHost 192.168.1.161
mqttPort 1883
mqttClientId ams-1267
mqttPublishTopic amsreader
mqttPayloadFormat 4
mqttSsl 0
homeAssistantDiscoveryPrefix 
homeAssistantDiscoveryHostname 
homeAssistantDiscoveryNameTag 
meterBaud 0
meterInvert 0
meterDistributionSystem 2
meterMainFuse 56
meterProductionCapacity 0
gpioHanPin 16
gpioHanPinPullup 1
gpioApPin 0
gpioLedPinRed 13
gpioLedPinGreen 14
gpioLedRgbInverted 1
gpioVccPin 10
gpioVccOffset 0.00
gpioVccMultiplier 1.000
gpioVccBootLimit 0.0
gpioVccResistorGnd 22
gpioVccResistorVcc 33
ntpEnable 1
ntpDhcp 1
ntpTimezone Europe/Oslo
ntpServer pool.ntp.org
priceEnabled 1
priceArea 10YNO-2--------T
priceCurrency NOK
thresholds 5 10 15 20 25 50 75 100 150 65535 3
dayplot 6 1722891612 164444.016 0 1296 1568 1680 2016 1136 1456 1312 1280 1280 1520 3808 3344 2512 1296 1360 2016 1360 1200 1408 2096 1840 1520 1456 1328
monthplot 7 1722808800 164404.464 0 44061 44061 44061 44061 44144 35616 48512 40848 33744 36512 35360 35776 36560 38928 34720 35216 38048 32848 45568 43744 44560 42106 42106 42106 42106 44061 44061 44061 44061 44061 44061
energyaccounting 6 8 9.45 34.77 261.24 0.00 0.00 0.00 4 2.69 5 3.81 0 0.00 0 0.00 0 0.00 39.62 0.00

catohagen, 2024-08-05:
wait a minute...
[image]
I noticed power saving settings on wifi was set to : off

I'm trying it on "Minimum" now, and it seems like voltage is normal....
Maybe this was it all the time...

ArnieO, 2024-08-06:
> Maybe this was it all the time...

Maybe, the recommended setting for Pow-U depends on the Wi-Fi environment - see this article: https://www.amsleser.no/module/ets_blog/blog?id_post=25

I recommend turning off "Allow 802.11b" (deactivated is the default setting). 
Modern routers do not need 802.11b compatibility, and we have seen that setting can bleed power in some cases.

closed 2024-08-06 by ArnieO