UtilitechAS/amsreader-firmware #829: Real time calculation Exporting

arnwik, 2024-08-06:
Real time calculation for production/exporting is only showing when the system actually are Exporting. The instance the system don't export anymore, no data for exporting is showing (like it does on the Consumption).
There should be a toggle to have production/export permanently on.

ThorAlex87, 2024-08-07:
There is, third to last section "user interface" on the configuration page has "enabled", "disabled" and "auto" settings for everything on the main page.

gskjold, 2024-08-07:
Also, if you configure "Production capacity", it will always show

ThorAlex87, 2024-08-07:
> Also, if you configure "Production capacity", it will always show

Hey I had not noticed production could be set, cool! But i prefer having both import and export meters set to auto, as they are mutually exclusive so there is no point in showing both?

ArnieO, 2024-08-07:
> > Also, if you configure "Production capacity", it will always show
> 
> Hey I had not noticed production could be set, cool! But i prefer having both import and export meters set to auto, as they are mutually exclusive so there is no point in showing both?

The choice is yours! 😁

ThorAlex87, 2024-08-07:
But if I understand you correctly I cannot choose to configure production and have the export meter only show when there is export?

(I'm currently testing this, so I will find out by the end of the day)

ArnieO, 2024-08-07:
> But if I understand you correctly I cannot choose to configure production and have the export meter only show when there is export?

Passing the question to @gskjold :  
If production is configured in Config/Meter; does that override an "Auto" setting in **User interface / Export gauge** so that Export gauge is always shown?

gskjold, 2024-08-07:
Correct, if production capacity is configured, gauge is always showing if set to auto

closed 2024-08-15 by gskjold