UtilitechAS/amsreader-firmware #830: Voltage, current, power and power factor do not add up

KDYR, 2024-08-07:
**Describe the bug**
Looking at the Voltage, current, power and cosphi (power factor) on the "front page" of the web interface it seems some of the values are wrong. The displayed power in Watt for each of the phases do not correspond to the values that can be calculated from U, I, and cosphi.

**To Reproduce**
In the setup for Interface, make sure Voltage, Amperage, and Power factor are Auto or Enabled, such that they all show.

**Expected behavior**
Displayed values should satisfy the basic formula for the relation between the four

**Screenshot**

[image]

Formula and manual calculations:

[image]

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled: Yes
 - AMS reader: Pow-K+

**Relevant firmware information:**
 - Version: 2.3.6  (on esp32s2)
 - MQTT: Yes
 - MQTT payload type: Home Assistant
 - HAN GPIO: 
 - HAN baud and parity: 2400 8N1
 - Temperature sensors:
 - ENTSO-E API enabled: No

**Additional context**
I am wondering if all four values are reported by the meter, of if one of them is calculated by the amsreader software. 

Attached is Telnet log file with three "batches" of data from the meeter. 
[Log.txt](https://github.com/user-attachments/files/16520927/Log.txt)
Not taken exactly at the time of the screenshot

KDYR, 2024-08-07:
Small correction: It should be "L3:" in last calculation - but numbers ARE from L3

ArnieO, 2024-08-07:
> I am wondering if all four values are reported by the meter, of if one of them is calculated by the amsreader software.

All these values are reported by the meter, not calculated by the firmware. 

In a 3 phase system, when cos phi / PF is not 1.0 on all phases (it never is!), these equations cannot be used, as you will have to take into account the phase angles (not reported by the meter) and do vector calculations. There is also a factor SQRT(3) involved, which you forgot.

If you really want to dive into this, you need to read up on power calculations in 3 phase installations; it is surprisingly complex. But bottom line is: 
**All power, current and voltage values are reported by the meter.**

closed 2024-08-07 by ArnieO

KDYR, 2024-08-07:
Hi @ArnieO 
The simple single phase formula in my report also applies in a three phase system with a null/neutral and a phase angle between the voltage and the current. And the sqrt(3) only comes into the picture when one has to joggle both phase-to-phase values and phase-to-neutral values. 
The phase angel (usually termed "phi") is indeed important, but that is represented by the power factor as the power factor is cos(phi).
In your answer you confirm that "power, current and voltage values are reported by the meter" - that is three of the four - so my final question is: What is the source of the power factor shown on the amsleser web interface? - is that also from the meter via the HAN?
BR

ArnieO, 2024-08-07:
I think I was a bit imprecise or wrong on some points in my previous post. Sorry for that, but the subject is complex and it is months between each time I have to refresh my memory. 

1. The issue is load balancing, not whether all PFs are close to 1,0. The equations you refer to work if load is well balanced between the phases, but in a small installation like a household that is the exception, not the rule. So in most cases the calculation will not work as you expect.

2. Power factor per phase is reported by the meter. Paragraph 5.2 in [this document](https://documentation.kamstrup.com/docs/HAN_NVE/en-GB/Information_sheet/CONTAAA62C8B611D4C4EB3186FD81AEE0152/) shows all parameters reported from a DLMS configured Kamstrup Omnipower meter in Denmark. 

3. As a general statement, **_all_** values presented in the user interface (and sent over MQTT) are coming from the power meter. You will be able to modify them by using Multipliers that can be activated on the config screen / [Meter tile](https://wiki.amsleser.no/en/firmware#meter). Multipliers are for instance needed on "industrial" installations with current transformers - but it is then user's responsibility to ensure the multiplier is set correctly.

**Side note**
Some power meter models only report current on two of the phases, and nothing on L2. In earlier versions of our firmware we attempted to calculate the length of the missing current vector from the available data. However, we realized that the calculation was only sufficiently accurate if the load was balanced. So we abandoned that, and now in stead report it as "Not available", see [this article](https://www.amsleser.no/module/ets_blog/blog?id_post=31).

KDYR, 2024-08-07:
@ArnieO 
Thanks for the reply and the links. I will study.
BR

KDYR, 2024-08-09:
Hi @ArnieO 
I have studied and have two findings:

1: I claimed earlier that the formula in my initial post _does_ apply to individual phases in a three-phase system; that is right as long as voltage and current have nice, sinusoidal shape. When voltage and/or current is "distorted", an additional element has to be considered and the formula has to be amended a bit:
[image]
You see that the concept of "power factor" is not just cosine of the usual (ideal) phase angle, but in a "Total Power Factor" one has to factor in also a power factor originating from the Total Harmonic Distrotion (THD), which is representing the 'deviation' from nice harmonic, sinusoidal curves (this is a result of all the electronics with semiconductors, that we have in our homes).
The Kamstrup meter is reporting the cosphi (power factor) via HAN, but depending on the THD one has to adjust the power calculated from, U, I and cosphi further. 
The Kamstrup meter is measuring also the PF_THD, but is is not available on the HAN port. It seems this parameter is available only to the utility company.

2: In the amsleser web interface the PF_cosphi is presented correctly. In above snip the values were 0.73, 0.44 and 0.73. This can also be presented as 73%, 44% and 73%. 
However, the values available in HomeAssistant (via MQTT) have been divided by 100: I see numbers like 0.73%, 0.44% and so on. That's a bug. If you agree, I will raise a separate issue in on this matter.

Thanks for your efforts!
BR

ArnieO, 2024-08-09:
> However, the values available in HomeAssistant (via MQTT) have been divided by 100: I see numbers like 0.73%, 0.44% and so on. That's a bug. If you agree, I will raise a separate issue in on this matter.

We have around 3000 users, and a large part of those are on Home Assistant - so congratulations with being the first one to noticing this! 😅 

As PF is per definition a number between 0 and 1, it seems to me that this is a bug in HA. 
But I think the easiest remedy is that we adjust the HA payload to send PF as an integer between 0 and 100.

You are welcome to raise a separate issue on this!