UtilitechAS/amsreader-firmware #832: Energy use history missing

sm1oii, 2024-08-12:
[image]

sandjo, 2024-08-12:
I also see the same after upgrading to version 2.3.7. It didn't occur immediately, but after 1-2 days.

If you click on "Edit data" everything comes up as zero.

kusmi, 2024-08-12:
for me it seems to start yesterday, e.g. in MQTT (Home Assistant) I still have the correct values, but in amsreader I see some gaps yesterday at 22:00 and the daily counter is only 0.8 kWh (instead of 20.81 kWh as reported in Home Assistant). I also don't see any "gaps" in MQTT on home assistant, so there are MQTT values delivered all the time.

### Home Assistant:
[image]
[image]

### amsreader:
[image]

ArnieO, 2024-08-13:
Thank you for reporting!
It seems to be two different errors (but could of course potentially be linked):
- @sm1oii and @sandjo lost history completely
- @kusmi has bugs in the history (both the hourly and the daily)

Since this is not a general problem (have only seen these reports, and our own devices are find), it could be interesting knowing which hardware you three are running.

sandjo, 2024-08-13:
I'm running:

Chip: esp32s2 (160MHz)
Device: Pow-P1

History has not only been lost, it is not filling up either.

To rule out client side issues I've tested with both Firefox and Safari.

ArnieO, 2024-08-13:
> Device: Pow-P1

OK - thank you. 
Can you do a [config backup](https://wiki.amsleser.no/en/firmware#backup-restore) (without "secrets") and post it here?

ArnieO, 2024-08-13:
It could potentially also be useful if you could run a Telnet Debug (in "Verbose" mode) to see if that reveals anything abnormal. https://www.amsleser.no/module/ets_blog/blog?id_post=24

kusmi, 2024-08-13:
I'm the one with the esp32c3 device, sorry :-)

What I recognized: As you see in my graph, I have a wrongly empty hour (22:00) and for the daily sum, only the kwh *after* that is added. That's why it shows only 0.8 kWh for the day. Also it seems mine sometimes "browns-out" and reboots, so it somehow lost some data? Funnily I see some reboots (as reported in the uptime value on MQTT) in the last month - but only this single day got messed up. All other reboots triggered no problem.

sandjo, 2024-08-13:
@ArnieO before I saw your request I had already downgraded to 2.3.6. History is still lost, but it starting adding new values again after the downgrade.

I have now upgraded to 2.3.7 again and will wait for the issue to trigger again. Last time it took 24+ hours before the problem showed up. If I see it again I'll grab the data you requested, if not I'll report back after at least three days have passed that it didn't trigger.

ArnieO, 2024-08-13:
> @ArnieO before I saw your request I had already downgraded to 2.3.6. History is still lost, but it starting adding new values again after the downgrade.
>
> I have now upgraded to 2.3.7 again and will [...]

OK, I understand. 

We just received a 3rd report of this issue by email, so this seems to be a bug that sneaked in with v2.3.7. 
I expect we now have around 2000 users (@gskjold - I believe you have a better estimate?) to be on v2.3.7 now, and so far only 3 reports of this issue - so it seems to be triggered by something marginal that we need to figure out.

Thank you for helping us figure out what is going on here!

sandjo, 2024-08-13:
This time the issue triggered faster.

Two backups are attached, the one with 2.3.6 in the name is from after the downgrade from 2.3.7 to 2.3.6 just prior to upgrading to 2.3.7 again and the one with 2.3.7 in the name is from after I saw that the issue reappeared.

[sandjo-configfile-2.3.6.cfg.txt](https://github.com/user-attachments/files/16602953/sandjo-configfile-2.3.6.cfg.txt)
[sandjo-configfile-2.3.7.cfg.txt](https://github.com/user-attachments/files/16602955/sandjo-configfile-2.3.7.cfg.txt)

The only difference is the version, dayplot, monthplot and energyaccounting fields.

I'll enable telnet debugging now and see if I can see anything there.

sandjo, 2024-08-13:
Telnet debugging gave this every second, which looks to be the expected outcome. After several minutes I gave up looking at it. Are there any specific time slots that I should examine more carefully?

```
(D) Received valid DSMR at 0
(V) DSMR frame:
[...]
```

ArnieO, 2024-08-13:
Thank you!

> The only difference is the version, dayplot, monthplot and energyaccounting fields.

Right. The historics stored in the "dayplot" line are effectively gone - hence lost. We must assume the same would be the case for the "monthplot" line, if there had been data there.

> Are there any specific time slots that I should examine more carefully?

Data is saved just after passing full hour, you'll have to look at it when that happens.

Pajaukkeli, 2024-08-13:
I have a similar issue, 2.3.6 working fine, but after installing 2.3.7, lost history data. And now the 24h daily graph is also not showing hourly status

Meter Aidon 7534

[configfile.cfg (1).txt](https://github.com/user-attachments/files/16603327/configfile.cfg.1.txt)

[image]

ArnieO, 2024-08-13:
> I have a similar issue, 2.3.6 working fine, but after installing 2.3.7, lost history data.

I assume from your nick that you are the customer in Finland with a Pow-P1 that I mentioned here? https://github.com/UtilitechAS/amsreader-firmware/issues/832#issuecomment-2286662686 

**I advice those are affected by this bug to roll back to v2.3.6, and wait for bugfix in next firmware revision.**

- **Download  [ams2mqtt-esp32s2-2.3.6.bin ](https://github.com/UtilitechAS/amsreader-firmware/releases/download/v2.3.6/ams2mqtt-esp32s2-2.3.6.bin)**
- **Go to Status/Info screen and click "Select firmware for upgrade"**
- **Select the downloaded file**

sandjo, 2024-08-13:
I tried to catch any debug output related to saving data at the past hour change. Only two additional lines as show below. Only thing I found odd was that it seemed to be triggered twice, once by the meter clock and once by the internal clock.

```
(D) Received valid DSMR at 0
(D) Updating data storage, triggered by meter clock 18:00:00 UTC (Internal clock: 17:59:59)
(V) DSMR frame:
[... one frame ...]
(D) Received valid DSMR at 0
(D) Updating data storage, triggered by internal clock 18:00:00 UTC (Meter clock: 18:00:01)
(V) DSMR frame:
```

ArnieO, 2024-08-14:
> I tried to catch any debug output related to saving data at the past hour change. Only two additional lines as show below. Only thing I found odd was that it seemed to be triggered twice, once by the meter clock and once by the internal clock.

Thank you very much!

leakim1957, 2024-08-14:
Missing energy use history occurd for me as well after latest upgrade,  I just did not report it.

ArnieO, 2024-08-15:
> Missing energy use history occurd for me as well after latest upgrade, I just did not report it.

Thank you for reporting. Are your symptoms as described above: Daily and monthly history erased, and not rebuilding?

gskjold, 2024-08-15:
I'm going to take a wild guess that you all have a P1 meter?

Anyhow, here is a firmware which hopefully fix it:
[esp32s2.zip](https://github.com/user-attachments/files/16628093/esp32s2.zip)
[esp32c3.zip](https://github.com/user-attachments/files/16628094/esp32c3.zip)
[esp32s3.zip](https://github.com/user-attachments/files/16628095/esp32s3.zip)
[esp32.zip](https://github.com/user-attachments/files/16628096/esp32.zip)
[esp8266.zip](https://github.com/user-attachments/files/16628098/esp8266.zip)
[esp32solo.zip](https://github.com/user-attachments/files/16628099/esp32solo.zip)

Flashing instructions: https://github.com/UtilitechAS/amsreader-firmware/wiki/Flashing-via-web-GUI

leakim1957, 2024-08-15:
Correct Gunnar.
I will upgrade tomorrow Friday and let you know asap.
///Mikael

Skickat från Outlook för Android<https://aka.ms/AAb9ysg>

leakim1957, 2024-08-18:
Hi Gunnar
No luck with the update.
I used the "esp32s2.zip" and it installed without problems, following your instruction.
I installed it today sunday at 12:17 but no records of  "Energy use last 24 hours " at this time (17:46)

Med Vänlig Hälsning
Mikael Bender Wilsson

ArnieO, 2024-08-18:
> No luck with the update.

Can du try a Factory Reset; red button bottom left on the Config screen?

nossaile, 2024-08-19:
I got the same issue with v2.3.7. No values presented in 24h and 31day history. Real time calculations looks only to summarize present hour, see below. Reporting to HA works fine.
[image]

nossaile, 2024-08-19:
[image]

ArnieO, 2024-08-19:
@gskjold : Similar issue reported in January by @nossaile: https://github.com/UtilitechAS/amsreader-firmware/issues/717

pagocs, 2024-08-19:
A have same issue after updated to 2.3.7. The 24 hours and 31 days charts are missing immediately after the devices was rebooted...
 
[image]

ArnieO, 2024-08-19:
> A have same issue after updated to 2.3.7.

Thank you for reporting - and again on a Pow-P1. 
I'm confident @gskjold will find the fix for this as soon as he finds time.

gskjold, 2024-08-19:
> I used the "esp32s2.zip" and it installed without problems, following your instruction.

I am curious, did you upload the zip file, or unzip and uploaded the bin file? What version does your device report that it is running now? Should not be 2.3.7

leakim1957, 2024-08-20:
I unzipped it and uploaded the bin file and could see that the version was 2.3.7
The problem I got after factory reset was that the P1 can not comunicate with my wifi.
The LED in the P1 is flashing blue and green which indicates that it is receiving the meter data correct.
I will continue work on the wifi. I include the config file, can yoy see anything strange with it? (I deleted my wifi pw)
///Mikael

Skickat från Outlook för Android<https://aka.ms/AAb9ysg>

ArnieO, 2024-08-20:
> The problem I got after factory reset was that the P1 can not comunicate with my wifi.

If you did a factory reset using the button in the bottom left of the config screen, the device is back to the state it had when you received it. You need to go through the [initial configuration procedure](https://wiki.amsleser.no/en/getting-started) again.

ArnieO, 2024-08-20:
Did anyone else try the fix posted by @gskjold earlier in this thread?

toast1234, 2024-08-20:
I have two Pow-P1 esp32s2 in two locations. Both lost the history. I tested the patch on one and did not notice any difference. E.g. the graphs did not get populated after a day (The build tag was shown as version so update went ok). I then rolled back that device to 2.3.6 and history is building up. My other device still has an uptime of 13 days since the upgrade to 2.3.7. Two days ago I opend the "edit data" for the 24h and 31day, pressed "clear all" and then save. Since then the history is still available. I will keep an eye on it. I use Home Assistant and seldom check in the GUI. As a side note I can mention that after upgrading my wifi AP I had to lower the DTIM intervall from 8 to 4 to not have random reboots. I have a few other IOT devices that did not have reboot issues with DTIM 8 on the newer AP FW.

ArnieO, 2024-08-20:
> I have two Pow-P1 esp32s2 in two locations. Both lost the history. I tested the patch on one and did not notice any difference. E.g. the graphs did not get populated after a day (The build tag was [...]

Thank you for quality feedback! 
So the patch does not seem to work.
Did you try a Factory Reset (red button lower left on config screen)? You will need to re-initialize it to connect it to your Wi-Fi.

**Thank you for very interesting input regarding DTIM interval.** 
We have 2-3 customers that have struggled with random reboots, I will propose them to try that. 
Most of us do not have access to such settings (modern routers for consumers...), but the persons having this issue all seem to have more advanced Wi-Fi systems (Unifi) where they can adjust the DTIM interval.

ArnieO, 2024-08-20:
@toast1234 
How much "free mem" do you see in the top bar on a failing device?
[image]

toast1234, 2024-08-20:
The device currently running 2.3.7 has an uptime of 13 days and have ~95 to ~97 kb free when checking. What it had when failing and before I cleared the history data I dont know. If the history fail again I will note down the memory and take a config backup. This device is in a remote location so can not "play" with it. For my other device that is currently rolled back to 2.3.6 working I can look in to doing another attempt on and a factory reset later in the week when I am in position to be able to initilize the wifi.

ArnieO, 2024-08-20:
> For my other device that is currently rolled back to 2.3.6 working I can look in to doing another attempt on and a factory reset later in the week when I am in position to be able to initilize the wifi. When I do that would you like me to start with the patch or the 2.3.7 release?

Thank you, that would be great! Just check in here before you do so, in case @gskjold has posted a new patch.

If a new patch is available, please try it first. If not, try v2.3.7 first and check if "Free mem" goes low (could indicate memory leakage etc).

toast1234, 2024-08-20:
I also want to add to the separate reboot DTIM issue as I actually made wifi 2 changes at the same time to solve it. I also changed broadcast filtering from "ARP" to "unicast ARP only". E.g. from converting ARP to unicast, drop broad/multicast to converting ARP to unicast and letting broad/multicast pass as is. It was very noticible as reboots occured multiple times during the day, at leased 20 times. After the change there are 0 reboots. I assumed DTIM but did not investigate futher.

gskjold, 2024-08-20:
I don't think DTIM causes reboots. I have three networks at home with DTIM=1, DTIM=4 and DTIM=10, all have test devices connected to them and I have never had any issues on any of the networks

ArnieO, 2024-08-20:
> I don't think DTIM causes reboots.

What about the other things he changed?

gskjold, 2024-08-20:
I'm not sure what the effect of that other change is, but multicast can affect WiFi wake time on all clients on the network

toast1234, 2024-08-20:
For the reboot issue, keep also in mind that the problem only came after I upgraded the AP firmware. I had no issue before. The reson I made those two adjustmets was becase I have the same setup in a different location and those two parameters were the only difference between the setups. After making the change for the SSID the random reboot stopped. I did not reboot the AP or made any other modification. The AP that had the issues is also in a much more wifi crowded location but otherwise setup is identical.

gskjold, 2024-08-20:
Here is another attempt at fixing the missing history issue (Version 1592449):
[esp32s2.zip](https://github.com/user-attachments/files/16680601/esp32s2.zip)
[esp32c3.zip](https://github.com/user-attachments/files/16680602/esp32c3.zip)
[esp32.zip](https://github.com/user-attachments/files/16680603/esp32.zip)
[esp8266.zip](https://github.com/user-attachments/files/16680604/esp8266.zip)
[esp32solo.zip](https://github.com/user-attachments/files/16680605/esp32solo.zip)
[esp32s3.zip](https://github.com/user-attachments/files/16680606/esp32s3.zip)

Flashing instructions: https://github.com/UtilitechAS/amsreader-firmware/wiki/Flashing-via-web-GUI

toast1234, 2024-08-20:
Yes, [1592449] passed the one hour rotate. I will report back tomorrow only if it has failed over the night.

After a debug session I see/saw the same as sandjo reported. Below is the result including the latest build.

ntpEnable 1
ntpDhcp 0
ntpTimezone Europe/Stockholm
ntpServer 192.168.5.1
Meter interval: 1s
MQTT interval: 10s

Build: 2.3.6
```
dayplot 6 1724173200 42744.918 0 1087 722 715 688 1331 2468 687 1085 719 907 1149 846 792 808 1733 1153 1380 1556 1313 931 1311 730 702 767 0.001 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
monthplot 7 1724104800 42725.179 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 33046 25331 0 0 0 0 0 0 0 0 0 0 0 0 0.001 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
energyaccounting 6 8 2.57 41.38 102.37 0.00 0.00 0.00 19 2.54 9 2.99 20 2.47 4 2.65 18 3.21 502.08 0.00

```

Build: 2.3.7 OTA update (history still intact after upgrade)
```
dayplot 6 1724173200 42744.918 0 1087 722 715 688 1331 2468 687 1085 719 907 1149 846 792 808 1733 1153 1380 1556 1313 931 1311 730 702 767 0.001 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
monthplot 7 1724104800 42725.179 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 33046 25331 0 0 0 0 0 0 0 0 0 0 0 0 0.001 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
energyaccounting 6 8 2.57 42.75 102.37 0.00 0.00 0.00 19 2.54 9 2.99 20 2.47 4 2.65 18 3.21 502.08 0.00
```

Hour rotate @20:00 (history lost)
```
(D t:1210346ms) Received valid DSMR at 0
(D t:1211345ms) Received valid DSMR at 0
(D t:1211508ms) Updating data storage, triggered by meter clock 18:00:00 UTC (Internal clock: 17:59:59)
(D t:1212368ms) Received valid DSMR at 0
(D t:1212532ms) Updating data storage, triggered by internal clock 18:00:00 UTC (Meter clock: 18:00:01)
(D t:1213337ms) Received valid DSMR at 0
(D t:1214346ms) Received valid DSMR at 0
(D t:1215346ms) Received valid DSMR at 0
(D t:1216345ms) Received valid DSMR at 0
dayplot 6 1724176800 42746.963 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0.001 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
monthplot 7 1724176800 42746.963 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0.001 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
energyaccounting 6 8 2.57 39.93 102.37 0.00 0.00 0.00 19 2.54 9 2.99 20 2.47 4 2.65 18 3.21 502.08 0.00
```

Build: c5c0b52 (Manual FW upload)
Hour rotate @21:00 (history lost)
```
(D t:3293580ms) Received valid DSMR at 0
(D t:3294562ms) Received valid DSMR at 0
(D t:3294727ms) Updating data storage, triggered by meter clock 18:00:00 UTC (Internal clock: 18:59:59)
(D t:3295651ms) Received valid DSMR at 0
(D t:3295817ms) Updating data storage, triggered by internal clock 19:00:00 UTC (Meter clock: 18:00:01)
(D t:3296569ms) Received valid DSMR at 0
(D t:3297562ms) Received valid DSMR at 0
dayplot 6 1724180400 42748.069 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0.001 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
monthplot 7 1724180400 42748.069 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0.001 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
energyaccounting 6 8 2.57 39.93 102.37 0.00 0.00 0.00 19 2.54 9 2.99 20 2.47 4 2.65 18 3.21 502.08 0.00
```
**Build: 1592449 (Manual FW upload)
Hour rotate @22:00  (history stored ok)**
```
(D t:3072728ms) Received valid DSMR at 0
(D t:3073816ms) Received valid DSMR at 0
(D t:3073981ms) Updating data storage, triggered by list 4 (internal clock 20:00:00 UTC, neter clock: 20:00:01)
(D t:3073989ms) Day is not happy
(D t:3074000ms)  - normal
(I t:3074004ms) Saving data
(D t:3074715ms) Received valid DSMR at 0
```

In a different location I have a meter that update every 10s. Maybe less likely to happen then. This meter also have export. See below:
Meter interval: 10s
MQTT interval: Real Time

Build: 2.3.7
Hour rotate @21:00 (history stored ok)
```
(D t:1172236188ms) Received valid DSMR at 0
(D t:1172246176ms) Received valid DSMR at 0
(D t:1172246393ms) Updating data storage, triggered by internal clock 19:00:01 UTC (Meter clock: 19:00:00)
(D t:1172246398ms) Day is not happy
(D t:1172246409ms)  - normal
(I t:1172246414ms) Saving data
(D t:1172256176ms) Received valid DSMR at 0
(D t:1172266185ms) Received valid DSMR at 0
dayplot 6 1724180401 15728.582 0 526 533 528 453 112 24 0 0 0 0 1 0 0 0 0 0 58 222 500 537 535 534 532 521 6576.638 0 0 0 0 4 245 499 1423 1926 2751 1085 2101 5370 2945 2171 1060 217 22 0 0 0 0 0 0
monthplot 7 1724104800 15724.572 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 2369 5770 0 0 0 0 0 0 0 0 0 0 0 0 6554.819 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 34725 44668 0 0 0 0 0 0 0 0 0 0 0 0
energyaccounting 6 8 0.44 9.23 39.05 3.88 54.40 220.39 12 1.41 10 2.34 11 1.79 13 2.64 14 4.13 195.37 1149.69
```
Build: 2.3.7
Hour rotate @22:00 (history stored ok)
```
(D t:3222794ms) Received valid DSMR at 0
(D t:3223007ms) Updating data storage, triggered by internal clock 20:00:01 UTC (Meter clock: 20:00:00)
(D t:3223013ms) Day is not happy
(D t:3223024ms)  - normal
(I t:3223029ms) Saving data
(D t:3232802ms) Received valid DSMR at 0

```

toast1234, 2024-08-20:
Maybe fix cosmetic: (internal clock 20:00:00 UTC, neter clock: 20:00:01) unless its an error in my terminal printout

fnorlund, 2024-08-21:
For whatever it's worth I have the same issue with this setup. It started sometime after 13-20 days uptime. 

[image]
[image]

Located in Sweden.

ArnieO, 2024-08-21:
> For whatever it's worth I have the same issue with this setup. It started sometime after 13-20 days uptime.

Thank you for reporting! 
Interesting - as all reports so far have been on the same hardware (Pow-P1 with ESP32-S2).
If the above identification in "Device information" is correct, you are on a [Wemos Lolin D32](https://www.wemos.cc/en/latest/d32/d32.html) that is based on an Espressif ESP32-WROOM-32 module.

Did you try the fix published yesterday? https://github.com/UtilitechAS/amsreader-firmware/issues/832#issuecomment-2299563461

senspix, 2024-08-21:
Same problem here on my **Pow-P1 on ESP32-S2**. I have rebooted, I have not checked with Telnet debug nor done a factory reset.
[image]

ArnieO, 2024-08-22:
> Same problem here 

Did you try the fix? https://github.com/UtilitechAS/amsreader-firmware/issues/832#issuecomment-2299563461

senspix, 2024-08-23:
I tried the firmware from #832 and it appears to work. I now have 12 hours of data across a calendar day change. 
I will report back if it stops working.

gskjold, 2024-08-24:
Found another edge case, would be good if you could test that also, to make sure I have not broken something i fixed before

[esp32s2.zip](https://github.com/user-attachments/files/16737768/esp32s2.zip)
[esp32c3.zip](https://github.com/user-attachments/files/16737769/esp32c3.zip)
[esp32.zip](https://github.com/user-attachments/files/16737770/esp32.zip)
[esp32s3.zip](https://github.com/user-attachments/files/16737771/esp32s3.zip)
[esp8266.zip](https://github.com/user-attachments/files/16737772/esp8266.zip)
[esp32solo.zip](https://github.com/user-attachments/files/16737773/esp32solo.zip)

toast1234, 2024-08-25:
Running f5c10c1 over the night. No issue seen so far.

fnorlund, 2024-08-25:
Installed f5c10c1 on [Wemos Lolin D32](https://www.wemos.cc/en/latest/d32/d32.html) that is based on an Espressif ESP32-WROOM-32 module this morning. Seems ok so far. Will let it run 24 hours.

gskjold, 2024-08-25:
Released new version containing the fix

closed 2024-08-25 by gskjold

fnorlund, 2024-08-26:
No issues with with f5c10c1 the last 26 hours. :-)