UtilitechAS/amsreader-firmware #833: Wrong export price if you use price region with support

aiboee, 2024-08-12:
**Describe the bug**
If you select a Norwegian price region that includes support, you get a wrong export price if the spot price is above the point where support kicks in.

**To Reproduce**
Select a Price region with support for example "NO2 with support"
Enable price fetch from remote server.
Observe the export price and compare it to the spot price when spot price is above lower limit for support.

**Expected behavior**
Export price must be the same as the spot price without being affected by the support.

**Hardware information:**
 - Country: Norway
 - AMS reader: Pow-U+

**Relevant firmware information:**
 - Version: 2.3.6

ArnieO, 2024-08-13:
Thank you for notifying us. 
Can you show an example of what the error is? Could be helpful in finding the bug.

aiboee, 2024-08-13:
For the hour between 19:00 and 20:00 on 12 august 2024 Nordpool export price is 2,2924 NOK and AMS reader Export price is 0,9 NOK.
Both integrations are configured to add 0.0125 NOK to the prices to include the compensation the grid company pays for export.

if I i do the following calculation i get same price as the AMS reader Export price.

Compensation from grid company=0,0125 NOK
Support is  90% of the amount that exceeds 73 øre/kWh
Support=(2,2924-0,0125-0,73) *0,9=1,39491
Nordpool price minus Support 2,2924-1,39491+0,0125=0,90999 or rounded to 0,9 NOK which is the same as reported by AMS reader.

Below is a screenshot showing the difference between AMS reader Export price and the Nordpool integration export price in Home Assistant for 12 august 2024.

[image]

ArnieO, 2024-08-14:
Thank you very much for your detailed analysis, very helpful!