UtilitechAS/amsreader-firmware #834: Spot price as export price

Belaial, 2024-08-26:
Hi,

I'm wondering if I could be missing something easy here, sorry if I have, I have tried to configure the "Price configuration", the scenario I am trying to achieve is the following.

Import = Static price
Export = Spot price

I have had no issues to configure a static import price but I see no way of selecting the import price as spot price, am I missing something or is this not currently implemented?

Reading here https://github.com/UtilitechAS/amsreader-firmware/wiki/Price-configuration it seems it's not to be possible currently, maybe this could be added in the future?

**Hardware information:**
 - Country: Finland (Belonging to SE3 price region)
 - Meter: Kamstrup
 - Encryption enabled: No
 - AMS reader: esp8266
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.3.7
 - MQTT: Yes
 - MQTT payload type: Home Asistant
 - HAN GPIO: GPIO5
 - HAN baud and parity: 2400 8N1
 - Temperature sensors
 - ENTSO-E API enabled: No

ArnieO, 2024-08-26:
Thank you for the enhancement suggestion, we will consider such enhancement for a future update.

gskjold, 2024-08-26:
We should update the documentation to be more clear, but the spot price is used unless a fixed price is set.

Belaial, 2024-08-26:
> We should update the documentation to be more clear, but the spot price is used unless a fixed price is set.

This is my current config regarding price

[image]

[image]

So, as I see it I have not set any fixed export price? Yet my graph show 0.00 €/kWh, should it not show current spot price if I understand your post correct?

[image]

Belaial, 2024-08-26:
Interesting, if I check in Home Assistant I do see a varying export price

[image]

I guess it just does not show in the GUI of the AMS reader?

Belaial, 2024-08-26:
Followup question, I assume this will work correctly even if no spot price is configured under Price configuration.

For every kWh I produce the power company takes 0.25 cent as a "transfer fee" of sorts, so I just add this to export price like so

[image]

I assume after adding this it will calculate `spot price - 0,25 = export price` ?

I will follow up this in Home Assistant, bad day today tho as the price is basically a flat line around 0....

gskjold, 2024-08-26:
> I assume after adding this it will calculate `spot price - 0,025 = export price` ?

Correct :)

> I will follow up this in Home Assistant, bad day today tho as the price is basically a flat line around 0....

I guess this is why the price in the export gauge shows 0.00? Probably currently it is less than 0.01? Let me know how that turns out.

Also, for NOK/SEK/DKK I have added subunit (øre) as an option. I guess we could to the same for EUR, so that when the price is really low it would show cent/kWh. Any thoughts?

Belaial, 2024-08-26:
> I guess this is why the price in the export gauge shows 0.00? Probably currently it is less than 0.01? Let me know how that turns out.

I will follow the graph and see :+1: , currently it looks like this, I added the transfer fee around 10, that's why there is a big drop there. Hopefully the prices are better tomorrow so I can see how it works, I will report back in a day or two.

[image]

> Also, for NOK/SEK/DKK I have added subunit (øre) as an option. I guess we could to the same for EUR, so that when the price is really low it would show cent/kWh. Any thoughts?

Please do if it's not to much work, it might be easier to read instead of `0.0255 EUR/kWh` like it is now

[image]

Also makes it more consistent I guess, if other currencies shows (øre) and so on that also Euro would behave the same.

ArnieO, 2024-08-26:
> Also, for NOK/SEK/DKK I have added subunit (øre) as an option. I guess we could to the same for EUR, so that when the price is really low it would show cent/kWh. Any thoughts?

Good idea! 👍

Belaial, 2024-08-30:
I guess we can close this issue, unless you want to keep it open and close it when the documentation is updated @gskjold

closed 2024-12-05 by gskjold