UtilitechAS/amsreader-firmware #837: 2.3.8, backup setting to file

borgemohus, 2024-09-01:
Function backup to file is not working when pressing "Download".
Respons "not found"

[image]

http://192.168.86.134/configfile.cfg?iw=true&im=true&ie=true&it=true&ih=true&ig=true&in=true&is=true

**Expected behavior**
Dialog-box where to save file

[image]

[image]

It seems to like some of the latest versions.
Tried with firefox 129.0.2 and chromium 128.0.6613.84

ArnieO, 2024-09-02:
I'm guessing from your nick and the customer database that you are on an older version Pow-U (layout v1.5), which uses ESP8266. 

I have tried to reproduce the problem with that same version Pow-U. 
- Firmware v2.3.7: No problem
- Firmware v2.3.8: No problem (tested on Firefox, Chrome and Edge).

That URL is strange, I don't see any URL in my browser - as this is a file download. 

What is that is assigned to the IP address 192.168.86.134? Is it the Pow-U or is it the device you're running the browser on?

A screendump of your congfig screen could potentially help.

ArnieO, 2024-09-04:
🤦 I read too fast yesterday eveneing: Your first post showed that 192.168.86.134 is the IP of the device. 

I see that you expect a dialogue box - but that will not happen. The button downloads to the download folder defined in the browser you're using. 

**Which browser are you using?** 
I have tried in Firefox, Chrome and Edge - and have not seen the URL shown as in your screenclip.

borgemohus, 2024-09-04:
> 🤦 I read too fast yesterday evening: Your first post showed that

192.168.86.134 is the ip of the amsreader, yes run via a google home-wifi

firefox and chromium version written earlier

Just tried chrome Version 128.0.6613.119 (Official Build) (64-bit) (on 
ubuntu 22.04), but same happends.

Børge

--------------TjZ0urJfkxNnnfbuhIMaih4u
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<!DOCTYPE html><html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    <p>192.168.86.134 is the ip of the amsreader, yes run via a google
      home-wifi<br>
    </p>
    <p>firefox and chromium version written earlier</p>
    <p>Just tried chrome Version 128.0.6613.119 (Official Build)
      (64-bit) (on ubuntu 22.04), but same happends.</p>
    <p><br>
    </p>
    Børge
    <div class="moz-cite-prefix">On 04.09.2024 16:52, ArnieO wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      <p dir="auto">🤦 I read too fast yesterday evening: Your first
        post showed that 192.168.86.134 is the IP of the device.</p>
      <p dir="auto">I see that you expect a dialogue box - but that will
        not happen. The button downloads to the download folder defined
        in the browser you're using.</p>
      <p dir="auto"><strong>Which browser are you using?</strong><br>
        I have tried in Firefox, Chrome and Edge - and have not seen the
        URL shown as in your screenclip.</p>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
          ID: <span>&lt;UtilitechAS/amsreader-firmware/issues/837/2329283421</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/UtilitechAS/amsreader-firmware/issues/837#issuecomment-2329283421",
"url": "https://github.com/UtilitechAS/amsreader-firmware/issues/837#issuecomment-2329283421",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
  </body>
</html>

--------------TjZ0urJfkxNnnfbuhIMaih4u--

borgemohus, 2024-09-05:
Sorry I don't have any windows enviroment.
I tried it from my Samsung PAD. Same result.

I will try to do some F12-trace with chrome soon.
No pri-case this

BR
Børge

ArnieO, 2024-09-12:
OK, we found the issue. You have set this:
[image]
And there is a bug, where"/ams" is not included in the URL.

ArnieO, 2024-09-12:
We see now that part of the bug is that if you remove the setting in Context, you will not be able to reach the device.
Do this:
- Set device in AP mode, and connect to it on 192.168.4.1
- Store a full backup (info/settings page)
- Do a Factory reset (bottom of config page)
- Reconfigure as when device was new
- Restore from backup

borgemohus, 2024-09-12:
I confirm that adding "ams" (context-value) to the url worked, and I got a dialog-box where to save the file.
The file contains valid values
http://192.168.86.134/ams/configfile.cfg?iw=true&im=true&ie=true&it=true&ih=true&ig=true&in=true&is=true&ic=true
I guess this is solved. Good work.
Børge

ArnieO, 2024-09-12:
Good to hear! 
And thank you for helping us notice that a bug was buried here.

closed 2024-11-09 by gskjold