UtilitechAS/amsreader-firmware #842: MQTT status switches to offline imediately after appearing online

shaohme, 2024-09-12:
**Describe the bug**
The AMS reader appears offline immediately after appearing online when connecting to MQTT.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup to publish to MQTT broker
2. Observe the MQTT topic configured for new messages
3. Reboot AMR reader
4. Observe that the topic <name>/status initially sets to 'online' after which it changes to 'offline'

**Expected behavior**
It should be kept in 'online' state as long as the device is running.

**Screenshots**
[image]
[image]

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup Omnipower
 - Encryption enabled yes
 - AMS reader: Pow-K+
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.3.8
 - MQTT: yes
 - MQTT payload type: Raw (full)
 - HAN GPIO:
 - HAN baud and parity: 2400 8E1
 - Temperature sensors:
 - ENTSO-E API enabled: yes

ArnieO, 2024-09-12:
Just to be sure what you experiencing: 
Does the device not transmit payloads? 
Or is it the problem that the transmitted status is set to "offline"?

shaohme, 2024-09-12:
> Just to be sure what you experiencing: Does the device not transmit payloads? 

I assume by 'payloads' you mean MQTT messages to the broker. If so, it does. 

> Or is it the problem that the transmitted status is set to "offline"?

Yes. It starts by setting the value 'online' but shortly after sets it to 'offline' and stays there, as show in the screenshot.

ArnieO, 2024-09-12:
> It starts by setting the value 'online' but shortly after sets it to 'offline' and stays there, as show in the screenshot.

And continues to send MQTT messages to the broker after that?

shaohme, 2024-09-12:
I have not setup a "Client ID" in my MQTT config before. When I set it, and reboot the reader, the problem disappeared :face_with_diagonal_mouth:

ArnieO, 2024-09-12:
I think each client has to "present itself" to the broker with an ID (that must not be the same as any other client). 
I guess that explains it!

If your setup now works as expected, I invite you to close this thread.

shaohme, 2024-09-13:
> I think each client has to "present itself" to the broker with an ID (that must not be the same as any other client). I guess that explains it!
> 
> If your setup now works as expected, I invite you to close this thread.

I guess, but should AMS not inform the user of this? Either by recommending or demand a client id?

ArnieO, 2024-09-13:
> I guess, but should AMS not inform the user of this? Either by recommending or demand a client id?

That is actually a good idea! This field could be auto populated with Hostname as default value. I will update [the Wiki](https://wiki.amsleser.no/e/en/HomeAutomation/Home-Assistant) now to remind that Client ID cannot be empty.

I mark this as an enhancement proposition for @gskjold to consider.

gskjold, 2024-11-03:
Making this field required when MQTT is enabled

closed 2024-11-09 by gskjold