UtilitechAS/amsreader-firmware #846: Two attributes have faulty state classes:

KDYR, 2024-09-24:
**Describe the bug**
In Home Assistant, in the "Developer Tool", tab statistics, two of the attributes of sensor.amsleser have a tag indicating there is an issue: "The state class '' of this entity is not supported."

Clicking "Fix issue" yields this text:

**To Reproduce**
1. Go to "Developer Tool" in Home Assistant
2. Go to Statistics tab
3. Find AMS-reader sensor/attribute

**Expected behavior**
As for other attributes, the right state class should be assigned by the integration

**Screenshots**
[image]

When clicking "Fix Issue"
[image]

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup]
 - Encryption enabled: yes
 - AMS reader: Pow-K+

**Relevant firmware information:**
 - Version: 2.3.8
 - MQTT: yes
 - MQTT payload type: Home-Assistant

ArnieO, 2024-09-25:
Thank you for reporting.

Just to be sure we have all relevant information to check/reproduce:
Which version HA are you running?

Eivind-V, 2024-09-27:
Same issue.

HA version 2024.10.0b2

https://rc.home-assistant.io/blog/2024/09/25/release-202410#repairs-for-long-term-statistics
[image]

[image]

Stewart555, 2024-10-03:
Seems to affect more sensors than the ones mentioned.

ArnieO, 2024-10-03:
Thank you for reporting. 
We will look into it for the upcoming release of v2.3.9 later this month.

danielcherubini, 2024-10-05:
An update, seems 2024.10.1 resolves this issue. 

https://www.home-assistant.io/changelogs/core-2024.10/#release-2024101---october-4

Nothing in the changelog jumps out at me

Stewart555, 2024-10-05:
For me the message about missing state classes disappeared right after
updating to 2024.10.1, but reappeared after some fairly short time. Unknown
how long, updated in the evening - issue not reported, next morning - issue
reported.
Maybe the issue is not detected by HA until there is an update of the
sensors in question, or the value is polled??

danielcherubini, 2024-10-05:
Yes. It's back. I gave it a good 30min wait, to check if it was just a reboot thing. But no.

Anyhow I'll wait till the next firmware, no big deal.

gskjold, 2024-11-03:
Although I cannot find the previous issue now, these sensors used to have a state class, but has since been removed because of the nature of these sensors. I cannot see that there is any suitable state class for these values... If you have a deep understanding of HA device and state classes, feel free to reopen with you knowledge of how to best configure these sensors.

Also, side note, this issue was based on warnings from a beta version of HA

closed 2024-11-03 by gskjold

Stewart555, 2024-11-03:
Just to be clear, the issue was based on warnings from a *release* version
of HA, not only a beta version.
I never run any beta versions of HA, and the issue was/is definitely still
there.
I personally do not use the information those sensors provided, so I just
deleted their entry, but the issue is still there for anyone not deleting
those entries.

joachimtingvold, 2024-12-25:
Not sure I see the problem of defining the state class? All five sensors that I have that error for, should fit into `
SensorStateClass.TOTAL_INCREASING`, at least according to the descriptions of the [three available classes](https://developers.home-assistant.io/docs/core/entity/sensor/#available-state-classes).

For clarity, that would be these five sensors;

```
sensor.amsreader1_max
sensor.amsreader1_threshold
sensor.amsreader1_hourcost
sensor.amsreader1_daycost
sensor.amsreader1_monthcost
```