UtilitechAS/amsreader-firmware #850: Real time gone from MQTT after power outage.

tux-, 2024-10-04:
We just had a power outage for about 1h 30m. Afterwards the only message sent to mqtt is a state message, no more real time data.

**Hardware information:**
 - Country: Norway
 - Meter: Aidon
 - Encryption enabled no
 - AMS reader: ESP32
 - Version: 2.3.8
 - ENTSO-E API enabled: no

NTP is enabled, tried multiple different servers, tried rebooting, but still date is set to 01.01.1970

Real Time (kW) graph is working on the web ui.

Communication with mqtt is working to some degree. as the periodic state does arrive at the broker.

ArnieO, 2024-10-05:
The date issue indicates that your device no longer has access to internet. 
The default NTP setting **_pool.ntp.org_** should work.

Can you share your config screen?

tux-, 2024-10-05:
I see no reason why it should have lost internet access, everything else on the network worked just fine. However, I left the device alone without rebooting or messing with it in any ways, and then it suddenly had correct datetime, and mqtt started working again. Maybe it was ntp.org that was down a little? No clue, but works again now. Lets hope this is not a problem on next outtake, cause I really want to log initial loads right after outages :)

ArnieO, 2024-10-06:
The question could then be whether the firmware is able to recover gracefully from a situation after a power outage, where there could be a subsequent delay during which Wi-Fi and internet connection is re-established. 

So you let it just sit there - and it recovered. Does the uptime counter (upper left) give an indication whether it had rebooted by itself while you left it alone?

tux-, 2024-10-06:
Oh, that it did not follow. I will keep an eye on it if I get a chance if there is a new outage one day. My guess is that it lost track of time when power was out, and then when booting back up the time server was unavailable so it just tried again a bit later and all was fine. I might look into setting up an internal time server so that it does not need internet for it just to make sure. Since this is more likely a time server issue, I will close this ticket. Thank for clearing things up!

closed 2024-10-06 by tux-

ArnieO, 2024-10-06:
> My guess is that it lost track of time when power was out, and then when booting back up the time server was unavailable so it just tried again a bit later and all was fine.

I think you are right, but need @gskjold to confirm.

Jerroder, 2024-11-03:
I had a power outage a few days ago and I just noticed I have the same problem, no data reach Home Assistant since then. My date is 1970 and rebooting/changing the NTP (even to pool.ntp.org) doesn't fix the problem. My uptime is 17 hours and it still hasn't updated the date.

My firewall does allow traffic from my amsreader's IPv4 and v6 to anywhere with protocol UDP and port 123. Any idea how to force update the date or set it manually maybe?

ArnieO, 2024-11-04:
Have you tried rebooting your device?

Jerroder, 2024-11-04:
Sorry I should have been clearer. The uptime was 17 hours, I rebooted before that (after the outage) and after 17 hours, and the date is still wrong. Right now the uptime is 19 hours, I rebooted yesterday before writing my comment and still nothing.

Jerroder, 2024-11-04:
> Wrong date indicates that it does not have contact out to internet.

Ok so I fixed it by allowing all protocol to any ports traffic from the PowP1's IP through the firewall. Allowing UDP:123 wasn't enough for some reason?

ArnieO, 2024-11-04:
> Allowing UDP:123 wasn't enough for some reason?

Sorry, that question surpasses my network competence. 😅 
-> @gskjold can maybe comment?