UtilitechAS/amsreader-firmware #851: Price service unavailable

ojessie, 2024-10-05:
Since the update to 2.3.8 the price service for Austria is not available. 

- Country: Austria
- Meter: Sagemcom
- Encryption: yes
- AMS Reader: Chip: esp32s2 (160MHz)
Version: 2.3.8
- MQTT: yes

The exact error message is: "Price service: Price unavailable, not found"

ArnieO, 2024-10-05:
This seems to be an issue with Entso-E (where we get the prices): 
[This link ](https://transparency.entsoe.eu/transmission-domain/r2/dayAheadPrices/show?name=&defaultValue=false&viewType=GRAPH&areaType=BZN&atch=false&dateTime.dateTime=05.10.2024+00:00|CET|DAY&biddingZone.values=CTY|10YAT-APG------L!BZN|10YAT-APG------L&resolution.values=PT15M&resolution.values=PT30M&resolution.values=PT60M&dateTime.timezone=CET_CEST&dateTime.timezone_input=CET+(UTC+1)+/+CEST+(UTC+2))should show today's price graph in Austria. 

What I get is this:
[image]

Unfortunately, Entso-E has data outages sometimes. You just have to wait for it to be restored.

ojessie, 2024-10-06:
> Von "ArnieO" ***@***.***>
> An "UtilitechAS/amsreader-firmware"
> Cc "ojessie" ***@***.***>; "Author"
> Datum 05.10.2024 16:33:14
> Betreff Re: [UtilitechAS/amsreader-firmware] Price service unavailable
> (Issue #851)

Thanks for clarification.

Br

cmock, 2024-10-18:
I'm gonna reopen this as the "old" ENTSO-E api seems to be deprecated, visiting the above link now shows:
[image]

ArnieO, 2024-10-19:
Thank you for the heads-up; we will look into this.

gskjold, 2024-10-20:
This is just the web view, API is not affected by tihs

closed 2024-10-20 by gskjold