UtilitechAS/amsreader-firmware #853: Option for how long in the future to look for cheapest price

MrAdam, 2024-10-07:
Sometimes the "cheapest 3hr period ahead" exposed to Home Assistant shows more than 12+ hours, which is a bit annoying when most of the time I just want to know when to start the dishwasher when going to bed, so looking so far into the future is not relevant for me. I know this is a thing which might be specific to just me, but making the lookahead time configurable would be a nice to have feature.

I have considered doing the calculate in Home Assistant based on the prices, but I didn't find this easy, especially since the prices are exposed as one entity per hour, instead of an array (see #838)

ArnieO, 2024-10-08:
Thank you for this enhancement proposition, we'll consider whether to implement it. 

As a general comment, we are approaching the available capacity on the ESP32 version we're using, so we need to be selective regarding which new features we implement.

MrAdam, 2024-10-08:
> Thank you for this enhancement proposition, we'll consider whether to implement it.
> 
> As a general comment, we are approaching the available capacity on the ESP32 version we're using, so we need to be selective regarding which new features we implement.

Thank you. And thanks for creating such a great little device, and keeping it up-to-date 😄

kaistian, 2024-10-14:
Since your are using HA you can use this macro https://github.com/TheFes/cheapest-energy-hours it's available in HACS.