UtilitechAS/amsreader-firmware #855: Negative Amp on L1

danielcherubini, 2024-10-09:
**Describe the bug**
Hey, so looking at my HA history, all of a sudden around sept 28th my L1 started reporting negative amps, I'm not sure what is going on, since I haven't updated the firmware or done anything to the system at that time. 

I first thought it might have been HA, since I do update that, but I checked the AMS dashboard, and yep, L1 was reporting negative amps. 

**Expected behavior**
I do believe it shouldn't be negative, that it should be most likely the postive value of what's being reported.

**Screenshots**
[image]
[image]

**Hardware information:**
 - Country: Norway
 - Meter: Aidon
 - Encryption enabled no
 - AMS reader: Pow-U+
 - M-bus adapter (if applicable): no

**Relevant firmware information:**
 - Version: 2.3.8
 - MQTT: yes
 - MQTT payload type: JSON
 - HAN GPIO: 
 - HAN baud and parity: 2400 8N1 (also tried 8E1
 - Temperature sensors None
 - ENTSO-E API enabled: yes

**Additional context**
I recently updated the firmware to the latest and the issue persists. Attatched is a telnet log with verbose logging.

[telnet.log](https://github.com/user-attachments/files/17302815/telnet.log)

ArnieO, 2024-10-09:
As indicated in this thread: https://github.com/UtilitechAS/amsreader-firmware/issues/764 some meters report negative current if there is export (solar panels etc). Do you have solar panels?

Can you include the power gauge on a screenshot taken while the current value is negative - so we can see whether the current values seem reasonable?

Also, please run a [Telnet debug](https://www.amsleser.no/module/ets_blog/blog?id_post=24) while current is negative - and post (attached text file) the output. In that way we can check the raw data from the meter. (Or you can do the check yourself by using this tool: https://www.gurux.fi/GuruxDLMSTranslator )

danielcherubini, 2024-10-09:
Hi,
Yeah I read that thread, and no I don't export any power, no solar, nothing. 

Here's with the power gauge

[image]

I noticed that if it goes over about 1100w then the value goes positive

[image]

Here's a log when it's negative
[telnet2.log](https://github.com/user-attachments/files/17305091/telnet2.log)

and here's one when it's positive.
[telnet3.log](https://github.com/user-attachments/files/17305096/telnet3.log)

danielcherubini, 2024-10-09:
Actually that part about where it's over 1100w going to positive.. i take that back

[image]

ArnieO, 2024-10-09:
Thank you, and let us look at the content of your first logfile (while current is negative), using https://www.gurux.fi/GuruxDLMSTranslator .

**ANALYSIS OF [telnet2.log](https://github.com/user-attachments/files/17305091/telnet2.log)**
We copy the first section in the logfile marked "DLMS frame", and convert it to XML:
```
<DataNotification>
  <LongInvokeIdAndPriority Value="40000000" />
  <DateTime Value="" />
  <NotificationBody>
    <DataValue>
      <Array Qty="0C" >
        <Structure Qty="02" >
          <!--1.1.0.2.129.255-->
          <OctetString Value="0101000281FF" />
          <String Value="AIDON_V0001" />
        </Structure>
        <Structure Qty="02" >
          <!--0.0.96.1.0.255-->
          <OctetString Value="0000600100FF" />
          <String Value="7359992898552476" />
        </Structure>
        <Structure Qty="02" >
          <!--0.0.96.1.7.255-->
          <OctetString Value="0000600107FF" />
          <String Value="6525" />
        </Structure>
        <Structure Qty="03" >
          <!--1.0.1.7.0.255-->
          <OctetString Value="0100010700FF" />
          <UInt32 Value="00000226" />
          <Structure Qty="02" >
            <Int8 Value="00" />
            <Enum Value="1B" />
          </Structure>
        </Structure>
        <Structure Qty="03" >
          <!--1.0.2.7.0.255-->
          <OctetString Value="0100020700FF" />
          <UInt32 Value="00000000" />
          <Structure Qty="02" >
            <Int8 Value="00" />
            <Enum Value="1B" />
          </Structure>
        </Structure>
        <Structure Qty="03" >
          <!--1.0.3.7.0.255-->
          <OctetString Value="0100030700FF" />
          <UInt32 Value="00000000" />
          <Structure Qty="02" >
            <Int8 Value="00" />
            <Enum Value="1D" />
          </Structure>
        </Structure>
        <Structure Qty="03" >
          <!--1.0.4.7.0.255-->
          <OctetString Value="0100040700FF" />
          <UInt32 Value="00000312" />
          <Structure Qty="02" >
            <Int8 Value="00" />
            <Enum Value="1D" />
          </Structure>
        </Structure>
        <Structure Qty="03" >
          <!--1.0.31.7.0.255-->
          <OctetString Value="01001F0700FF" />
          <Int16 Value="FFEB" />
          <Structure Qty="02" >
            <Int8 Value="FF" />
            <Enum Value="21" />
          </Structure>
[... 69 lines cut ...]

So the first payload / data frame of your file [telnet2.log](https://github.com/user-attachments/files/17305091/telnet2.log) reports negative current on L1, positive on L3.

The next 3 frames are short List 1 frames (see Interface spec), only containing Instantaneous active import power (OBIS code 1.7.0).

The second List 2 frame in this logging: Current on L1 is reported as 0xFFEA -> **-2,2 A**

So there are two consecutive frames with negative current on phase L1. 

******************
**ANALYSIS OF Logfile: [telnet3.log](https://github.com/user-attachments/files/17305096/telnet3.log)**

The first List 2 frame reports 0x0034 (5,2 A) on L1 and 0x0049 (7,3 A) on L3. So no negative number in this frame.

The second List 2 frame reports 0x0034 (5,2 A) on L1 and 0x004A (7,4 A) on L3. No negative number.

******************
**CONCLUSION**
Analysis of the raw data confirms that your meter sometimes reports negative current values on phase 1. 
As you do not have power generation in your installation, this is not normal.
I recommend you to contact your grid company and ask if they are aware of such a situation in your installation (they should be, if they do a follow-up of the data they receive), and ask them what the reason could be.

I think this **_could_** be an indication of a problem at grid level in your area/neighborhood - so it could be that your grid company is already aware of the situation and working on it.

**Please be so kind as to report back here when you have been in contact with your grid company, as we would like to learn from this situation; as it might happen again. **

danielcherubini, 2024-10-17:
I'll comment here when, and if i get an answer back from my grid company... Chances are slim, the person I talked to had no idea what i was talking about.