UtilitechAS/amsreader-firmware #856: crashes/endless reboots with ENTSO-E API key configured

cmock, 2024-10-18:
**Describe the bug**
I got a ENTSO-E API key today, configured it, and my ESP started rebooting endlessly.

It comes up, establishes the MQTT connection, publishes some things, and then reboots. Maximum uptime I've seen was 15 seconds.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure ENTSO-E API key (just put the API key in the relevant field, no other changes)
2. Save
3. Watch in horror 

**Expected behavior**

Given #851 and no prices shown in the AMS reader web UI, I expected either prices to be shown or an error message to be displayed.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: Austria
 - Meter: Sagemcom T210-D
 - Encryption enabled yes
 - AMS reader: ESP32
 - M-bus adapter (if applicable): self-built

**Relevant firmware information:**
 - Version: 2.3.8
 - MQTT: yes
 - MQTT payload type: HomeAssistant
 - HAN GPIO: 16
 - HAN baud and parity: 2400 8E1
 - Temperature sensors: n/a
 - ENTSO-E API enabled: yes

**Additional context**
Add any other context about the problem here.

ArnieO, 2024-10-19:
Thank you for reporting. 
Potentially linked to changes in Entso-E API: https://github.com/UtilitechAS/amsreader-firmware/issues/851

cmock, 2024-10-19:
Probably. Looking at the parser code in EntsoeA44Parser.cpp, I don't think this can handle the fact that the XML I get from the API does contain two `<TimeSeries>` elements, the first with 15 minute date and thus 96 values, the second with hourly data... (Wild guess, I have no clue what it returned before the changes).

ArnieO, 2024-10-20:
Are you reading the Entso-E API directly?
The recommended method has for quite a while been to read from our intermediate server, where we have implemented (for some countries) backup solutions to Entso-E.

To do that, you activate "price fetch" but without entering Entso-E API key. 
This is how it looks for me:
[image]

cmock, 2024-10-20:
For my above comment, I looked at the URL the code generates when an API key is set and requested that with curl.

In the AMS reader, I had no API key configured until #851 happened, which left me with no prices. So I requested API access from ENTSO-E and configured the key, and then the reboot loop happened (which I resolved by blocking access to the ENTSO-E server in my firewall, because there was no other way to access the config page to remove the key).

ArnieO, 2024-10-20:
> For my above comment, I looked at the URL the code generates when an API key is set and requested that with curl.
>
> In the AMS reader, I had no API key configured until #851 happened, which left me [...]

OK, thank you for additional information. I tried to enter a Swedish price zone in my Pow-K now - and can just confirm our intermediate server is not delivering prices for Sweden (and probably other countries) due to the API change. 
Please give us a few days to fix it; priority will be on getting the the API running on our server - which should remedy this unfortunate situation.

tronde-ams, 2024-10-20:
I have API key and get prices for NO1. Was a hickup a few hours a couple a days ago, but else OK. 
Prices are the same as shown on https://newtransparency.entsoe.eu

FW 2.3.8 and generic ESP32.

gskjold, 2024-10-20:
There is no API change, the documentation is exactly the same as before (https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html)

As for the reboot loop, I am unable to reproduce this....

gskjold, 2024-10-20:
I see now what it could be, the AT document returns a time series for every 15 minutes, the code does not account for this. Will need some adjustments if we are to support this. How many countries change the price every 15m?

gskjold, 2024-10-20:
Try this firmware:
[esp32s3.zip](https://github.com/user-attachments/files/17451604/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/17451605/esp32solo.zip)
[esp32.zip](https://github.com/user-attachments/files/17451606/esp32.zip)
[esp8266.zip](https://github.com/user-attachments/files/17451607/esp8266.zip)
[esp32c3.zip](https://github.com/user-attachments/files/17451608/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/17451609/esp32s2.zip)

ArnieO, 2024-10-20:
> I see now what it could be, the AT document returns a time series for every 15 minutes, the code does not account for this. Will need some adjustments if we are to support this. How many countries change the price every 15m?

Both Sweden and Finland reports issues with fetching data from our intermediate web server, presumably because of this. I do not know which other countries could be affected.

ArnieO, 2024-10-20:
> There is no API change, the documentation is exactly the same as before (https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html)
> 
> As for the reboot loop, I am unable to reproduce this....

Data presentation has moved from https://transparency.entsoe.eu to https://newtransparency.entsoe.eu, where things look a bit... unfinished. The site seems to be a transition phase. I have logged in, but when trying to find information I find a lot of non-working links. 

Are we sure the API description on https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html is still valid for fetching data from newtransparency.entsoe.eu?

cmock, 2024-10-20:
I flashed the firmware above and now both without and with API key everything works as expected. No crashes, and prices are displayed.

closed 2024-11-09 by gskjold