UtilitechAS/amsreader-firmware #857: Add capacity grid sensors to adjust power consumption to stay below a given threshold

fraas, 2024-10-21:
This is not related to a problem. We could always write our own sensors, but the best thing would have been if this device does it all internally and then we can make use of the sensors the way we want.

I want pretty much the same sensors as these powersaver sensors provide: https://powersaver.no/examples/example-grid-tariff-capacity-part.html

How it is being handled doesn't have to be exactly the same as powersavers example. The most important is a sensor that makes it easy to understand if we have to reduce the current power usage to stay below a configured capacity threshold, and some information about how close or far away we are. Based on these sensors it should be easier to write automation's to turn off elements/entities with high power usage to stay below the configured threshold. And turn them back on again at the start of a new hour.

This should make it much easier to add capacity automation's for those who's not capable of writing their own sensors. I also think this would make the device more complete as a all-in-one power-metering-device.

fraas, 2026-11-12:
Hi @ArnieO,

Just wanted to follow up on this enhancement request, as it's been a while since it was originally suggested.

Has there been any further consideration of implementing these capacity-related sensors, or is this something that's likely to remain on the backlog indefinitely? I completely understand that development priorities change, but it would be nice to know whether this is still being considered.

You previously mentioned that the ESP32's 4 MB flash capacity was becoming a limitation, which is understandable.

With that in mind, could this functionality potentially be implemented as an optional add-on through the marketplace instead of being included in the core firmware? If the architecture supports it, this might be a way to provide additional functionality without increasing the size of the base firmware for everyone.

I still believe these sensors would be a valuable addition, especially for users who want to automate their power consumption to stay below a configured grid capacity threshold without having to create complex sensors themselves.

Thanks again for all the work you put into this project!

ArnieO, 2026-11-13:
Ping @gskjold